riscv-collab/riscv-openocd #589: compile error when ./configure --enable-jlink

iysheng, 2021-04-03:
When I want to use jtag interface to connect with riscv chip, so i try to compile openocd from source, but I get this compile error like:
[image]
I add the ```log.h``` to file ```src/jtag/drivers/usb_common.h```, 
``` diff
diff --git a/src/jtag/drivers/usb_common.h b/src/jtag/drivers/usb_common.h
index 4d2bd2686..655feeb10 100644
--- a/src/jtag/drivers/usb_common.h
+++ b/src/jtag/drivers/usb_common.h
@@ -19,6 +19,7 @@
 #define OPENOCD_JTAG_DRIVERS_USB_COMMON_H
 
 #include <usb.h>
+#include "log.h"
 
 int jtag_usb_open(const uint16_t vids[], const uint16_t pids[],
 		struct usb_dev_handle **out);
```
then I could compile success. 
I get confused why could this happen?

timsifive, 2021-04-06:
@JanMatCodasip mentioned "The issue can be reproduced with ./configure --enable-jlink [...]." but when I try that I don't see this error. What arguments did you use for the configure script?

The correct fix is to include helper/types.h, and possibly helper/replacements.h as well. (This is what I had to do in jtag_usb_common.h.)

JanMatCodasip, 2021-04-06:
> @JanMatCodasip mentioned "The issue can be reproduced with ./configure --enable-jlink [...]." but when I try that I don't see this error. 

That is strange. The only arguments to "./configure" I used were `./configure --prefix=`pwd`/install --enable-jlink` and that was sufficient to reproduce this issue. I started with clean copy of the source - no previous compilation. The platform was CentOS 7.

timsifive, 2021-04-07:
I just did:
```
git clone https://github.com/riscv/riscv-openocd
cd riscv-openocd/
./bootstrap 
./configure --prefix=`pwd`/install --enable-jlink
make
```
and it built fine.

If it's reproducing for you, can you try:
```
diff --git a/src/jtag/drivers/usb_common.h b/src/jtag/drivers/usb_common.h
index 4d2bd2686..7f8bd0a45 100644
--- a/src/jtag/drivers/usb_common.h
+++ b/src/jtag/drivers/usb_common.h
@@ -20,6 +20,9 @@
 
 #include <usb.h>
 
+#include <helper/replacements.h>
+#include <helper/types.h>
+
 int jtag_usb_open(const uint16_t vids[], const uint16_t pids[],
                struct usb_dev_handle **out);
 
```
to see if that fixes it? (Really replacements.h might not even be required.)

TommyMurphyTM1234, 2021-04-07:
I tried to reproduce this on CentOS 7.9 but can't even get past bootstrap even though I have "OpenOCD dependencies" mentioned here installed:

https://github.com/riscv/riscv-openocd

```
[vagrant@centos7 riscv-openocd]$ ./bootstrap
+ aclocal --warnings=all -I src/m4
+ libtoolize --automake --copy
+ autoconf --warnings=all
configure.ac:25: warning: The macro `AC_PROG_CC_C99' is obsolete.
configure.ac:25: You should run autoupdate.
src/m4/std-gnu11.m4:9: AC_PROG_CC_C99 is expanded from...
configure.ac:25: the top level
configure.ac:13: error: possibly undefined macro: AC_MSG_WARN
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:34: error: possibly undefined macro: AC_DISABLE_SHARED
configure.ac:210: error: possibly undefined macro: AC_DEFINE
configure.ac:351: error: possibly undefined macro: AC_MSG_ERROR
configure.ac:650: error: possibly undefined macro: AC_MSG_NOTICE
``` 
If I run autoupdate as advised I and then bootstrap again I get this:

```
[vagrant@centos7 riscv-openocd]$ autoupdate
configure.ac:24: warning: AC_PROG_CC_C99 is obsolete; use AC_PROG_CC

[vagrant@centos7 riscv-openocd]$ ./bootstrap
+ aclocal --warnings=all -I src/m4
configure.ac:25: error: AC_REQUIRE(): cannot be used outside of an AC_DEFUN'd macro
configure.ac:25: the top level
autom4te: /usr/bin/m4 failed with exit status: 1
aclocal: error: echo failed with exit status: 1
```

Perhaps the main readme https://github.com/riscv/riscv-openocd is not up to date with the actual versions of the tools that are required?

timsifive, 2021-04-07:
@TommyMurphyTM1234, this sounds like a general OpenOCD issue and nothing to do with this fork.

iysheng, 2021-04-09:
> ./configure --prefix=`pwd`/install --enable-jlink

I did this with , but when i first make ,i get error as:
```
src/jtag/drivers/cmsis_dap_usb_hid.c:102:19: error: implicit declaration of function ‘strlen’ [-Werror=implicit-function-declaration]
  102 |     size_t len = (strlen(serial) + 1) * sizeof(wchar_t);
      |                   ^~~~~~
src/jtag/drivers/cmsis_dap_usb_hid.c:102:19: error: incompatible implicit declaration of built-in function ‘strlen’ [-Werror]
src/jtag/drivers/cmsis_dap_usb_hid.c:42:1: note: include ‘<string.h>’ or provide a declaration of ‘strlen’
   41 | #include "cmsis_dap.h"
  +++ |+#include <string.h>
   42 |
src/jtag/drivers/cmsis_dap_usb_hid.c: In function ‘cmsis_dap_hid_write’:
src/jtag/drivers/cmsis_dap_usb_hid.c:190:2: error: implicit declaration of function ‘memset’ [-Werror=implicit-function-declaration]
  190 |  memset(dap->packet_buffer + txlen, 0, dap->packet_size - txlen);
      |  ^~~~~~
src/jtag/drivers/cmsis_dap_usb_hid.c:190:2: error: incompatible implicit declaration of built-in function ‘memset’ [-Werror]
src/jtag/drivers/cmsis_dap_usb_hid.c:190:2: note: include ‘<string.h>’ or provide a declaration of ‘memset’
cc1: all warnings being treated as errors
```
after I add ```<string.h>``` in file ```src/jtag/drivers/cmsis_dap_usb_hid.c```,
i continue to make get error as i mentioned before:
```
In file included from src/jtag/drivers/usb_common.c:21:
src/jtag/drivers/usb_common.h:23:25: error: unknown type name ‘uint16_t’
   23 | int jtag_usb_open(const uint16_t vids[], const uint16_t pids[],
      |                         ^~~~~~~~
src/jtag/drivers/usb_common.h:23:48: error: unknown type name ‘uint16_t’
   23 | int jtag_usb_open(const uint16_t vids[], const uint16_t pids[],
      |                                                ^~~~~~~~
src/jtag/drivers/usb_common.c:36:5: error: conflicting types for ‘jtag_usb_open’
   36 | int jtag_usb_open(const uint16_t vids[], const uint16_t pids[],
      |     ^~~~~~~~~~~~~
In file included from src/jtag/drivers/usb_common.c:21:
src/jtag/drivers/usb_common.h:23:5: note: previous declaration of ‘jtag_usb_open’ was here
   23 | int jtag_usb_open(const uint16_t vids[], const uint16_t pids[],
      |     ^~~~~~~~~~~~~
make[2]: *** [Makefile:3853: src/jtag/drivers/libocdjtagdrivers_la-usb_common.lo] Error 1
make[2]: Leaving directory '/tmp/riscv-openocd'
make[1]: *** [Makefile:5373: all-recursive] Error 1
make[1]: Leaving directory '/tmp/riscv-openocd'
make: *** [Makefile:2231: all] Error 2
```
``` bash
▸ neofetch
          /:-------------:\          yangyongsheng@red.WS
       :-------------------::        --------------------
     :-----------/shhOHbmp---:\      OS: Fedora 32 (Thirty Two) x86_64
   /-----------omMMMNNNMMD  ---:     Host: -------------------------------
  :-----------sMMMMNMNMP.    ---:    Kernel: 5.8.15-201.fc32.x86_64
 :-----------:MMMdP-------    ---\   Uptime: 15 hours, 9 mins
,------------:MMMd--------    ---:   Packages: 2574 (rpm)
:------------:MMMd-------    .---:   Shell: bash 5.0.17
:----    oNMMMMMMMMMNho     .----:   Resolution: 1366x768
:--     .+shhhMMMmhhy++   .------/   WM: dwm
:-    -------:MMMd--------------:    Theme: Adwaita [GTK2/3]
:-   --------/MMMd-------------;     Icons: Adwaita [GTK2/3]
:-    ------/hMMMy------------:      Terminal: alacritty
:-- :dMNdhhdNMMNo------------;       CPU: Intel i5-7200U (4) @ 3.100GHz
:---:sdNMMMMNds:------------:        GPU: Intel HD Graphics 620
:------:://:-------------::          GPU: NVIDIA GeForce 940MX
:---------------------://            Memory: 1803MiB / 7717MiB
```

closed 2021-07-19 by timsifive