riscv-collab/riscv-openocd #591: openocd: src/rtos/FreeRTOS.c:324: populate_offset_size: Assertion `info[i].size > 0' failed

t-8ch, 2021-04-09:
Since 737f013ada404fdd620c2c0d4f1cb318a0ff64b2 I receive the error from the subject directly after starting when using `target/gd32vf103.cfg`.

It seems that
* `freertos->ubasetype_size == -1`
* because `riscv_data_bits() == -1`
* because `riscv_xlen() == -1`
* because it's not overwritten after `riscv_info_init()`.

Steps to reproduce:
```tcl
# test.cfg
source [find interface/dummy.cfg]
source [find target/gd32vf103.cfg]
$_TARGETNAME configure -rtos FreeRTOS
```
```sh
$ openocd -f test.cfg
Open On-Chip Debugger 0.11.0-rc2+dev-00994-g50a5971be (2021-04-09-17:15)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
openocd: src/rtos/FreeRTOS.c:324: populate_offset_size: Assertion `info[i].size > 0' failed.
Aborted (core dumped)
```

timsifive, 2021-04-09:
I don't have access to one of those targets. Can you reproduce this with `openocd -d` and attach the full output to this ticket?

t-8ch, 2021-04-09:
You should not need to have access to the target. It segfaults before it even tries to interact with the hardware.
(And it uses the dummy adapter)

<details>
<summary>debug.log</summary>

```
Open On-Chip Debugger 0.11.0-rc2+dev-00994-g50a5971be (2021-04-09-17:15)  
Licensed under GNU GPL v2                                                                                 
For bug reports, read                                                                                     
        http://openocd.org/doc/doxygen/bugs.html                                                          
User : 13 0 options.c:64 configuration_output_handler(): debug_level: 3      
User : 14 0 options.c:64 configuration_output_handler():                    
Debug: 15 0 options.c:245 add_default_dirs(): bindir=/tmp/openocd/bin
Debug: 16 0 options.c:246 add_default_dirs(): pkgdatadir=/tmp/openocd/share/openocd
Debug: 17 0 options.c:247 add_default_dirs(): exepath=/tmp/openocd/bin       
Debug: 18 0 options.c:248 add_default_dirs(): bin2data=../share/openocd       
Debug: 19 0 configuration.c:44 add_script_search_dir(): adding /home/t-8ch/.config/openocd
Debug: 20 0 configuration.c:44 add_script_search_dir(): adding /home/t-8ch/.openocd
Debug: 21 0 configuration.c:44 add_script_search_dir(): adding /tmp/openocd/bin/../share/openocd/site
Debug: 22 0 configuration.c:44 add_script_search_dir(): adding /tmp/openocd/bin/../share/openocd/scripts
Debug: 23 0 configuration.c:99 find_file(): found /tmp/test.cfg
Debug: 24 0 configuration.c:99 find_file(): found /tmp/openocd/bin/../share/openocd/scripts/interface/dummy.cfg
Debug: 25 0 command.c:146 script_debug(): command - adapter driver dummy
Info : 27 0 transport.c:118 allow_transports(): only one transport option; autoselect 'jtag'
Debug: 28 0 configuration.c:99 find_file(): found /tmp/openocd/bin/../share/openocd/scripts/target/gd32vf103.cfg
Debug: 29 0 command.c:146 script_debug(): command - reset_config srst_nogate
Debug: 31 0 command.c:146 script_debug(): command - jtag newtap gd32vf103 cpu -irlen 5 -expected-id 0x1000563d
Debug: 32 0 tcl.c:571 jim_newtap_cmd(): Creating New Tap, Chip: gd32vf103, Tap: cpu, Dotted: gd32vf103.cpu, 4 params
Debug: 33 0 tcl.c:596 jim_newtap_cmd(): Processing option: -irlen
Debug: 34 0 tcl.c:596 jim_newtap_cmd(): Processing option: -expected-id   
Debug: 35 0 core.c:1485 jtag_tap_init(): Created Tap: gd32vf103.cpu @ abs position 0, irlen 5, capture: 0x1 mask: 0x3
Debug: 36 0 command.c:146 script_debug(): command - jtag newtap gd32vf103 bs -irlen 5 -expected-id 0x790007a3
Debug: 37 0 tcl.c:571 jim_newtap_cmd(): Creating New Tap, Chip: gd32vf103, Tap: bs, Dotted: gd32vf103.bs, 4 params
Debug: 38 0 tcl.c:596 jim_newtap_cmd(): Processing option: -irlen
Debug: 39 0 tcl.c:596 jim_newtap_cmd(): Processing option: -expected-id
Debug: 40 0 core.c:1485 jtag_tap_init(): Created Tap: gd32vf103.bs @ abs position 1, irlen 5, capture: 0x1 mask: 0x3
Debug: 41 0 command.c:146 script_debug(): command - target create gd32vf103.cpu riscv -chain-position gd32vf103.cpu
Debug: 42 0 target.c:2174 target_free_all_working_areas_restore(): freeing all working areas
Debug: 43 0 riscv.c:428 riscv_create_target(): riscv_create_target()
Debug: 44 0 command.c:146 script_debug(): command - gd32vf103.cpu riscv set_enable_virt2phys off
Debug: 46 0 command.c:146 script_debug(): command - riscv set_mem_access progbuf          
Debug: 48 0 command.c:146 script_debug(): command - gd32vf103.cpu configure -work-area-phys 0x20000000 -work-area-size 0x1000 -work-area-backup 1
Debug: 49 0 target.c:2174 target_free_all_working_areas_restore(): freeing all working areas
Debug: 50 0 target.c:2174 target_free_all_working_areas_restore(): freeing all working areas
Debug: 51 0 target.c:2174 target_free_all_working_areas_restore(): freeing all working areas
Debug: 52 0 command.c:146 script_debug(): command - flash bank gd32vf103.flash gd32vf103 0x08000000 0 0 0 gd32vf103.cpu                                                                                              
Debug: 54 0 tcl.c:1320 handle_flash_bank_command(): 'gd32vf103' driver usage field missing
Debug: 55 0 command.c:146 script_debug(): command - flash bank gd32vf103.flashalias virtual 0x0 0 0 0 gd32vf103.cpu gd32vf103.flash                                                                                  
Debug: 57 0 tcl.c:1320 handle_flash_bank_command(): 'virtual' driver usage field missing
Debug: 58 0 command.c:146 script_debug(): command - gd32vf103.cpu configure -event reset-assert           
        set dmcontrol           0x10                 
        set dmcontrol_dmactive  [expr 1 << 0]
        set dmcontrol_haltreq   [expr 1 << 31]

        global _RESETMODE                                                                                 
        global _TARGETNAME

        # Halt the core so that we can write to memory. We do this first so               
        # that it doesn't clobber our dmcontrol configuration.                              
        halt                                                                                              
                                                     
        # Set haltreq appropriately for the type of reset we're doing. This           
        # replicates what the generic RISC-V reset_assert() function would
        # do if we weren't overriding it. The $_RESETMODE hack sucks, but                                 
[... cut ...]
                                                                                                          
Debug: 59 0 command.c:146 script_debug(): command - gd32vf103.cpu configure -rtos FreeRTOS
openocd: src/rtos/FreeRTOS.c:324: populate_offset_size: Assertion `info[i].size > 0' failed.                                                                                                                         
Debug: 60 0 server.c:617 sig_handler(): Terminating on Signal 6                             
Aborted (core dumped)  
```
</details>

timsifive, 2021-04-09:
Thanks for explaining. Sorry for my sloppy reading earlier. I have reproduced this.

timsifive, 2021-04-09:
I didn't run into this earlier because I was using `-rtos auto` instead of `-rtos FreeRTOS`. In my experience `-rtos FreeRTOS` didn't work at all, but maybe that was something about my setup.

I will fix this assertion, but I am curious if it works for you if you use `-rtos auto`.

t-8ch, 2021-04-10:
> I will fix this assertion, but I am curious if it works for you if you use `-rtos auto`.

Indeed it does work with `-rtos auto`

t-8ch, 2021-04-10:
> In my experience `-rtos FreeRTOS` didn't work at all, but maybe that was something about my setup.

FYI I used `-rtos FreeRTOS` because the autodetection didn't work for my OS in the past. Now it does and this is just an artifact of that history.
But `-rtos FreeRTOS` works fine for me in general.

PR #593 (2021-04-13): Don't check target_address_bits() until we need it.
Fixes #591, and also makes `-rtos FreeRTOS` work (in addition to `-rtos
auto`).

Change-Id: Ieafe703a2929a0959fcc81997096e8beeeb42496
commit 05e962e0 (2021-04-13): Don't check target_address_bits() until we need it. (#593)

Fixes #591, and also makes `-rtos FreeRTOS` work (in addition to `-rtos
auto`).

Change-Id: Ieafe703a2929a0959fcc81997096e8beeeb42496
closed 2021-04-13 by timsifive