riscv-collab/riscv-openocd #596: cmd error when writing to the DPC

duaneellissd, 2021-04-18:
what I'm not sure of - is who/where to go with this for help/guidence. 

I would consider this place the "the head/master" of all riscv development :-) so I think I should start here.

### Problem statement

Hi - I'm using the MicroSemi polarfire (non-SOC, I have their polarfire devkit) the version of OpenOCD comes with Soft console v6.5 and I seem be able to cause a "stuck busy" then a cmderr when openocd is writing to what it says is the DPC

I can't get past this and need help to understand what is going wrong

### Versions

Using Windows 10,  OpenOCD version I have, says this at startup:
>> xPack OpenOCD (Microchip SoftConsole build), x86_64 Open On-Chip Debugger 0.10.0+dev-00859-g95a8cd9b5-dirty (2020-10-19-17:30)

### I have an A/B test - one works, one does not trying to narrow down the problem

I have 2 apps 

- I have app(A) - that I can load onto the target, seems to work all the time
- I have app(B) - that I can't seem to load, it hangs (as described below) 
- All debug settings are identical, same start up scripts etc.

App (B) is new, and I know its going to have bugs ... Thats why I want to set breakpoints etc and debug HOWEVER....  I have a chicken and egg problem.  I can't load the code because OpenOCD & GDB lockup and don't function correctly.

### What I have done

I turn on openocd debug logs,  then examined the openocd log file...

Starting with: the "accepting gdb" connection log message (in both cases) I can see what openocd and GDB are doing.

For example - in both cases GDB queries for the register list xml and seems to do things identically between the two apps.

the relevent point where things fail is shown below

I first thought .. ok maybe the debugger is writing to a wacky place (causing an exception) or something like that but... 

However, if I examine the logs starting from the GDB connection forward, I don't see any memory access operations, the first "X" packet occurs much later.  I've also tried power cycling the board (a few times, in case stale code was present)

### THIS IS THE GOOD CASE
```
Debug: 89045 832910 riscv.c:1974 riscv_openocd_poll(): polling all harts
Debug: 89046 832910 riscv.c:3010 riscv_set_current_hartid(): setting hartid to 0, was 0
Debug: 89047 832910 riscv.c:1926 riscv_poll_hart(): polling hart 0, target->state=2
Debug: 89048 832910 riscv-013.c:398 scan(): 41b r 00000000 @11 -> + 00000000 @00; 3i
Debug: 89049 832910 riscv-013.c:398 scan(): 41b - 00000000 @11 -> + 00030382 @11; 3i
Debug: 89050 832910 riscv-013.c:408 scan():  ->  allresumeack anyresumeack allhalted anyhalted authenticated version=2
Debug: 89051 832910 command.c:143 script_debug(): command - fpga.miv fpga.miv invoke-event reset-init
Debug: 89052 832910 target.c:4597 target_handle_event(): target(0): fpga.miv (riscv) event: 13 (reset-init) action: init_regs
Debug: 89053 832910 command.c:143 script_debug(): command - reg reg pc 0
Debug: 89055 832910 gdb_server.c:401 gdb_put_packet_inner(): sending packet '$O#4f'
Debug: 89056 832910 target.c:2873 handle_reg_command(): -
Debug: 89057 832910 riscv.c:3559 register_set(): [0]{0} write 0x00000000 to pc (valid=0)
Debug: 89058 832910 riscv.c:3132 riscv_set_register_on_hart(): {0} pc <- 0
Debug: 89059 832910 riscv-013.c:3689 riscv013_set_register(): [0] writing 0x0 to register pc on hart 0
Debug: 89060 832910 riscv.c:3010 riscv_set_current_hartid(): setting hartid to 0, was 0
Debug: 89061 832910 riscv-013.c:3697 riscv013_set_register(): [0] writing PC to DPC: 0x0
Debug: 89062 832910 riscv-013.c:1298 register_write_direct(): {0} dpc <- 0x0
Debug: 89063 832910 riscv-013.c:398 scan(): 41b r 00000000 @11 -> + 00000000 @00; 3i
Debug: 89064 832910 riscv-013.c:398 scan(): 41b - 00000000 @11 -> + 00030382 @11; 3i
Debug: 89065 832910 riscv-013.c:408 scan():  ->  allresumeack anyresumeack allhalted anyhalted authenticated version=2
Debug: 89066 832910 riscv-013.c:779 execute_abstract_command(): command=0x221008; access register, size=32, postexec=0, transfer=1, write=0, regno=0x1008
Debug: 89067 832910 riscv-013.c:398 scan(): 41b w 00221008 @17 -> + 00000000 @00; 3i
Debug: 89068 832910 riscv-013.c:398 scan(): 41b r 00000000 @16 -> + 00231008 @17; 3i
Debug: 89069 832910 riscv-013.c:398 scan(): 41b - 00000000 @16 -> + 0e000001 @16; 3i
Debug: 89070 832910 riscv-013.c:408 scan():  ->  progbufsize=14 datacount=1
Debug: 89071 832910 riscv-013.c:398 scan(): 41b r 00000000 @04 -> + 00000000 @00; 3i
Debug: 89072 832910 riscv-013.c:398 scan(): 41b - 00000000 @04 -> + 00000000 @04; 3i
Debug: 89073 832910 riscv-013.c:1491 register_read_direct(): {0} s0 = 0x0
Debug: 89074 832910 riscv-013.c:1298 register_write_direct(): {0} s0 <- 0x0
[... 53 lines cut ...]
I doubt making the 2 second timeout is going to solve this. Something else seems to be wrong, this is not an operation that I would expect to take 2 seconds on a 50mhz cpu. Something else is obviously wrong.

```
Debug: 148737 895630 riscv-013.c:408 scan():  ->  progbufsize=14 busy datacount=1
Debug: 148738 895630 riscv-013.c:398 scan(): 41b r 00000000 @16 -> + 00000000 @00; 3i
Debug: 148739 895645 riscv-013.c:398 scan(): 41b - 00000000 @16 -> + 0e001001 @16; 3i
Debug: 148740 895645 riscv-013.c:408 scan():  ->  progbufsize=14 busy datacount=1
Error: 148741 895645 riscv-013.c:764 wait_for_idle(): Timed out after 2s waiting for busy to go low (abstractcs=0xe001001). Increase the timeout with riscv set_command_timeout_sec.
Debug: 148742 895645 gdb_server.c:401 gdb_put_packet_inner(): sending packet '$O#04'
Debug: 148743 895645 riscv-013.c:802 execute_abstract_command(): command 0x221008 failed; abstractcs=0xe001001
Debug: 148744 895645 riscv-013.c:398 scan(): 41b w 00000700 @16 -> + 00000000 @00; 3i
Debug: 148745 895645 riscv-013.c:408 scan():  cmderr=7 -> 
Debug: 148746 895645 riscv-013.c:398 scan(): 41b - 00000000 @16 -> + 0e001001 @16; 3i
Debug: 148747 895645 riscv-013.c:408 scan():  ->  progbufsize=14 busy datacount=1
Debug: 148748 895645 riscv-013.c:779 execute_abstract_command(): command=0x221008; access register, size=32, postexec=0, transfer=1, write=0, regno=0x1008
Debug: 148749 895645 riscv-013.c:398 scan(): 41b w 00221008 @17 -> + 00000000 @00; 3i
Debug: 148750 895645 riscv-013.c:398 scan(): 41b r 00000000 @16 -> + 00221008 @17; 3i
Debug: 148751 895645 riscv-013.c:398 scan(): 41b - 00000000 @16 -> + 0e001101 @16; 3i
Debug: 148752 895645 riscv-013.c:408 scan():  ->  progbufsize=14 busy cmderr=1 datacount=1
```
From that point - forward - nothing works, everything has cascaded into a giant train wreck.

NOTE: I would suggest that the POLLING loop that tests if something is busy - SHOULD die/exit if there is a cmderr

Suggestions would be helpful. Otherwise I'm sort of stuck.

timsifive, 2021-04-19:
Congratulations on writing the best github ticket! It contains everything somebody might want to know, with some helpful analysis to boot.

Just to narrow things down more, the actual operation that's failing is the abstract command register read of s0 (which OpenOCD does as part of writing to DPC because it needs a scratch register). I don't see anything that OpenOCD is doing wrong here. Did you try reporting this to MicroSemi? @TommyMurphyTM1234 do you know anything about this?

TommyMurphyTM1234, 2021-04-19:
Hi @timsifive - not offhand.

@duaneellissd - I presume that you're using a soft RV32 core in the PolarFire fabric?
What is the provenance or nature of your (Libero SoC) design?
Can you share it?

Cheers
Tommy

TommyMurphyTM1234, 2021-04-19:
Also @duaneellissd, can you clarify which PolarFire FPGA board that you are using here please? Thanks.

https://www.microsemi.com/product-directory/fpgas/3854-polarfire-fpgas#resources

duaneellissd, 2021-04-19:
the code is from this:

https://github.com/RISCV-on-Microsemi-FPGA/MPF300T-PolarFire-Eval-Kit/tree/master/Modify_The_FPGA_Design

I just followed the instructions... I'm no verilog/vhdl guy....  Just software... who can dabble.

Using Libero v12.6, help about says:  12.900.20.24

duaneellissd, 2021-04-19:
Why did I give such a good report?  I was at one time one of the maintainers of openocd so I know what you are looking for.

A bit more background 

I have 2 software apps - (GOOD) and (BAD)  the only difference is the APP that I am loading.

That's why I went through the debug log - carefully and all of this happens before GDB starts loading the app.  It is happening during the  attach/reset/halt stage.

I'm also trying to figure out if the problem is (A) caused by my code(the bad code) or (B) openocd or (C) the HDL... it may well be there are layers of problems I'm figuring out.

Also It seems that - (A) my code can get the FPGA so wrapped up/crashed that ... the GOOD code will not load which makes it really confusing...

As for sharables :

I have 2 APPS (GOOD) and (BAD)   I can't really share the two apps.

My luck is this: If I create APP(third-sharable) it will work.... Murphys law!

I have 2 HDL images (HDL_1 internal board cannot share) and (HDL_2 board DEV KIT, as above, probably can share this, it is the microsemi demo)

The HDL-1 (internal board) is for a 500 part, 
the devboard is a 300 part 
So I cannot load (HDL) onto other boards...
I'm also not a VHDL/VERILOG guy...

That said - the basic memory map is the same , "ram at 0x8000.0000" so - this much should be the same:  attach, reset, load should be the same ... I should be able to single step through the startup code, or through the first few things in main() - then all hell would break loose because the peripherals are different.

But I can't get to that point - I cannot attach and load.

Reporting this to Microsemi, and getting/communicating a good clear understanding of where the problem is is very non-trivial due to the complexity and their reporting system is non-optimal

TommyMurphyTM1234, 2021-04-20:
> I have 2 software apps - (GOOD) and (BAD) the only difference is the APP that I am loading.

The app that you load from the debugger should have no impact on the debug startup behaviour. I presume that you're debugging from SoftConsole and using a debug launch configuration with the recommended settings (i.e. modelled on one of the existing examples)? In particular do you have the initial reset init enabled to ensure that the debugger is putting the target into a known good/clean debug state before doing anything?

I need to check if I have a PolarFire eval kit board here (working from home) but if I do I'll have a look in more detail later and post back.

Cheers
Tommy

duaneellissd, 2021-04-20:
1) debug from softconsole 6.5, and the open ocd scripts that come with it.

2) using “remote” to localhoat so i can get to the telnet window and turn on logging before gdb connects

3)  this “recommended settings” is documented where? are there differences between the 300 vrs 500 part that effect gdb/openocd? i would be surprised

4) existing examples? Please point me to one i will use that

5) i am copying the launch config from (GOOD) to (BAD) and using windiff to compare update the path and project name otherwise the files are identical

6) could this be a timing closure / constraints problem in vhdl? i don’t know where to look, or what to look for, the hdl build takes about 5 minutes on my laptop and i would expect this to be properly constrained by the automated TCL scripts that build / create the project

if the build took long then i would expect timing issues as the tool has to work harder to resolve timing (we have some projects that take 8hrs or more to build) this build takes 5 minutes

TommyMurphyTM1234, 2021-04-21:
Hi @duaneellissd - apologies for the delay but I was out of action yesterday.
I will post back with some info later today.

TommyMurphyTM1234, 2021-04-21:
Hi @duaneellissd 

It looks like you may be using outdated Mi-V RV32 soft core material.

Can you try this instead:

https://github.com/RISCV-on-Microsemi-FPGA/PolarFire-Eval-Kit

In particular the 12.6-v1.0 tagged release for use with Libero SoC v12.6:

https://github.com/RISCV-on-Microsemi-FPGA/PolarFire-Eval-Kit/releases

(There will be a new release tag soon for Libero SoC v2021.1).

When you have your board programmed with this design you can use the Firmware Catalog bundled with Libero SoC to generate the MIV_RV32 Hardware Abstraction Layer (HAL) v3.0.109 SysTick Timer Example project and use that.

You may need to refer to the HAL User Guide regarding the build options/configurations for the example project depending on which variant you are targeting.

Let me know if you have any issues getting this stuff going.

Hope this helps

Regards
Tommy

duaneellissd, 2021-04-22:
@TommyMurphyTM1234  - thanks for that pointer

Things act a little different, using that - I've been tied up on some other projects/fire-fights
If this works out - I'll close this ticket (eta: end of the week)

TommyMurphyTM1234, 2021-04-22:
Hi @duaneellissd 

When you run/debug the SysTick Timer Example and if you find that it ends up in a trap handler then please change the project settings to define MIV_LEGACY_RV32 in the GNU RISC-V Cross Assembler and GNU RISC-V Cross C Compiler tool configurations.

If you hit any issues and need any assistance post back when you get a chance.

Regards
Tommy

TommyMurphyTM1234, 2021-04-22:
Hi @duaneellissd 

Apologies for the confusing instructions. Ignore my previous post and just do this:

If you are using an MPF300T PolarFire Eval Kit board then use this FlashPro Express job file to program your board:

https://github.com/RISCV-on-Microsemi-FPGA/PolarFire-Eval-Kit/blob/master/FlashPro_Express_Projects/Programming_Files/PF_Eval_Kit_MiV_RV32_CFG1_BaseDesign.job

If you are using an MPF300T_ES (Engineering Sample) PolarFire Eval Kit board then use this FlashPro Express job file to program your board:

https://github.com/RISCV-on-Microsemi-FPGA/PolarFire-Eval-Kit/blob/master/FlashPro_Express_Projects/Programming_Files/PF_Eval_Kit_ES_MiV_RV32_CFG1_BaseDesign.job

If you want to create the actual Libero design from scratch then you can follow the instructions here:

https://github.com/RISCV-on-Microsemi-FPGA/PolarFire-Eval-Kit/tree/master/Libero_Projects

using the relevant Tcl script here:

MPF300T: https://github.com/RISCV-on-Microsemi-FPGA/PolarFire-Eval-Kit/blob/master/Libero_Projects/PF_Eval_Kit_MiV_RV32_BaseDesign.tcl

MPF300T_ES: https://github.com/RISCV-on-Microsemi-FPGA/PolarFire-Eval-Kit/blob/master/Libero_Projects/PF_Eval_Kit_ES_MiV_RV32_BaseDesign.tcl

And then you can generate the MIV_RV32 Hardware Abstraction Layer (HAL) v3.0.109 SysTick Timer Example project from the Firmware Catalog, import it into SoftConsole, build it and debug it as is (i.e. without the need for the MIV_LEGACY_RV32 define mentioned above).

It might help to clarify that the MiV-RV32 is our current/latest configurable soft RV32 core and others (MiV-RV32IMA, MiV-RV32IMAF etc.) are older legacy cores. If at all possible you should focus on the former (MiV-RV32) and collateral related to that.

Hope that helps but if there's any remaining confusion or issues let me know.

Regards
Tommy

duaneellissd, 2021-04-22:
@TommyMurphyTM1234  - Thanks - That closes the ticket.

Wow - What a PAIN! This was not a simple thing to find, fix and get right.

I would suggest - on the microsemi end - that your startup code, and/or your example code does some type of check for the correct CPU version etc via registers like:  "misa", "march", "mimpid" etc ... 
and if it is not correct... then end in a while 1 loop.

You would stop people like me .. and others who slog through issues like this.

duaneellissd, 2021-04-22:
@timsifive  - thanks for your help with this also.

closed 2021-04-22 by duaneellissd

TommyMurphyTM1234, 2021-04-23:
Hi @duaneellissd 

Can I just confirm that you were able to make progress based on what I posted above? If so that's great. But if you have any other issues/questions in this context post back.

Regards
Tommy

duaneellissd, 2021-04-23:
yes i am porting the rest of my sw stack 

i just wish things like this had some automated error checking.

for example in the entry.S, or main.c, or other init function read CSRs and if they are not correct goto some hard fault fail