riscv-collab/riscv-openocd #597: Unsupported DTM Version

Bill94l, 2021-04-19:
Hi everyone, I'm using OpenOCD to connect the FPGA board via the (JTAG) HS2 programming cable

> Info : clock speed 1000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: riscv.cpu: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unsupported DTM version: 15
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet
> 
> Error: Unsupported DTM version: 15
Erreur de segmentation (core dumped)

**Is this a known issue?**

Thanks

JanMatCodasip, 2021-04-19:
Hi @Bill94l , 

I assume you are using a correct configuration of the JTAG-HS2 programming cable. (If unsure, you may take a look at: https://github.com/riscv/riscv-openocd/blob/riscv/tcl/interface/ftdi/digilent-hs2.cfg)

The most important message from your log is: **Error: JTAG scan chain interrogation failed: all ones**.

This means there is a hardware problem - the TDO input of the JTAG-HS2 adapter is at constant logical '1'.  This could be due to:
* problems with your FPGA design - the JTAG interface is not working as it should
* poor connection between the JTAG-HS2 adapter and the board (swapped pins, bad contacts in the connectors ....)

So please check again your hardware and the connection of the JTAG-HS2 adapter to the board. This does not appear to be a problem with OpenOCD.

TommyMurphyTM1234, 2021-04-19:
Almost certainly there is something wrong with your hardware (target, cable etc.) that is causing TDO to be stuck at 1 and basic JTAG connectivity not working correctly.

Bill94l, 2021-04-19:
Hi @JanMatCodasip  @TommyMurphyTM1234 ,
Thanks for your quick reply, the problem was in using a correct configuration of the JTAG-HS2 programming cable. (I corrected the configuration from the link you suggested:    https://github.com/riscv/riscv-openocd/blob/riscv/tcl/interface/ftdi/digilent-hs2.cfg)
Thanks a lot

closed 2021-04-19 by Bill94l