riscv-collab/riscv-openocd #601: Does not compile on RedHat 7

agrobman, 2021-05-05:
Hi,

I had to change 

```
diff --git a/src/gnulib/Makefile.am b/src/gnulib/Makefile.am
index 62accdf..33cd573 100644
--- a/src/gnulib/Makefile.am
+++ b/src/gnulib/Makefile.am
@@ -181,7 +181,7 @@ stdint.h: stdint.in.h $(top_builddir)/config.status
              -e 's/@''BITSIZEOF_WINT_T''@/$(BITSIZEOF_WINT_T)/g' \
              -e 's/@''HAVE_SIGNED_WINT_T''@/$(HAVE_SIGNED_WINT_T)/g' \
              -e 's/@''WINT_T_SUFFIX''@/$(WINT_T_SUFFIX)/g' \
-             -e 's/@''GNULIB_OVERRIDES_WINT_T''@/$(GNULIB_OVERRIDES_WINT_T)/g' \
+             -e 's/@''GNULIB_OVERRIDES_WINT_T''@/1/g' \
              < $(srcdir)/stdint.in.h; \
        } > $@-t && \
        mv $@-t $@
```

because somehow $(GNULIB_OVERRIDES_WINT_T) var was not set with  bootstrap/configure on our RH7 servers
and stdint.h created with 
# if
# if !
lines.

What's wrong?

timsifive, 2021-05-05:
I don't know. This sounds like the same problem as mentioned at https://bug-gnulib.gnu.narkive.com/ZZGizTsg/stdint-h-if-with-no-expression which suggests some solutions.

I tried to reproduce this in a Centos 7 docker image, but I gave up when Centos has an automake that's too old for OpenOCD, after `./bootstrap` failed with a bunch of macro errors.

Did you confirm you have autoconf >= 2.69 and automake >= 1.14, as required by the README?

agrobman, 2021-05-05:
Yes Tim, we used required versions, but with not out of box as correct version tools are installed privately (not in standard locations as our IT wants to keep RH distribution versions, which are pretty old). So we had to do some m4 macro collection locally and made this directory to -I option of the aclocal call

agrobman, 2021-05-05:
Oh, we have pretty old wint_t.m4 macro : # wint_t.m4 serial 5 (gettext-0.18.2)
I've found serial 7, which is much longer and it sets GNULIB_OVERRIDES_WINT_T

closed 2021-07-19 by timsifive