riscv-collab/riscv-openocd #602: `fillb` command doesn't work with HiFive1b?

bradjc, 2021-05-06:
I'm running into the following error when using the `fillb` command:

```
$ openocd -c " source [find board/sifive-hifive1-revb.cfg]; init; reset init; halt; flash fillb 0x20040000 0xff 512;  exit"
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : J-Link OB-K22-SiFive compiled May 27 2019 15:39:22
Info : Hardware version: 1.00
Info : VTarget = 3.300 V
Info : clock speed 4000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x20000913 (mfg: 0x489 (SiFive Inc), part: 0x0000, ver: 0x2)
Info : datacount=1 progbufsize=16
Info : Disabling abstract command reads from CSRs.
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101105
Info : starting gdb server for riscv.cpu.0 on 3333
Info : Listening on port 3333 for gdb connections
Info : Found flash device 'issi is25lp032' (ID 0x0016609d)
Ready for Remote Connections
Info : JTAG tap: riscv.cpu tap/device found: 0x20000913 (mfg: 0x489 (SiFive Inc), part: 0x0000, ver: 0x2)
Error: ip.txwm didn't get set.
Info : Disabling abstract command writes to CSRs.
Error: Algorithm timed out after 10001 ms.
Error: ra = 0x80000094
Error: sp = 0x80000790
Error: gp = 0x80001100
Error: tp = 0x0
Error: t0 = 0x0
Error: t1 = 0x80000000
Error: t2 = 0x80002330
Error: s0 = 0x400
Error: s1 = 0x20
Error: a0 = 0x10014000
Error: a1 = 0x800000dc
Error: a2 = 0x0
Error: a3 = 0x0
Error: a4 = 0x0
Error: a5 = 0x80002970
Error: a6 = 0x1
Error: a7 = 0x0
Error: s2 = 0x800011ac
Error: s3 = 0x1
Error: s4 = 0xffffffff
Error: s5 = 0x80000900
Error: s6 = 0x8
Error: s7 = 0x200004
Error: s8 = 0x1f
Error: s9 = 0x80002430
Error: s10 = 0x3
Error: s11 = 0x20013808
Error: t3 = 0x2001ebfc
Error: t4 = 0x2001d2fc
Error: t5 = 0x80001104
Error: t6 = 0x8
Error: pc = 0x80000060
Error: mstatus = 0x1800
Error: mepc = 0x20011bf8
Error: mcause = 0x8
Error: Failed to execute algorithm at 0x80000000: -302
Error: error writing to flash at address 0x20000000 at offset 0x00040000
```

I'm on macOS 10.14.6 (intel) and

```
$ openocd --version
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
```

which I installed via brew.

I tried installing openocd via `brew install --HEAD openocd` but that gave me an error, and I also tried building this repo but ran into too many compilation errors before I gave up. So it's entirely possible this is fixed in later versions and I will wait until I can get one of those to work, but I wanted to check in case this is a more obscure command that isn't used much.

en-sc, 2024-01-22:
Hi, @bradjc. Could you please provide a verbose OpenOCD log?
This can be done by either `-d3` flag or `debug_level 3` command.
Please, note that the log is quite large. It can be redirected to a file by `-l <filename>` or `log_output <filename>`

TommyMurphyTM1234, 2024-01-22:
FWIW I tried this against a HiFive1 Rev A01 (not Rev B) and get the same results as the original post:
```
openocd -c " source [find board/sifive-hifive1.cfg]; init; reset init; halt; flash fillb 0x20040000 0xff 512;  exit"
Open On-Chip Debugger 0.12.0+dev-03592-g4fc0d86ff-dirty (2024-01-17-15:20)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x10e31913 (mfg: 0x489 (SiFive Inc), part: 0x0e31, ver: 0x1)
Warn : Got exception 0xffffffff when reading tinfo
Warn : Got exception 0xffffffff when reading tinfo
Info : [riscv.cpu] Found 2 triggers
halted at 0x400 due to software breakpoint
Info : Examined RISCV core; XLEN=32, misa=0x40001105
[riscv.cpu] Target successfully examined.
Info : [riscv.cpu] Examination succeed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : Found flash device 'issi is25lp128d' (ID 0x0018609d)
Info : JTAG tap: riscv.cpu tap/device found: 0x10e31913 (mfg: 0x489 (SiFive Inc), part: 0x0e31, ver: 0x1)
halted at 0x400 due to software breakpoint
Info : Retrying memory read starting from 0x80000000 with more delays
Error: [riscv.cpu] Algorithm timed out after 1027 ms.
halted at 0x204000b4 due to debug interrupt
Error: [riscv.cpu] ra = 0x20400102
Error: [riscv.cpu] sp = 0x68ccc1f8
Error: [riscv.cpu] gp = 0x8000096c
Error: [riscv.cpu] tp = 0x0
Error: [riscv.cpu] t0 = 0x80000370
Error: [riscv.cpu] t1 = 0x80000370
Error: [riscv.cpu] t2 = 0x204000b4
Error: [riscv.cpu] fp = 0x80000028
Error: [riscv.cpu] s1 = 0x0
Error: [riscv.cpu] a0 = 0x1
Error: [riscv.cpu] a1 = 0xe7f7b2
Error: [riscv.cpu] a2 = 0x80000028
Error: [riscv.cpu] a3 = 0x800000d8
Error: [riscv.cpu] a4 = 0x8000016c
Error: [riscv.cpu] a5 = 0xe7f7b3
Error: [riscv.cpu] a6 = 0x200c000
Error: [riscv.cpu] a7 = 0x7ffffffb
Error: [riscv.cpu] s2 = 0x0
Error: [riscv.cpu] s3 = 0x0
Error: [riscv.cpu] s4 = 0x0
Error: [riscv.cpu] s5 = 0x0
Error: [riscv.cpu] s6 = 0x0
Error: [riscv.cpu] s7 = 0x0
Error: [riscv.cpu] s8 = 0x0
Error: [riscv.cpu] s9 = 0x0
Error: [riscv.cpu] s10 = 0x0
Error: [riscv.cpu] s11 = 0x0
Error: [riscv.cpu] t3 = 0x0
Error: [riscv.cpu] t4 = 0x0
Error: [riscv.cpu] t5 = 0x0
Error: [riscv.cpu] t6 = 0x0
Error: [riscv.cpu] pc = 0x204000b4
Error: [riscv.cpu] mstatus = 0x1800
Error: [riscv.cpu] mepc = 0x204000b6
Error: [riscv.cpu] mcause = 0x7
Error: Failed to execute algorithm at 0x80000000: -302
Error: error writing to flash at address 0x20000000 at offset 0x00040000
```
Here is the zipped verbose log: [openocd.zip](https://github.com/riscv/riscv-openocd/files/14015504/openocd.zip)

I haven't looked at it in detail yet.

TommyMurphyTM1234, 2024-01-28:
Isn't the address `0x2004_0000` used in the original post incorrect?
Shouldn't it be `0x2040_0000`?
See here for example:

* https://forums.sifive.com/t/bootloader-restore/2429/2

However, even if I use `0x2040_0000` I still get the same behaviour so there's still some issue even if the correct (?) flash address is used:
```
openocd -c " source [find board/sifive-hifive1.cfg]; init; reset init; halt; flash fillb 0x20400000 0xff 512;  exit"
Open On-Chip Debugger 0.12.0+dev-03592-g4fc0d86ff-dirty (2024-01-17-15:20)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x10e31913 (mfg: 0x489 (SiFive Inc), part: 0x0e31, ver: 0x1)
Warn : Got exception 0xffffffff when reading tinfo
Warn : Got exception 0xffffffff when reading tinfo
Info : [riscv.cpu] Found 2 triggers
halted at 0x204000b4 due to debug interrupt
Info : Examined RISCV core; XLEN=32, misa=0x40001105
[riscv.cpu] Target successfully examined.
Info : [riscv.cpu] Examination succeed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : Found flash device 'issi is25lp128d' (ID 0x0018609d)
Info : JTAG tap: riscv.cpu tap/device found: 0x10e31913 (mfg: 0x489 (SiFive Inc), part: 0x0e31, ver: 0x1)
halted at 0x400 due to software breakpoint
Info : Retrying memory read starting from 0x80000000 with more delays
Error: [riscv.cpu] Algorithm timed out after 1027 ms.
halted at 0x204000b4 due to debug interrupt
Error: [riscv.cpu] ra = 0x20400102
Error: [riscv.cpu] sp = 0x254d39f8
Error: [riscv.cpu] gp = 0x8000096c
Error: [riscv.cpu] tp = 0x0
Error: [riscv.cpu] t0 = 0x80000370
Error: [riscv.cpu] t1 = 0x80000370
Error: [riscv.cpu] t2 = 0x204000b4
Error: [riscv.cpu] fp = 0x4000f063
Error: [riscv.cpu] s1 = 0x0
Error: [riscv.cpu] a0 = 0x10014000
Error: [riscv.cpu] a1 = 0x100
Error: [riscv.cpu] a2 = 0x8000031c
Error: [riscv.cpu] a3 = 0x400000
Error: [riscv.cpu] a4 = 0x200
Error: [riscv.cpu] a5 = 0x2
Error: [riscv.cpu] a6 = 0x200c000
Error: [riscv.cpu] a7 = 0x7fffeffb
Error: [riscv.cpu] s2 = 0x0
Error: [riscv.cpu] s3 = 0x0
Error: [riscv.cpu] s4 = 0x0
Error: [riscv.cpu] s5 = 0x0
Error: [riscv.cpu] s6 = 0x0
Error: [riscv.cpu] s7 = 0x0
Error: [riscv.cpu] s8 = 0x0
Error: [riscv.cpu] s9 = 0x0
Error: [riscv.cpu] s10 = 0x0
Error: [riscv.cpu] s11 = 0x0
Error: [riscv.cpu] t3 = 0x0
Error: [riscv.cpu] t4 = 0x0
Error: [riscv.cpu] t5 = 0x0
Error: [riscv.cpu] t6 = 0x0
Error: [riscv.cpu] pc = 0x204000b4
Error: [riscv.cpu] mstatus = 0x1800
Error: [riscv.cpu] mepc = 0x204000b6
Error: [riscv.cpu] mcause = 0x7
Error: Failed to execute algorithm at 0x80000000: -302
Error: error writing to flash at address 0x20000000 at offset 0x00400000
```