riscv-collab/riscv-openocd #610: How to quit GDB without interrupt running program

Kendidi, 2021-05-19:
Hi all,

I use OpenOCD and GDB to load and run application on RISCV32 core-0.  How can I let the application continue to run, and quit/exit GDB and then disconnected OpenOCD?

The reason is after I start running application on core 0, I want to connect to core 1 and do the same.  Please advise!  Thanks a lot in advance!

timsifive, 2021-05-20:
I'd suggest leaving gdb running, and simply connect a second one. (Note that for that you don't want to use `-rtos auto` so that each hart shows up as its own core, and OpenOCD will listen on a second port for a second gdb.)

Kendidi, 2021-05-20:
Thanks Tim for your advice! I will try. 

Best Regards
Kenneth Wong

closed 2021-07-19 by timsifive