riscv-collab/riscv-openocd #623: How to use OpenOCD to control the "SB" module in the DM to access the bus when there is no CPU?

zhuazizhua, 2021-06-11:
Recently I was accessing the bus through the "sb"(Located in the dm 0x38-0x3f) module of the DM, but I ran into a problem."target create *.cpux riscv -chain-position * -coreid 0xX" cannot be set in *.cfg without CPU, so that "riscv set_prefer_sba on" cannot be set.So I have four questions that I need help with:
1. How to set "riscv set_prefer_sba on" when "target create *.cpux riscv-chain-position * -coreid 0xX" cannot be set?
2. Is there a way in OpenOCD to force access to memery using "sb"?
3. How do you force "sb" to access memery in "riscv set_prefer_sba on"?
4. I read "OpenOCD User's Guide", but I still can't understand the meaning of "Debug: 292 96 riscv-013.c:393 scan(): 41b 0i - 00000000 @10 -> b 00000000 @10".
Can someone help me? Thank you very ### much!!

JanMatCodasip, 2021-06-14:
Hi @zhuazizhua,

>     1. How to set "riscv set_prefer_sba on" when "target create *.cpux riscv-chain-position * -coreid 0xX" cannot be set?

It is not possible to access target's memory with OpenOCD without having the target defined in your cfg file. In other words, `target create ....` is mandatory.

>     2. Is there a way in OpenOCD to force access to memery using "sb"?
>     3. How do you force "sb" to access memery in "riscv set_prefer_sba on"?

You can use `riscv set_mem_access sysbus` to force the use of sysbus (and ignore all other methods). But still, a target has to be defined (`target create ....`).

>     4. I read "OpenOCD User's Guide", but I still can't understand the meaning of `Debug: 292 96 riscv-013.c:393 scan(): 41b 0i - 00000000 @10 -> b 00000000 @10`.

These low-level debug prints show the accesses to the debug module's registers. To my knowledge, they are not documented.
To understand the prints:
* familiarize yourself with the chapter 6 of the [RISC-V debug spec](https://github.com/riscv/riscv-debug-spec/blob/master/riscv-debug-stable.pdf) - DTM, Debug Transport Module
* then look directly at the OpenOCD source to find out what each of the numbers mean: https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L399

zhuazizhua, 2021-06-15:
Thank you very much. It seems that OpenOCD currently accesses the bus through the "SB" module of the DM without CPU, because there are two problems with this without CPU:.
1. The premise of "riscv set_prefer_sba on" is "target create *.cpux riscv ".
2. The premise of "target create *.cpux riscv" is "jtag newtap cpu"(I don't know if "JTAG newtap unknowN" is OK?).
I have now solved this problem in another way, with the Laterbach debugger, which has a "prepare" mode.Hopefully oepnocd can provide a similar solution.

closed 2021-06-16 by zhuazizhua