riscv-collab/riscv-openocd #628: stepping over compressed instructions

duaneellissd, 2021-07-09:
Hi - Internally (I'm reporting for a team member) - we have an issue when in GDB, we are trying to *step*over* compressed instructions.

The sequence is (disassembly) causing problems is:

```
800011d2 <main>:
800011d2:  1101                  addi sp,sp,-32
800011d4:  ce06                  sw   ra,28(sp)
800011d6:  cc22                  sw   s0,24(sp)
800011d8:  1000                  addi s0,sp,32
800011da:  64018223              sb   zero,1604(gp) # 8000bc64 <sendTest>
800011de:  67f1                  lui  a5,0x1c
800011e0:  20078593              addi a1,a5,512 # 1c200 <RAM_SIZE+0xc200>
800011e4:  7e000537              lui  a0,0x7e000
800011e8:  2411                  jal  800013ec <DEBUG_por_init>
```
When stepping over C lines of code - GDB gets very lost

My guy describes the problem like this:

--------------------------
Here’s where the debugger goofs… after executing the lui (actually, c.lui) instruction, the PC should be $800011e0, but the debugger has it set to $800011e2.  If I step another instruction, I land at trap_entry with mcause set to 2, “illegal instruction”.  The mtval register contains half of the addi instruction, and half of the succeeding lui instruction (which is a full 4-byte lui, not another compressed one.)

I’m running into this in other places.  Trying to step through parts of the serial power-on reset code to see if it’s correctly setting up the IRQs, it happens even if I’m stepping through lines of C code, not lines of disassembly.
--------------------------

I can get more detail but it's not easy.. I'm not on that project so its a small game of "broken telephone" and lost in translation.

This is using the MicroSemi  Soft Console 2021.1 release.

This problem could be: (A) in OpenOCD, or (B) In GDB, or (C) Something else -if more details are needed, and  if we had a better understanding of where to go next, we can probably get some more details

our next step is to disable the Compressed Instructions... 

Thanks

TommyMurphyTM1234, 2021-07-10:
Hi @duaneellissd 

I work for Microchip (Microsemi) on SoftConsole and can look into this for you if you can provide the code which you are trying to debug and any relevant instructions on how you are doing this. 

I don't think that here is the appropriate place for this ticket so maybe you can contact me directly and we can take this offline? Obviously if it does turn out to the an OpenOCD issue it can be revived here.

I am not aware of any fundamental problem in OpenOCD or SoftConsole with regard to debugging code containing compressed instructions but obviously the first step in investigating your problem is to reproduce it.

Regards
Tommy

TommyMurphyTM1234, 2021-10-04:
Can this issue be closed @duaneellissd?

TommyMurphyTM1234, 2023-03-16:
Candidate for closure?

closed 2023-03-17 by timsifive