riscv-collab/riscv-openocd #637: Unable to load programs with progbufsize =1 and impebreak = 1.

stanleyleee, 2021-08-09:
The RISCV_MEM_ACCESS_PROGBUF method requires progbufsize + impebreak >= 3.  
        
For a implementation supports "PROGBUF with progbufsize =1, impebreak = 1",  and does not support "SYSBUS_ACCESS", "ABSTRACT_MEMORY_ACCESS", there are no available method in write_memory function at riscv-013.c.
           
Would it be a optional methods to fill PROGBUF with "store s1, 0(s0)" only and always fall into branch of if(setup_needed){ ... } to enable write_memory_progbuf function for the implementation with progbufsize =1, impebreak = 1 ?

timsifive, 2021-08-09:
I'd happily review a PR that implements read/write memory using progbuf when progbufsize=1.

Generally I really encourage hardware implementations to have progbufsize of 2. With a size of 1, memory reads/writes will be at least 4 times slower than with a size of 2. (But maybe if your target only has 4KB of RAM that doesn't matter.)

timsifive, 2023-03-17:
Fair enough. There could be such an implementation, and if there was then it would be reasonable to have an issue open here, but I still wouldn't have time to work on it.

closed 2023-03-17 by timsifive