riscv-collab/riscv-openocd #641: command 'opneocd' not found

ghita92, 2021-08-31:
Hello,
I'm new to riscv, i would like to know what are the paths envirements for openocd.
I installed openocd without any options in the configuration and the command openocd is not recognized.
Thanks

TommyMurphyTM1234, 2021-08-31:
This is not really a RISC-V issue.
But you need to explain exactly what you did.
You provide no useful info above.

If you are building OpenOCD yourself then git clone, bootstrap, configure and make install should do what you need (at least on GNU/Linux).
And then you just have to ensure that the OpenOCD install bin directory is on your path.

ghita92, 2021-08-31:
I did install using 
sudo ./bootstrap
sudo ./configure
make
but when then i cannot find openocd.
I have this error Command 'openocd' not found, but can be installed with:

TommyMurphyTM1234, 2021-09-01:
You didn't do make install to install it to the default location.

TommyMurphyTM1234, 2021-09-01:
> I did install using
> sudo ./bootstrap
> sudo ./configure
> make

You didn't do make install?

> but when then i cannot find openocd.
> I have this error Command 'openocd' not found, but can be installed with:
> 
> My $path variable is the riskv install path do i set it multiple Time ?
> Export path=path to RISCv install$path
> Export path=path to openocd install$path ?

I don't really understand any of this.

ghita92, 2021-09-01:
Thanks, i manage to solve the problem by adding the bin path to the $PATH variable.

closed 2021-09-02 by JanMatCodasip