riscv-collab/riscv-openocd #643: openocd timeout occurs when trying to load elf file using command 'load_image'.

nikhill-agnisys, 2021-09-01:
Hello all,

I am trying to debug SweRV RISCV RTL v1.9 during simulation using bitbang bridge and openocd in my environment. For this, I have instantiated SimJTAG module in my testbench and connected it with swerv core. Now, using bitbang a server socket is created and I am able to listen on port 9999.

[image]

In another terminal (terminal 2) , I launched openocd (vOpen On-Chip Debugger 0.11.0+dev-01861-g6edf98db7-dirty) using my 'sim_swervjtag.cfg' file, the contents of config file are shown below,

`
adapter speed 100

adapter driver remote_bitbang
remote_bitbang_host localhost

remote_bitbang_port 9999

set _CHIPNAME riscv

jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x01

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME

riscv set_enable_virt2phys off

riscv set_mem_access abstract

riscv expose_csrs 1988

proc swerv_eh1_execute_fence {} {
reg csr1988 0x3
}

$_TARGETNAME configure -event resume-start {
swerv_eh1_execute_fence
}

$_TARGETNAME configure -event step-start {
swerv_eh1_execute_fence
}

init

halt
`
On another terminal (terminal 3), When I connect using telnet localhost 4444 and try to issue load_image command after reset I get timeout error as below,

[image]

On terminal 2, it outputs as shown without -d debug switch,

[image]

With debug switch -d, please find attached the log file 'openocd.log',

[openocd.log](https://github.com/riscv/riscv-openocd/files/7089354/openocd.log)

I have tried increasing the timeout interval using 'riscv set_command_timeout_sec' to 2000 but then it never comes out.

If possible, please point me to any pointer on how to resolve this. Thanks.

TommyMurphyTM1234, 2021-09-01:
Your image seems to be getting loaded at address 0x00000000. Is that correct for your target? Is there read/write RAM at that location?

`Debug: 1118 19714 configuration.c:99 find_file(): found B1_test.elf
Debug: 1119 19714 image.c:76 autodetect_image_type(): ELF image detected.
Debug: 1120 19714 configuration.c:99 find_file(): found B1_test.elf
Debug: 1121 19714 image.c:641 image_elf_read_headers(): ELF32 image detected.
Debug: 1122 19714 image.c:670 image_elf32_read_section(): load segment 0 at 0x0 (sz = 0x4e8)
Debug: 1123 19714 image.c:677 image_elf32_read_section(): read elf: size = 0x4e8 at 0x1000
Debug: 1124 19714 target.c:2393 target_write_buffer(): writing buffer of 1256 byte at 0x00000000`

nikhill-agnisys, 2021-09-01:
Okay thanks for the response @TommyMurphyTM1234. Can you please provide me some pointers on how to update the target address where the image gets loaded via jtag.

TommyMurphyTM1234, 2021-09-01:
Read the OpenOCD user's guide about how to use the load_image command?

nikhill-agnisys, 2021-09-01:
I have updated my target memory offset to 0x20000000 which is the location of SRAM memory on the SweRV EH1.9 core, but the issue still persist. Even if I directly use the openocd command like 'mww' the same issue is reflected. 

[image]

Looks like, some busy signal is not getting low, even it I increase timeout using 'riscv set_command_timeout_sec', the busy signal remains high throughout. The debug file shows that the process stuck at 'abstractcs=0x1002', attached is the short snippet for reference,

[image]

Are there any ways to debug this issue or to get any relevant information that could help in the process.

TommyMurphyTM1234, 2021-09-01:
Can you post a -d verbose log?
It might be interesting to see exactly why the write to 0x20000000 is failing.
Are you sure that RAM is properly implemented at this address?
The fact that target examination is presumably working and reset init does not trigger any error suggests that basic debug access is working even if RAM access seems to be failing,
Do any other openocd commands work?
E.g. read/write registers, read RAM etc.?

timsifive, 2021-09-01:
Simulated targets are really slow. Try increasing the timeout as the error message you get suggests.

TommyMurphyTM1234, 2021-09-01:
> Simulated targets are really slow. Try increasing the timeout as the error message you get suggests.

According to the first post in this thread even 2000 seconds didn't help so I don't think that that's the problem?

TommyMurphyTM1234, 2021-09-01:
> How big is the program? 

Even a simple mww 0x20000000 10 seems to fail/timeout.

nikhill-agnisys, 2021-09-13:
Hello everyone,

It was really helpful to have your inputs, thank you all for the wonderful contributions. After spending some additional time in target simulation I was able to find why my abstract commands were failing. My SB was indeed tied of, causing debug state to remain as 'SB_CMD_RESP' for rest of the simulation. 

[image]

Adding crossbar to route the transactions worked for me. I am able to fix this issue now.

closed 2021-09-13 by nikhill-agnisys