riscv-collab/riscv-openocd #646: System reset is not tirggered during the OpenOCD initialization routine

Hzv587, 2021-09-24:
Dear community,

I have a debug setup consists of J-Link and Xilinx FPGA. It occurs to me that the system reset is not triggered during the OpenOCD initialization routine as shown in the Figure
[image]

However, the "monitor reset" in GDB terminal works as expected, i.e., both system and test reset are triggered.

TommyMurphyTM1234, 2021-09-24:
What openocd script are you using?

Hzv587, 2021-09-27:
Hi, I tested with both riscv-openocd-0.10.0-2020.04.6-x86_64-w64-mingw32 and riscv-openocd-0.10.0-2020.12.1-x86_64-w64-mingw32 downed from SiFive. Following settings are used in the openocd configuration script:

interface jlink
transport select jtag
adapter_khz 1000
set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 8
set _TARGETNAME $_CHIPNAME.cpu
reset_config trst_and_srst srst_pulls_trst
target create $_TARGETNAME riscv -chain-position $_TARGETNAME
gdb_report_data_abort enable
gdb_breakpoint_override hard
init
halt

TommyMurphyTM1234, 2021-09-27:
Can you attach your full verbose (-d3) OpenOCD log please?

Hzv587, 2021-09-27:
Please see the log file attached.
[openocd.log](https://github.com/riscv/riscv-openocd/files/7235720/openocd.log)

TommyMurphyTM1234, 2021-09-27:
Doesn't this indicate that the resets have been asserted (active low) and then deasserted/released correctly?

```
Debug: 300 90 jlink.c:939 jlink_reset(): TRST: 0, SRST: 0.
Debug: 305 90 core.c:729 jtag_add_reset(): SRST line released
Debug: 306 90 core.c:753 jtag_add_reset(): TRST line released

```

Hzv587, 2021-09-27:
From the log file, yes, the openOCD shall do this. 
But the logic analyzer detects no transitions on the system reset signal in the FPGA. However, the transitions are detected for the system reset signal when I gave the "monitor reset" command in the GDB terminal.

Hzv587, 2021-09-28:
Do you mean to replace "init" in the configuration script to "reset init"?
After doing this, I got an error message in the openOCD log, saying
`User : 162 30 command.c:695 command_run_line(): .\openOCD.cfg:18: Error: invalid command name "reset"
`
Am I understanding you correctly?

TommyMurphyTM1234, 2021-09-28:
> Do you mean to replace "init" in the configuration script to "reset init"?

No. You need to figure out (maybe based on looking at other exist example scripts) what init/reset is appropriate for your specific target and debugging needs.

My point is that nothing (other than maybe the JTAG TAP) will be reset automatically by OpenOCD unless you tell it to happen and in your original example there doesn't seem to have been any reset effected.

I generally let the debugger connect and only do a `reset init` via gdb in the case where I want to debug "from scratch" rather than connecting to send debugging an already running target. I don't generally put the reset in the OpenOCD script. But different people and situations have different requirements.

TommyMurphyTM1234, 2021-10-01:
Can this issue be closed @Hzv587 ?

Hzv587, 2021-10-04:
Thanks for the information. I have adapted my OpenOCD configuration script to achieve my purpose. Basically, this issue can be closed. 

But I still want to point out the inconsistency between the log and what OpenOCD really does. That is, while the log file told me that system reset is triggered (lines 219-221 in the attached log file), the OpenOCD did not do that at all. This is the main reason why I raise this issue. It would be nice, we can fix this inconsistency and do not confuse other developers.

TommyMurphyTM1234, 2021-10-04:
> But I still want to point out the inconsistency between the log and what OpenOCD really does. That is, while the log file told me that system reset is triggered (lines 219-221 in the attached log [...]

I don't think that this is the case.
The log shows SRST/TRST being set to 0 - i.e. deasserted.
It does *not* show them being *toggled* - i.e. asserted and then deasserted.
As such I believe that you're mistaken in inferring that there's any reset being effected here.
My understanding is that a reset only happens if you tell it to happen via OpenOCD commands in the script, passed via the command line, via GDB's monitor command etc.

Hzv587, 2021-10-04:
I get your point, thanks for sharing your understanding. Let me close this issue.

closed 2021-10-04 by Hzv587