riscv-collab/riscv-openocd #663: Custom JTAG setup sequence before target examine

fborisovskii, 2021-11-18:
Hello,

After last openocd update from this repo I've found that my custom JTAG setup procedure is broken. It seems that 
after this change https://github.com/riscv/riscv-openocd/pull/654 target examine is called before JTAG setup.
Is there any way to setup JTAG TAPs and to override reset procedure after this change? Currently I've reverted this change locally.

The problem is that we have the system with two RISC-V subsystems with
individual debug module each connected in JTAG chain. Both TAPs are online after power up.
One of the subsystems is kept in reset while another one is doing system reset sequence.
ndmreset signal of only one of them is connected to the rest of the system (and to the cores of the second subsystem)

I use overridden jtag setup  procedure to make reset sequence before targets examine.

With that changes targets examine is called before jtag setup and that breaks functionality.
Do you know another mechanism to make initial reset sequence before targets examine?
The mechanism which can be applied with that changes.

I attach openocd config (sys_cfg.txt) and logs with and without that patch.

The problem as I see it is in target.c "handle_target" function.
In my case it runs examine target before jtag setup is called.

Thank you,
Fedor Borisovskii

[sys_run_with_patch.log](https://github.com/riscv/riscv-openocd/files/7564589/sys_run_with_patch.log)
[sys_run_without_patch.log](https://github.com/riscv/riscv-openocd/files/7564590/sys_run_without_patch.log
[sys_cfg.txt](https://github.com/riscv/riscv-openocd/files/7564592/sys_cfg.txt)
)

timsifive, 2021-11-18:
I'm going to look into this next week. Somehow OpenOCD is trying to halt the target before examine() is called.

If I understand your setup correctly, change 654 should remove the need for a lot of the complexity in your config script.

fborisovskii, 2021-11-18:
Many thanks,

I add a simplified picture of the system. So one RISC-V subsystem is MCU which is online after power up.
And another one is dual-hart CPU (CPU0, CPU1 are controlled by common Debug Module) which is under reset. 
So the goal is to release the second one's reset
before its examine. And the second goal is to override monitor reset procedure to use RESET Control block to 
flush the system state.

```
         ◄────────────────────────────────┐
JTAG         ┌───────┐          ┌───────┐ │
         ───►│ TAP1  ├─────────►│ TAP0  ├─┘ ┌──────── reset_vector is to SRAM
             └───┬───┘          └───┬───┘   │
                 ▼                  ▼       ▼
             ┌─────────┐     ┌────────────────┐   ┌───────┐
             │         │     │                │──►│SRAM   │
             │   MCU   │     │  CPU0  CPU1    │   └───────┘
             │         │     │                │   ┌───────┐
             └──────┬──┘     │                │──►│DRAM   │
                   MMR       └────────────────┘   └───────┘
                    ▼                ▲                 ▲     
                ┌──────────┐         │                 │     
                │ RESET    │         │                 │     
                │ Control  ├───reset─┘─────────────────┘     
                │          │
                └──────────┘
```

fborisovskii, 2021-11-30:
Thank you Tim,
I've checked change 665 and my configuration works as expected (as it worked previously).
The issue can be closed after this change is committed.

closed 2021-12-03 by timsifive

fborisovskii, 2022-05-05:
Hello Tim,
I've tried -defer-examine option in target create line. And I've got no changes in tool behavior. It seems that handle_target is called right after tap is reset:
```
Debug: 227 12 jlink.c:953 jlink_reset(): TRST: 0, SRST: 0
Debug: 232 12 core.c:824 jtag_add_reset(): SRST line released
Debug: 233 12 core.c:849 jtag_add_reset(): TRST line released
Debug: 234 12 core.c:322 jtag_call_event_callbacks(): jtag event: TAP reset
Debug: 235 12 command.c:166 script_debug(): command - jtag arp_init
Debug: 236 12 target.c:1858 target_call_event_callbacks(): target event 0 (gdb-halt) for core riscv.cpu0
Debug: 237 12 target.c:3075 handle_target(): [riscv.cpu0] Polling failed, trying to reexamine
Debug: 238 12 target.c:1858 target_call_event_callbacks(): target event 19 (examine-start) for core riscv.cpu0
Debug: 239 12 target.c:5101 target_handle_event(): target(0): riscv.cpu0 (riscv) event: 19 (examine-start) action:
```
And without guard with examine_attempted variable it calls examine-start before jtag setup event.
I 've tried even to override examine-start with empty proc and it did no changes to this either:

```
$_TARGETNAME configure -event examine-start "";
```
Also tool reported that "arp_examine" is to called after "init"
I suppose it is too late in my case.
Talking in RISC-V debug terminology I need manually control reset sequence using Debug Module (via SBA)
before target examination (Debug Module is online while Core is in reset) and reset is controlled via external MMRs module. Because I have no direct access to Core's CSRs (only via program buffer) I can't complete examination process while core is in reset.

timsifive, 2022-05-05:
Thank you for checking on that. Let me think about this some more... We may be able to revert #654, which is the cause of your trouble in the first place.

PR #665 (2021-12-03): Don't reexamine targets until it's time.
Don't reexamine targets until we're past the examine stage of init().
Earlier than that, examine() will likely fail because the scan chain
hasn't been examined yet.

This will likely fix #663.

Change-Id: I76ee9181f35cedcdb1a3e0f8ac33ab361c68d3af
commit f631c906 (2021-12-03): Don't reexamine targets until it's time.

Don't reexamine targets until we're past the examine stage of init().
Earlier than that, examine() will likely fail because the scan chain
hasn't been examined yet.

This will likely fix #663.

Change-Id: I76ee9181f35cedcdb1a3e0f8ac33ab361c68d3af