riscv-collab/riscv-openocd #667: How to add new commands to read and write custom registers in JTAG DTM TAP？

OpenEDF, 2021-12-04:
Hi，
I added a new register at address 0x19 in JTAG DTM TAP Register to help me read and write Debug Bus information when debugging, but I don’t know how to add new commands to openocd to operate it. The operation command is added and it can run successfully, but it only executes a simple print output when the command is received. I don't know how to read and write the data in this register. How can I do it? Can you tell me how to add the operation process and modify the file or function? Any answer is of great help to me.

timsifive, 2021-12-07:
What Tommy describes should work. You'll probably have to disable polling when you do that, otherwise IR will likely be changed before you get to access DR. See https://openocd.org/doc/html/Server-Configuration.html#Event-Polling

OpenEDF, 2021-12-07:
Hi @timsifive, @TommyMurphyTM1234,

thanks for your reply.

According to your answers and email from @borneoa, I can read the register data using the following command.

poll off 
irscan  $tapname 0x19 
drscan $tapname $len $data 
poll on

closed 2021-12-07 by OpenEDF

TommyMurphyTM1234, 2021-12-07:
Glad it helped @OpenEDF