riscv-collab/riscv-openocd #668: OPENOCD issue with NEXYS_VIDEO fpga

sreya-12, 2021-12-07:
Hello, I am new to riscv .I am trying to use openocd for nexys_video fpga.But the openocd is not working.having a error at 6 th line in configuration file  which i am unable to rectify
[image]
.

TommyMurphyTM1234, 2021-12-07:
Your screenshot is difficult to read and posting screenshots isn't really a good way to provide info in any case.
You would be better off posting a verbose -d OpenOCD log and details of the script(s) that you're using.

sreya-12, 2021-12-08:
**# SPDX-License-Identifier: GPL-2.0-or-later

# Digilent Nexys Video with Xilinx Artix-7 FPGA
# https://reference.digilentinc.com/programmable-logic/nexys-video/start

adapter driver ftdi
adapter speed 30000

ftdi device_desc "Digilent USB Device"
ftdi vid_pid 0x0403 0x6010

# channel 0 is dedicated for Digilent's DPTI Interface
# channel 1 is used for JTAG
ftdi channel 1

# just TCK TDI TDO TMS, no reset
ftdi layout_init 0x0088 0x008b
reset_config none

# Enable sampling on falling edge for high JTAG speeds.
ftdi tdo_sample_edge falling

transport select jtag

source [find cpld/xilinx-xc7.cfg]
source [find cpld/jtagspi.cfg]**

this is the configuration file I have used.

sreya-12, 2021-12-08:
adapter driver ftdi
adapter speed 30000

ftdi device_desc "Digilent USB Device"
ftdi vid_pid 0x0403 0x6010

# channel 0 is dedicated for Digilent's DPTI Interface
# channel 1 is used for JTAG
ftdi channel 1

# just TCK TDI TDO TMS, no reset
ftdi layout_init 0x0088 0x008b
reset_config none

# Enable sampling on falling edge for high JTAG speeds.
ftdi tdo_sample_edge falling

transport select jtag

source [find cpld/xilinx-xc7.cfg]
source [find cpld/jtagspi.cfg]**

this is the configuration file I have used.

sreya-12, 2021-12-16:
Following is my nexys_video.cfg file for openocd

# SPDX-License-Identifier: GPL-2.0-or-later

# Digilent Nexys Video with Xilinx Artix-7 FPGA
# https://reference.digilentinc.com/programmable-logic/nexys-video/start

adapter driver ftdi
adapter speed 30000

ftdi device_desc "Digilent USB Device"
ftdi vid_pid 0x0403 0x6010

# channel 0 is dedicated for Digilent's DPTI Interface
# channel 1 is used for JTAG
ftdi channel 1

# just TCK TDI TDO TMS, no reset
ftdi layout_init 0x0088 0x008b
reset_config none

# Enable sampling on falling edge for high JTAG speeds.
ftdi tdo_sample_edge falling

transport select jtag

source [find cpld/xilinx-xc7.cfg]
source [find cpld/jtagspi.cfg]

------------------
I am trying to run a baremetal program on nexys-video FPGA .I am stuck at this point.Is there any need to edit this openocd cfg file

TommyMurphyTM1234, 2021-12-16:
You didn't attach a verbose OpenOCD log generated using `-d`.

sreya-12, 2021-12-16:
Open On-Chip Debugger 0.10.0+dev-00849-gcbb15587d (2021-11-23-18:54)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 2 options.c:63 configuration_output_handler(): debug_level: 3
User : 14 2 options.c:63 configuration_output_handler(): 
Debug: 15 2 options.c:187 add_default_dirs(): bindir=/opt/rockettools/bin
Debug: 16 2 options.c:188 add_default_dirs(): pkgdatadir=/opt/rockettools/share/openocd
Debug: 17 2 options.c:189 add_default_dirs(): exepath=/opt/rockettools/bin
Debug: 18 2 options.c:190 add_default_dirs(): bin2data=../share/openocd
Debug: 19 2 configuration.c:42 add_script_search_dir(): adding /home/sreya/.openocd
Debug: 20 2 configuration.c:42 add_script_search_dir(): adding /opt/rockettools/bin/../share/openocd/site
Debug: 21 2 configuration.c:42 add_script_search_dir(): adding /opt/rockettools/bin/../share/openocd/scripts
User : 22 2 command.c:695 command_run_line(): embedded:startup.tcl:26: Error: Can't find openocd.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 26
Info : 23 2 server.c:310 add_service(): Listening on port 6666 for tcl connections
Info : 24 2 server.c:310 add_service(): Listening on port 4444 for telnet connections
Debug: 25 2 command.c:143 script_debug(): command - init init
Debug: 27 2 command.c:143 script_debug(): command - target target init
Debug: 29 2 command.c:143 script_debug(): command - target target names
Debug: 30 2 target.c:1428 handle_target_init_command(): Initializing targets...
Error: 31 2 core.c:1340 adapter_init(): Debug Adapter has to be specified, see "interface" command
Debug: 32 2 command.c:629 run_command(): Command 'init' failed with error code -101
User : 33 2 command.c:695 command_run_line(): embedded:startup.tcl:26: Error: 
in procedure 'script' 
at file "embedded:startup.tcl", line 26

TommyMurphyTM1234, 2021-12-17:
What is your command line for invoking OpenOCD?
The following suggests that you're not passing any script(s) from the command line so it falls back to openocd.cfg (in the current directory) and doesn't find it.

> embedded:startup.tcl:26: Error: Can't find openocd.cfg
in procedure 'script'
at file "embedded:startup.tcl", line 26

sreya-12, 2021-12-29:
I used the command openocd -f nexys_video.cfg
and the output is as follows

Open On-Chip Debugger 0.10.0+dev-00849-gcbb15587d (2021-11-23-18:54)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter
  adapter usb
    adapter usb location [<bus>-port[.port]...]
adapter_khz [khz]
adapter_name
adapter_nsrst_assert_width [milliseconds]
adapter_nsrst_delay [milliseconds]
interface driver_name
interface_list
reset_config [none|trst_only|srst_only|trst_and_srst]
          [srst_pulls_trst|trst_pulls_srst|combined|separate]
          [srst_gates_jtag|srst_nogate] [trst_push_pull|trst_open_drain]
          [srst_push_pull|srst_open_drain]
          [connect_deassert_srst|connect_assert_srst]
Error: invalid subcommand "driver ftdi"
nexys_video.cfg:6: Error: 
in procedure 'script' 
at file "embedded:startup.tcl", line 26
at file "nexys_video.cfg", line 6

TommyMurphyTM1234, 2021-12-30:
> Error: invalid subcommand "driver ftdi"

That's a different error to your earlier post and suggests some problem with your script that needs to be fixed according to the OpenOCD documentation.

sreya-12, 2022-01-03:
Thanks i will check on that

en-sc, 2024-01-22:
@sreya-12, is the issue still relevant? I would like to close it.

closed 2024-01-26 by en-sc