riscv-collab/riscv-openocd #680: flash dt-bl10 (risc-v) board

ntn888, 2022-02-14:
sorry to bother with a simple question :).
I've successfully connected to the above board using openocd (via jtag):

```
> sudo openocd -f ~/my_vsllink.cfg 
Open On-Chip Debugger 0.11.0-rc2
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'adapter speed' not 'adapter_khz'
jtag
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Versaloon(0x15)by Simon(compiled on Dec 21 2018)
Info : USB_TO_XXX abilities: 0x00000208:0x010001E3:0xC0000007
Info : clock speed 100 kHz
Warn : There are no enabled taps.  AUTO PROBING MIGHT NOT WORK!!
Info : JTAG tap: auto0.tap tap/device found: 0x20000c05 (mfg: 0x602 (<unknown>), part: 0x0000, ver: 0x2)
Warn : AUTO auto0.tap - use "jtag newtap auto0 tap -irlen 5 -expected-id 0x20000c05"
Warn : gdb services need one or more targets defined
```
and I have a *.bin file compiled...but im not sure how do I flash it from here, the openocd manual horrifies me like a novel. i'm strugglong to make anything out of it...how do I issue the flash command?

ntn888, 2022-02-14:
okay... actually i'm having troubles just connecting. here's what i did:

cloned the repo: https://github.com/bouffalolab/bl_mcu_sdk

built openocd with following options:
```./configure --enable-vsllink --with-sysroot```

moved the resulting binary into the above cloned directory:
```https://github.com/bouffalolab/bl_mcu_sdk/tree/master/tools/openocd```
to replace it. because my flasher* is not supported by theirs.

but i get:
```
sudo ./openocd -f my_vsllink.cfg -f tgt_602.cfg
Open On-Chip Debugger 0.11.0+dev-00562-g5ab74bde0 (2022-02-14-14:03)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
jtag
Ready for Remote Connections
Warn : `riscv set_prefer_sba` is deprecated. Please use `riscv set_mem_access` instead.
Info : Versaloon(0x15)by Simon(compiled on Dec 21 2018)
Info : USB_TO_XXX abilities: 0x00000208:0x010001E3:0xC0000007
Info : clock speed 100 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x20000c05 (mfg: 0x602 (Open HW Group), part: 0x0000, ver: 0x2)
Info : datacount=1 progbufsize=2
Info : Disabling abstract command reads from CSRs.
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40801125
Info : starting gdb server for riscv.cpu.0 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: riscv.cpu tap/device found: 0x20000c05 (mfg: 0x602 (Open HW Group), part: 0x0000, ver: 0x2)
reset-assert-pre
reset-deassert-post
Error: Fail to receive usb data.
Error: vsllink_jtag_execute failure
Error: dmi_scan failed jtag scan
Error: failed write at 0x4, status=2
Error: Fail to send usb data.
Error: vsllink_jtag_execute failure
Error: dmi_scan failed jtag scan
Error: failed write at 0x17, status=2
Error: Fail to send usb data.
Error: vsllink_jtag_execute failure
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: Could not write to register 'mstatus'
Error executing event reset-deassert-post on target riscv.cpu.0:
embedded:startup.tcl:854: Error: 
in procedure 'ocd_process_reset' 
in procedure 'ocd_process_reset_inner' called at file "embedded:startup.tcl", line 854
Error: Fail to send usb data.
Error: vsllink_jtag_execute failure
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: Fail to send usb data.
Error: vsllink_jtag_execute failure
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: timed out while waiting for target halted
TARGET: riscv.cpu.0 - Not halted
Error: Fail to send usb data.
```

*https://github.com/zoobab/versaloon

ntn888, 2022-02-14:
does the siFive openocd support the bl602 & this one doesnt?

TommyMurphyTM1234, 2022-02-14:
> does the siFive openocd support the bl602 & this one doesnt?

This *is* the SiFive fork of the OpenOCD repo!

ntn888, 2022-02-14:
thanks for your attention!
i'm referring to: https://github.com/sifive/freedom-openocd. i did notice that repo recursive pulled this one inside... but i was unable to run ```./configure``` command on that. does this incorporate the risc-v changes from that?

thanks again!

ntn888, 2022-02-14:
are you by chance aware if this supports the bl602?

ntn888, 2022-02-14:
actually yes. this link suggests that it's this repo that's being officlally used:
https://nuttx.apache.org/docs/latest/platforms/risc-v/bl602/index.html#debuging-with-openocd

maybe fault with the probe...

ntn888, 2022-02-15:
hi, i'm using another firmware dongle: CMSIS-DAPv2

could i know how to start openocd with this? i tried:
```
openocd -c "source [find interface/CMSIS-DAPv2.cfg]" 
Open On-Chip Debugger 0.11.0+dev-02228-gbdce7319b (2022-02-15-02:17)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Can't find interface/CMSIS-DAPv2.cfg
```
many thanks

ntn888, 2022-02-15:
```
openocd -c "source [find interface/cmsis-dap.cfg]"
Open On-Chip Debugger 0.11.0+dev-02228-gbdce7319b (2022-02-15-02:17)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Warn : Using CMSIS-DAPv2 interface 1 with wrong class 10 subclass 0 or protocol 0
Info : Using CMSIS-DAPv2 interface with VID:PID=0x0483:0x572a, serial=0001A0000000
Warn : could not claim interface: Resource busy
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error writing data: Input/Output Error
Error: CMSIS-DAP command CMD_INFO failed.
```

ntn888, 2022-02-15:
okay i installed ```libhidapi-dev ```, rebuilt openocd, and ran again, still no success

ntn888, 2022-02-15:
scrapped the bluepill* and switched to the openocd's ft232r method (using the ft232rl module) much simpler, and works too...

```
openocd -f ~/openocd-cfg/my_ft232r.cfg -f tgt_602.cfg 
Open On-Chip Debugger 0.11.0+dev-02230-g347395372 (2022-02-16-03:05)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
DEPRECATED! use 'ft232r restore_serial' not 'ft232r_restore_serial'
FT232R restore serial: 0x0015 (enabled)

Ready for Remote Connections
Warn : `riscv set_prefer_sba` is deprecated. Please use `riscv set_mem_access` instead.
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 3000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x20000c05 (mfg: 0x602 (Open HW Group), part: 0x0000, ver: 0x2)
Info : [riscv.cpu.0] datacount=1 progbufsize=2
Info : Disabling abstract command reads from CSRs.
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40801125
[riscv.cpu.0] Target successfully examined.
```

didn't know openocd had this handy feature under the hood!!!

closed 2022-02-15 by ntn888