riscv-collab/riscv-openocd #685: Enabling ebreaks after reset

erhankur, 2022-03-02:
I wonder which part of the code will be responsible to set `dcsr.ebreak` bits immediately after reset? 

There are 3 flags inside riscv.c 
```c
bool riscv_ebreakm = true;
bool riscv_ebreaks = true;
bool riscv_ebreaku = true;
```
They are in use only in `riscv013_on_step_or_resume()`  But what about the reset conditions which occurred while OpenOCD is running? `dcsr.ebreakX` bits will return to default state (zero) and needs to be set again. 

Is it expected to handle in `TARGET_EVENT_RESET_ASSERT` ?

JanMatCodasip, 2022-03-02:
I confirm your point is valid - dcsr.ebreakX bits are currently not handled after reset. 

One consequence of this issue is that syscalls (semihosting) won't work if the program execution is started by `reset run` TCL command.

For resets triggered by OpenOCD itself -  I'd recommend to set the ebreak bits in the `deassert_reset()` callback (https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L2382). 

As for resets triggered outside of OpenOCD - I do not know how to best handle them, or if there is a reliable enough solution.

erhankur, 2022-03-02:
> As for resets triggered outside of OpenOCD - I do not know how to best handle them, or if there is a reliable enough solution.

In this case, we are enabling ebreaks in `on_reset()`callback which I mentioned at this [PR](https://github.com/riscv/riscv-openocd/pull/683) 

This question came into my mind while I was thinking to replace it with more reliable solution.

We need to understand the core was reset, then halt the core, set `dcsr.ebreakX` bits and then resume the core again.

timsifive, 2022-03-02:
It seems reasonable to have OpenOCD always set resethaltreq, so that if the hart resets for any reason, it will halt. Then OpenOCD can set the ebreak bits and resume the hart if the user expects it to be running. resethaltreq must be cleared before exiting.

erhankur, 2022-06-06:
Hi Tim, I would like to implement this feature. 

As a summary; we need to handle 2 reset scenarios;

1- OpenOCD resets; as @JanMatCodasip suggested, it can be handled in the `deassert_reset()` callback
2- External resets; you said `resethaltreq` is a good option for this. Can you elaborate the flow in your mind?

timsifive, 2022-06-06:
I didn't think about any of the details, but essentially the debugger should make sure `resethaltreq` is always set. When an unexpected reset occurs, the target will halt. OpenOCD will notice while polling and then can set the ebreak* bits to their expected values. Then if the target is supposed to be running, OpenOCD can let it run.

biergaizi, 2024-07-19:
I was implementing semihosting for my microcontroller today, but I just couldn't get it to work despite all attempts, unless the `ebreak` instructions are running in an infinite loop. More than an hour was wasted due to this issue, before a lucky Web search brought me here.

A note for all newcomers,  `ebreak` is not enabled in OpenOCD upon reset... To use semihosting with OpenOCD, one must reset the target via `reset halt; resume` instead of a plain `reset`,