riscv-collab/riscv-openocd #695: Which RISC-V target to use for testing the support of 0.11 debug spec?

JanMatCodasip, 2022-05-02:
When making RISC-V related changes to OpenOCD code, mostly in the 0.13 part of the code, I'd like to have a way how to test that I did not break anything in the support for the RISC-V debug spec 0.11.

What target (what hardware, or even better, SW simulator) can I use for that testing?

What are the future plans with the 0.11 code - any plans for deprecation?

Thank you.

timsifive, 2022-05-02:
I'm using a HiFive1 board for this, and rarely actually run the tests. (Note that not every test passes, either.)

I haven't given any thought to deprecating 0.11 support. I suspect it's not in common usage, but maybe there are some users out there. I'm inclined to leave the code in there until we're sure nobody is using it, but also not to spend much effort on it except to fix clear bugs reported by users.

JanMatCodasip, 2022-05-03:
Thank you for replying.

Are both the original HiFive1 (now deprecated) and HiFive1 rev. B based on the 0.11 version of the debug spec?

timsifive, 2022-05-03:
HiFive1 rev B is almost certainly 0.13.

closed 2022-05-04 by JanMatCodasip