riscv-collab/riscv-openocd #715: [Q] Parallel TPIU trace on Segger J-Trace Pro?

MountaineerGit, 2022-07-31:
I'm exploring the parallel TPIU features of the Segger J-Trace Pro using the 4+1 additional trace pins.
See https://www.segger.com/products/debug-probes/j-trace/models/j-trace/
Is it feasible to get this working with win/libusb drivers via openocd (without using the Segger-dll)?
Anyone has made some try-outs?
I would assume just opening opening a suited usb-stream descriptor is probably not enough?

TommyMurphyTM1234, 2022-07-31:
This repo is for the RISC-V fork of OpenOCD where development on RISC-V support is done before being upstreamed to the main OpenOCD project.

TPIU is a feature of the Arm CoreSight debug infrastructure.

I don't really see the link between the two.
Maybe you can explain?
If this is a general OpenOCD/Arm trace question then is would be better posted to the upstream OpenOCD mailing lists or repo rather than here.

The OpenOCD user manual explains the support that OpenOCD has for Arm trace, TPIU etc. in case that helps at all. OpenOCD doesn't seem to have any support for J-Trace as opposed to J-Link.

MountaineerGit, 2022-08-01:
Hi TommyMurphyTM1234,
indeed upstream seems to be better suited for the arm-support. At present, the TPIU upstream version only supports SWO output, without using the additional trace pins. I will place the question there as well and link it.

>I don't really see the link between the two.
>Maybe you can explain?
RISCV Nexus and ARM TPIU tracing are very similar, also the connectors are sharing same pins.
Background of my project:
We have an ongoing university project with an RISCV core that shall be extended with a Nexus Trace via MIPI20 connector (see https://github.com/riscv-non-isa/tg-nexus-trace/blob/master/pdfs/RISC-V-Trace-Connectors-0-9-2.pdf) by using 4-bit parallel trace. Due to some resource shift, I do not get Nexus trace integrated before end of this year :(. To unblock my integration of the tracing feature in OpenOCD, I was looking into the TPIU of ARM (STM32Fx), since at the end I just need a way to get the data out of the J-Trace probe. Btw. J-Trace Pro is working fine as a debug-bridge for casual debugging on RISCV with OpenOCD and the winusb driver via JTAG.

TommyMurphyTM1234, 2022-08-01:
> RISCV Nexus and ARM TPIU tracing are very similar, also the connectors are sharing same pins.

I still don't understand. Perhaps the information conveyed by the two trace mechanisms is similar but, as far as I can see, the physical transport mechanism, never mind board level issues such as physical connections, is not even specified by the RISC-V trace specification. Thee trace specification

https://wiki.riscv.org/plugins/servlet/mobile?contentId=26574878#content/view/26574878

actually states in chapter 7:

> _The infrastructure used to transport these packets is outside the scope of this document, and as such the manner in which packets are encapsulated for transport is not specified._

> Background of my project: We have an ongoing university project with an RISCV core that shall be extended with a Nexus Trace via MIPI20 connector (see https://github.com/riscv-non-isa/tg-nexus-trace/blob/master/pdfs/RISC-V-Trace-Connectors-0-9-2.pdf) by using 4-bit parallel trace.

Bear in mind that that document is not an official ratified RISC-V specification (yet).

> Btw. J-Trace Pro is working fine as a debug-bridge for casual debugging on RISCV with OpenOCD and the winusb driver via JTAG.

I presume that OpenOCD can use the J-Trace as a J-Link, i.e. using the debug subset of the J-Trace's full debug and trace capabilities? Certainly, there's no mention of J-Trace specifically in the OpenOCD documentation.

closed 2024-10-02 by MountaineerGit