riscv-collab/riscv-openocd #728: hello,may i ask,this project just for debug riscv chip or this tools can be run on a riscv64 board?

lgkgkfg, 2022-09-19:
I want to use may riscv64 board to run openocd,is this project can be use for this ?

TommyMurphyTM1234, 2022-09-19:
OpenOCD is for cross debugging from a debug host (usually Windows, Linux, MacOS, FreeBSD etc.) against a RISC-V (or many other CPU/MCU targets) target board using JTAG. I guess that the host debug platform could also be RISC-V (e.g. running Linux) with OpenOCD built to run on that system but debugging a separate RISC-V target via JTAG. But I'm not sure if that's ever been done? So OpenOCD is compiled for the relevant debug host OS and supports cross/remote debugging of the target system (including RISC-V systems) using JTAG (and some other debug mechanisms).

However if you want to debug ON a RISC-V platform then you could run Linux (or another OS) on that RISC-V platform and use gdb to debug user applications there. It probably wouldn't really make sense to run OpenOCD on the RISC-V platform for debugging because there would be no JTAG connection needed to debug on the platform itself. If the RISC-V platform is running bare metal or an RTOS then I don't think that there's any option other than to run OpenOCD on a separate debug host and debug remotely over JTAG.

Hope that makes sense? It might be worth reading the OpenOCD documentation to understand the purpose of and rationale for OpenOCD.

* https://openocd.org/pages/documentation.html

TommyMurphyTM1234, 2022-10-19:
Candidate for closure?

closed 2022-10-20 by timsifive