riscv-collab/riscv-openocd #730: Removal of "test_sba_config_reg" command

JanMatCodasip, 2022-09-21:
I'd like to remove the command `test_sba_config_reg` and the associated function `riscv013_test_sba_config_reg()`. These are leftover from the compliance tests, and those were already removed in https://github.com/riscv/riscv-openocd/pull/567/files#diff-701864e2be3210275085c84b8f6fa7f9eb6171a2a18ca71638be42a155c08c63L4854.

@timsifive:
- Would you agree? 
- If so, do you prefer 
  - a) to remove them directly? 
  - b) or shall I add a deprecation notice first?

Thanks.

timsifive, 2022-09-21:
Just go and remove it. If people want it, they can dig in github and build an old OpenOCD that supports it.

JanMatCodasip, 2023-02-03:
`test_sba_config_reg` has already been removed here: https://github.com/riscv/riscv-openocd/pull/780

Closing this issue.

closed 2023-02-03 by JanMatCodasip