riscv-collab/riscv-openocd #737: OScan1 mode doesn't work

mrv96, 2022-09-29:
After PR #727 OScan1 mode doesn't work anymore, init command fails. In the next days I'll try to figure out the issue and fix it in PR #736.

timsifive, 2022-09-29:
That is very curious. Thank you for taking a look at it.

mrv96, 2022-10-03:
The issue is due to this line: https://github.com/riscv/riscv-openocd/blob/550a66e72094bc59b33fd2f020bf78acf731f60c/src/target/target.c#L3053

Increasing `polling_interval` to 400 solves the issue.

I didn't understood well the code involved in PR #727, @timsifive what do you suggest to fix this issue?

timsifive, 2022-10-03:
In my test case, Oscan1 still works after #727. #727 just changes the timing of when some functions are called. Sounds like you're running into a race condition. Can you collected `openocd -d` output from before the change (which works) and after the change (where it doesn't work)? Then we can sift through that and hopefully get an idea of what's going wrong.

TommyMurphyTM1234, 2022-10-03:
Apologies if this is a dumb question/connect, but is a race condition actually possible in OpenOCD given that, apart from asynchronous signal/Ctrl-C handling, it is single threaded? As far as I know all OpenOCD side JTAG interface/driver and related logic is also synchronous and controlled by the host? Or am I mistaken?

timsifive, 2022-10-03:
If OpenOCD looks at the wall time to decide what to call next, then the speed of the host, connection, etc. can all affect the behavior of OpenOCD. That might not technically be a race condition, but that's what I mean.

sifiverobert, 2022-10-03:
I agree with Tim - IMO changing time from 100ms(?) to 400ms is just making problem less probable to happen.
I do not see how polling interval may help with Oscan problem ...

mrv96, 2022-10-03:
Output when OpenOCD works: 
[working.txt](https://github.com/riscv/riscv-openocd/files/9699493/working.txt)

Output when OpenOCD doens't work: 
[not_working.txt](https://github.com/riscv/riscv-openocd/files/9699495/not_working.txt)

TommyMurphyTM1234, 2022-10-03:
> If OpenOCD looks at the wall time to decide what to call next, then the speed of the host, connection, etc. can all affect the behavior of OpenOCD. That might not technically be a race condition, but that's what I mean.

Thanks @timsifive - apologies for the pedantry. I understand your point now. 👍

mrv96, 2022-10-03:
> I agree with Tim - IMO changing time from 100ms(?) to 400ms is just making problem less probable to happen. I do not see how polling interval may help with Oscan problem ...

OScan1 multiplexes 3 signals on TMSC, it is three times slower. Maybe the issue is related to this, even if I tried to change adapter speed from 1000 to 3000 without success

sifiverobert, 2022-10-03:
IMO in 'not_working' case things are happening too quickly after release of nSRST (and or nTRST). Is your design using nTRST?
Usually there are some parameters to control 'delays' after each type of reset, so there is no JTAG scanning too early (before HW is ready for it). 
Also, nSRST is open-source, so it may take long time to release from reset. If you start talking after 100ms it may be too early.
I think with proper configuration script (and NOT change in sources ...) it should be possible to make it working.

timsifive, 2022-10-03:
In both logs, the first go through examine() fails in the same way. In the old version, examine() is called again before a halt attempt, and all is well. In the new version examine() is not called as quickly in succession, and the halt fails.
I suspect that @sifiverobert is right and adding some reset delays would solve the issue. Possibly we could solve this by having halt() call examine() if it hasn't happened yet, instead of simply returning an error message. I'm not sure what the right solution is.
@mrv96 can you play with adding delays to at least confirm that we understand what's going on?

sifiverobert, 2022-10-03:
@timsifive - first 'examine' fails the same but with different interval there is different delay before second 'examine'.

I remember, that in one of cases I was forced to disable polling (in OPENOCD script) and 'manually' initialize things using different commands (and 'pause ...' as well). At and of OpenOCD script I enabled polling again (but it was AFTER access to target was OK.

timsifive, 2022-10-03:
Thanks. @mrv96 can you see if https://github.com/riscv/riscv-openocd/tree/halt_examine also works, but without your added delay?

mrv96, 2022-10-03:
@timsifive https://github.com/riscv/riscv-openocd/tree/halt_examine works

sifiverobert, 2022-10-03:
This seems like a good change - it cannot break much.
The very reason I once played with low-level script was fact that examine did NOT work first time.
I would only add a log message, so it is visible that target was automatically 're-examined' at halt request.

sifiverobert, 2022-10-04:
I see that @mrv96 would like to commit this change (as it is original problem ...), but **I am still not certain WHY examine was NOT working first time**. There are different error messages of incorrect IDs etc. Maybe procedure to enter Oscan1 is not 'good enough'?
@mrv96 said, that he is NOT using nSRST nor nTRST, so I do NOT see why we cannot connect to the target on first-try.
@mrv96 - could you do as follows:
1. Connect
2. Disconnect (without powering-down the target - just close OpenOCD).
3. Connect again. If first examine is FAILING it means something basic (in JTAG initialization ...) is wrong as I see no reason why first examine must fail here.

mrv96, 2022-10-04:
>I see that @mrv96 would like to commit this change

@sifiverobert PR #736 aims to add JScan3 support to OpenOCD. JScan3 feature is not directly related to this issue. IMO the PR #736 can be merged, then it can be created a new PR to solve this issue. This issue has been highlighted by PR #727.

>@mrv96 - could you do as follows:
> 
>     1. Connect
> 
>     2. Disconnect (without powering-down the target - just close OpenOCD).
> 
>     3. Connect again. If first examine is FAILING it means something basic (in JTAG initialization ...) is wrong as I see no reason why first examine must fail here.

I'll do it in the next hours

timsifive, 2022-10-04:
> @timsifive - is it the case, that with delay there is polling which actually does 'examine' before 'halt'?

Before #727, examine() would be called a lot more often, which hid whatever the real problem is.

> But I am thinking that we still did NOT find root cause and we fixed something else ...

Correct. However, we found a problem that is exposed by an earlier change. This problem led somebody to come here and ask questions. The proposed fix works around the problem in the same way as used to happen, and makes it so I don't have to diagnose pre-existing problems exposed by #727.

mrv96, 2022-10-05:
> 3\. Connect again. If first examine is FAILING it means something basic (in JTAG initialization ...) is wrong as I see no reason why first examine must fail here.

@sifiverobert the first examine fails always. Step 1 and 3 produce the same output during init. Have you any idea about what could be wrong in my configuration script?

mrv96, 2022-10-05:
What can I do to help with the debugging?

P.S. in your opinion, can this issue be a consequence of a HW bug in the JTAG IP inside chip? From my side I cannot exclude this possibility in advance, even if unlikely IMO.

sifiverobert, 2022-10-05:
Sorry - I am experienced in JTAG/cJTAG and RISC-V debug principles, but I am not 'fluent' in OPENOCD.

I would enable some deep enough logging and compare what is being done in 'first examine' and 'second examine'.
Most likely first examine (this one which fails ...) puts JTAG in some 'better-state' and then 'second examine' is working OK.

I cannot help you more here ...

mrv96, 2022-10-06:
Maybe I found a solution.

https://github.com/riscv/riscv-openocd/blob/0f12a0100757378099f0ec0678ed1d222fadbc76/src/jtag/drivers/ftdi.c#L688-L691

Moving line 688 down to 691 seems to solve the issue

mrv96, 2022-10-07:
I tried to figure out this issue in PR #744.

Resulting log: [working_pr.txt](https://github.com/riscv/riscv-openocd/files/9734708/working_pr.txt)

mrv96, 2022-10-10:
Should we keep this issue open until PR #739 merge? Is it going to be accepted and closed?

timsifive, 2022-10-12:
I thought that #744 fixed this problem, which seems like a better fix than #739. I'd like to discard #739 if we don't need it.

mrv96, 2022-10-12:
I agree with you Tim, however I agree with you also about this: https://github.com/riscv/riscv-openocd/issues/737#issuecomment-1267556019

From my side the issue is closed, but considerations in your comment are still valid. Maybe also #739 should be merged, honestly I don't know... I have not the right expertise to take a decision

timsifive, 2022-10-12:
I'll close #739. If we run across a problem that requires it, we can always revive it. (Part of my motivation is that #739 would be yet another target-independent change to upstream, which tends to be slow and I'm already way behind.)

closed 2022-10-13 by mrv96