riscv-collab/riscv-openocd #749: Error: [riscv.cpu0] Hart is not halted! in riscv-013 examine

fborisovskii, 2022-10-18:
Hello,

Seems after changes https://github.com/riscv/riscv-openocd/pull/743
examine function in riscv-013.c causes error message for harts with no halt on reset:

```
static int examine(struct target *target)
...
	if (!halted)
		riscv013_step_or_resume_current_hart(target, false);
```
The function riscv013_step_or_resume_current_hart prints error when target is not in HALTED state.
```
static int riscv013_step_or_resume_current_hart(struct target *target,
		bool step)
{
	if (target->state != TARGET_HALTED) {
		LOG_TARGET_ERROR(target, "Hart is not halted!");
		return ERROR_FAIL;
	}
...
```

And target->state = TARGET_HALTED is set later in deassert_reset function

I've combined old and new approaches together to overcome this issue
(and not calling riscv_is_halted every time):
```
static int riscv013_step_or_resume_current_hart(struct target *target,
		bool step)
{
	if (target->state != TARGET_HALTED && !riscv_is_halted(target)) {
		LOG_TARGET_ERROR(target, "Hart is not halted!");
		return ERROR_FAIL;
	}
...
```

Please check it out.
Log with error message is in attachment.
[ocd.log](https://github.com/riscv/riscv-openocd/files/9812566/ocd.log)

Regards,
Fedor

fborisovskii, 2022-10-19:
I've tried #750 but seems it cannot help in this case. #750 is about deassert_reset but this behavior is caused by the code inside examine as I can investigate. 

I also thought that maybe there is a need to set target->state after calling riscv13_halt_go in examine func:

```
static int examine(struct target *target)
{
	...
	bool halted = riscv_is_halted(target);
	if (!halted) {
		r->prepped = true;
		if (riscv013_halt_go(target) != ERROR_OK) {
			LOG_TARGET_ERROR(target, "Fatal: Hart %d failed to halt during examine()",
					info->index);
			return ERROR_FAIL;
		}
	}

	/* Without knowing anything else we can at least mess with the
		* program buffer. */
	r->debug_buffer_size = info->progbufsize;

	int result = register_read_abstract(target, NULL, GDB_REGNO_S0, 64);
	if (result == ERROR_OK)
		r->xlen = 64;
	else
		r->xlen = 32;

	/* Save s0 and s1. The register cache hasn't be initialized yet so we
	 * need to take care of this manually. */
	uint64_t s0, s1;
	if (register_read_direct(target, &s0, GDB_REGNO_S0) != ERROR_OK) {
		LOG_TARGET_ERROR(target, "Fatal: Failed to read s0.");
		return ERROR_FAIL;
	}
	if (register_read_direct(target, &s1, GDB_REGNO_S1) != ERROR_OK) {
		LOG_TARGET_ERROR(target, "Fatal: Failed to read s1.");
		return ERROR_FAIL;
	}

	if (register_read_direct(target, &r->misa, GDB_REGNO_MISA)) {
		LOG_TARGET_ERROR(target, "Fatal: Failed to read MISA.");
		return ERROR_FAIL;
	}

	if (riscv_supports_extension(target, 'V')) {
		if (discover_vlenb(target) != ERROR_OK)
			return ERROR_FAIL;
	}

	/* Now init registers based on what we discovered. */
	if (riscv_init_registers(target) != ERROR_OK)
		return ERROR_FAIL;

	/* Display this as early as possible to help people who are using
	 * really slow simulators. */
	LOG_TARGET_DEBUG(target, " XLEN=%d, misa=0x%" PRIx64, r->xlen, r->misa);

	/* Restore s0 and s1. */
	if (register_write_direct(target, GDB_REGNO_S0, s0) != ERROR_OK) {
		LOG_TARGET_ERROR(target, "Fatal: Failed to write back s0.");
		return ERROR_FAIL;
	}
	if (register_write_direct(target, GDB_REGNO_S1, s1) != ERROR_OK) {
		LOG_TARGET_ERROR(target, "Fatal: Failed to write back s1.");
		return ERROR_FAIL;
	}

	if (!halted)
		riscv013_step_or_resume_current_hart(target, false);
        ...
}

```

At the start of code section above riscv013_halt_go is called when !halted detected, but target->state is not changed to TARGET_HALTED.
In the end of the code section above riscv013_step_or_resume_current_hart is called but state is still not halted.
Sorry I'm not familiar with the concepts of this code. Just trying to investigate why Error is in the log.

timsifive, 2022-10-21:
Calling is_halted() checks the state on the target, but doesn't set it. That might be the problem. I will push a change that gets rid of is_halted() altogether next week, and that should fix the issue properly.

fborisovskii, 2022-10-21:
Thank you. That seems the root cause. I'll use my fix locally till your change.

fborisovskii, 2022-11-24:
Hello Tim,
After latest update I've investigated a bit more. Can you explain a bit about comment in riscv013_halt_go function?

```
	if (dm->current_hartid == HART_INDEX_MULTIPLE) {
		target_list_t *entry;
		list_for_each_entry(entry, &dm->target_list, list) {
			struct target *t = entry->target;
			t->state = TARGET_HALTED;
			if (t->debug_reason == DBG_REASON_NOTHALTED)
				t->debug_reason = DBG_REASON_DBGRQ;
		}
	}
	/* The "else" case is handled in halt_go(). */
```
In my implementation I have no multiple harts selection. So I never get into this  сondition.
And never set TARGET_HALTED thereafter.
I've added else branch to this condition and set TARGET_HALTED state to original target.
```
	if (dm->current_hartid == HART_INDEX_MULTIPLE) {
		target_list_t *entry;
		list_for_each_entry(entry, &dm->target_list, list) {
			struct target *t = entry->target;
			t->state = TARGET_HALTED;
			if (t->debug_reason == DBG_REASON_NOTHALTED)
				t->debug_reason = DBG_REASON_DBGRQ;
		}
	}
	else {
		target->state = TARGET_HALTED;
	}
```
This fixes my issue. 
Is it a right solution?

timsifive, 2022-11-25:
That breaks some other things.

Looking at the code, the comment is lying though. target->state clearly needs to be updated. Let me investigate a bit more to figure out how to properly solve this.

timsifive, 2022-11-28:
Can you see if #772 fixes this for you?

fborisovskii, 2022-11-30:
Yes. I've checked it out. No Error in log. After this change merge this issue can be closed.

PR #750 (2022-10-21): target/riscv: Correctly set target->state in deassert_reset
This bug didn't lead to problems that I can see, but it would with some upcoming changes. It might fix #749.

Change-Id: I552acbae9977150c4c9e573f8852033bc80fcebb
commit 86e84d3f (2022-11-29): target/riscv: Set target->state in riscv013_halt_go()

Then also set it when we resume in examine(), which doesn't use the full
abstractions because not all required data structures are filled out
yet.

Hopefully fixes #749.

Change-Id: I0c6ab16da1f035ca2fbdb9f7be1462d44ddce3a0
PR #772 (2022-11-30): target/riscv: Set target->state in riscv013_halt_go()
Then also set it when we resume in examine(), which doesn't use the full abstractions because not all required data structures are filled out yet.

Hopefully fixes #749.

Change-Id: I0c6ab16da1f035ca2fbdb9f7be1462d44ddce3a0
closed 2022-11-30 by timsifive