riscv-collab/riscv-openocd #757: riscv 0.11 debug not able to do flash programming

fanghuaqi, 2022-10-31:
As commit  64f3f8877e9dbedf7b80ea3d544e5a6a5b17ed61 do,  

~~~c
	LOG_DEBUG("resume at 0x%" TARGET_PRIxADDR, entry_point);
	if (riscv_resume(target, 0, entry_point, 0, 1, true) != ERROR_OK)
		return ERROR_FAIL;
~~~

it might cause riscv 0.11 can't do flash programing, which I met as follows.

[image]

we can solve it by change `0, 1` to `0, 0`, but not sure it will make any bad influnce on openocd other features, could you help check it.

Thanks

timsifive, 2022-10-31:
I don't have time to dig into it in detail this week. I think #679 was right, and riscv-011 needs to be updated. The update is probably really trivial. As a first step, I'd try to remove the debug_execution check on line 1200 of riscv-011.c. There's nothing really special that happens when debug_execution is true.

fanghuaqi, 2022-11-01:
Thanks @timsifive , we checked that removing the debug_execution check works, not sure whether it will affect other features in riscv 0.11 version debug.

commit fc210e86 (2022-11-21): target/riscv: Ignore debug_execution in 0.11 resume

It's only used to change what callback events are generated, and there
are none anyway. (That's probably a bug, but since 0.11 is so rare I'm
not going to worry about it.)

Fixes #757.

Change-Id: I5b5df3a9bec927fb0368304229533e2875a83f6b
closed 2022-11-21 by timsifive