riscv-collab/riscv-openocd #760: about openocd install question

crstial-zy, 2022-11-08:
i want install openocd , but [after](url) i done ./configure , execute the make setup , hint src/Makefile.am:17: *** missing separato rmissing separator %s。 stop , I thought this file came with you. How could it go wrong. 
# ./configure --enable-ftdi --enable-dummy
..........
libjaylink configuration summary:
 - Package version ................ 0.2.0
 - Library version ................ 1:0:1
 - Installation prefix ............ /usr/local
 - Building on .................... x86_64-unknown-linux-gnu
 - Building for ................... x86_64-unknown-linux-gnu

Enabled transports:
 - USB ............................ yes
 - TCP ............................ yes

OpenOCD configuration summary
--------------------------------------------------
MPSSE mode of FTDI based devices        yes (auto)
cJTAG (OScan1, JScan3) tunneled thru MPSSEyes (auto)
ST-Link Programmer                      yes (auto)
TI ICDI JTAG Programmer                 yes (auto)
Keil ULINK JTAG Programmer              yes (auto)
Altera USB-Blaster II Compatible        yes (auto)
Bitbang mode of FT232R based devices    yes (auto)
Versaloon-Link JTAG Programmer          yes (auto)
TI XDS110 Debug Probe                   yes (auto)
CMSIS-DAP v2 Compliant Debugger         yes (auto)
OSBDM (JTAG only) Programmer            yes (auto)
eStick/opendous JTAG Programmer         yes (auto)
Olimex ARM-JTAG-EW Programmer           yes (auto)
Raisonance RLink JTAG Programmer        yes (auto)
USBProg JTAG Programmer                 yes (auto)
Andes JTAG Programmer (deprecated)      no
CMSIS-DAP Compliant Debugger            no
Nu-Link Programmer                      no
Cypress KitProg Programmer              no
Altera USB-Blaster Compatible           no
ASIX Presto Adapter                     no
OpenJTAG Adapter                        no
Linux GPIO bitbang through libgpiod     no
SEGGER J-Link Programmer                yes (auto)
Bus Pirate                              yes (auto)
Use Capstone disassembly framework      no

#make
/home/ICer/openocd_riscv/src/Makefile.am:17: *** missing separator missing separator %s。 停止。

timsifive, 2022-11-16:
I don't know what's going wrong. My guess would be that you're running an old version of automake or autoconf. I believe you need at least automake 1.14 and autoconf 2.64.

TommyMurphyTM1234, 2022-11-28:
My most recent build worked fine - if you are still having problems then you need to provide more information.

https://github.com/riscv/riscv-openocd/issues/760

TommyMurphyTM1234, 2023-03-16:
Candidate for closure?