riscv-collab/riscv-openocd #771: Error building riscv-openocd

everetto89, 2022-11-28:
Hi, 

when I try to build the riscv-openocd I get the following error: 

```
src/jtag/drivers/arm-jtag-ew.c: In function ‘armjtagew_debug_buffer’:
src/jtag/drivers/arm-jtag-ew.c:796:22: error: ‘%04x’ directive output may be truncated writing between 4 and 8 bytes into a region of size 5 [-Werror=format-truncation=]
   snprintf(line, 5, "%04x", i);
                      ^~~~
src/jtag/drivers/arm-jtag-ew.c:796:21: note: directive argument in the range [0, 2147483632]
   snprintf(line, 5, "%04x", i);
                     ^~~~~~
In file included from /usr/include/stdio.h:862:0,
                 from ./jimtcl/jim.h:74,
                 from ./src/helper/jim-nvp.h:47,
                 from ./src/helper/command.h:28,
                 from ./src/helper/log.h:28,
                 from ./src/jtag/jtag.h:26,
                 from ./src/jtag/interface.h:28,
                 from src/jtag/drivers/arm-jtag-ew.c:23:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:64:10: note: ‘__builtin___snprintf_chk’ output between 5 and 9 bytes into a destination of size 5
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
Makefile:3437: recipe for target 'src/jtag/drivers/src_jtag_drivers_libocdjtagdrivers_la-arm-jtag-ew.lo' failed
make[2]: *** [src/jtag/drivers/src_jtag_drivers_libocdjtagdrivers_la-arm-jtag-ew.lo] Error 1
make[2]: Leaving directory '/home/user/RISC_V/riscv-openocd'
Makefile:4577: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/user/RISC_V/riscv-openocd'
Makefile:1679: recipe for target 'all' failed
make: *** [all] Error 2
```

Can you please help here? 

Best regards, 
Christian

TommyMurphyTM1234, 2022-11-28:
Please list the clone, configure, and make commands that you use, details of the platform and tools used, and the build log.

TommyMurphyTM1234, 2022-12-01:
Any update @everetto89?

TommyMurphyTM1234, 2022-12-15:
Candidate for closure - not reproducible?

everetto89, 2022-12-16:
Hi @TommyMurphyTM1234, 
here is my configuration: 

```
./configure --enable-verbose --enable-dummy --enable-ftdi --enable-ftdi-cjtag --enable-stlink --enable-ulink --enable-usb-blaster-2 --enable-ft232r --enable-cmsis-dap-v2 --enable-usbprog --enable-cmsis-dap --enable-nulink --enable-usb-blaster --enable-openjtag --enable-jlink --enable-remote-bitbang
```
Which results in the following Configuration summary: 

```
OpenOCD configuration summary
--------------------------------------------------
MPSSE mode of FTDI based devices        yes
cJTAG (OScan1, JScan3) tunneled thru MPSSEyes
ST-Link Programmer                      yes
TI ICDI JTAG Programmer                 yes (auto)
Keil ULINK JTAG Programmer              yes
Altera USB-Blaster II Compatible        yes
Bitbang mode of FT232R based devices    yes
Versaloon-Link JTAG Programmer          yes (auto)
TI XDS110 Debug Probe                   yes (auto)
CMSIS-DAP v2 Compliant Debugger         yes
OSBDM (JTAG only) Programmer            yes (auto)
eStick/opendous JTAG Programmer         yes (auto)
Olimex ARM-JTAG-EW Programmer           yes (auto)
Raisonance RLink JTAG Programmer        yes (auto)
USBProg JTAG Programmer                 yes
Andes JTAG Programmer (deprecated)      no
CMSIS-DAP Compliant Debugger            yes
Nu-Link Programmer                      yes
Cypress KitProg Programmer              yes (auto)
Altera USB-Blaster Compatible           yes
ASIX Presto Adapter                     yes (auto)
OpenJTAG Adapter                        yes
Linux GPIO bitbang through libgpiod     no
SEGGER J-Link Programmer                yes
Bus Pirate                              yes (auto)
Use Capstone disassembly framework      no

```

I'm using a fresh clone, `git log` shows me that I try to build on the following commit: 

```
commit b337b0cfb4612c0474c109abeb511189bfd5559c (HEAD -> riscv, origin/riscv, origin/HEAD)
Author: Dolu1990 <charles.papon.90@gmail.com>
Date:   Fri Dec 2 18:38:40 2022 +0100
```
Can you help here?

TommyMurphyTM1234, 2022-12-16:
> Hi @TommyMurphyTM1234, here is my configuration:

You haven't provided the other stuff that I mentioned above:

> Please list the clone, configure, **and make commands that you use, details of the platform and tools used, and the build log.**

TommyMurphyTM1234, 2022-12-16:
OK - when I use your `configure` command I can reproduce it.
But I can also reproduce it when I clone from the upstream OpenOCD repository:

https://openocd.org/pages/repos.html

So I think that it probably needs to be reported there since it's not a RISC-V specific issue relevant to this repo?

https://openocd.org/pages/bug-tracker.html

TommyMurphyTM1234, 2022-12-16:
A workaround, unless you explicitly need these two specific debug interfaces, is to add the following to your configure command and then build: `--disable-armjtagew --disable-opendous`.

TommyMurphyTM1234, 2022-12-17:
It's very odd - whether I pass the various `--enable-<interface>` options to `configure` or pass none, the compilation commands for `arm-jtag-ew.c` seem to be identical and yet, in the former case we get the compilation error but in the latter we do not...

```
./configure --prefix=`pwd`/installed-tools
make
...
/bin/bash ./libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.   -I./src -I./src -DPKGDATADIR=\"/home/tommy/Downloads/riscv-openocd/installed-tools/share/openocd\" -DBINDIR=\"/home/tommy/Downloads/riscv-openocd/installed-tools/bin\" -DFD_SETSIZE=128 -I./jimtcl -I./jimtcl -isystem /usr/include/libusb-1.0 -I/usr/include/libftdi1 -I/usr/include/libusb-1.0  -I/usr/include/hidapi  -I./src/jtag/drivers/libjaylink/libjaylink -I./src/jtag/drivers/libjaylink  -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Wundef -Wno-error=deprecated-declarations -Werror -DFD_SETSIZE=128 -g -O2 -MT src/jtag/drivers/libocdjtagdrivers_la-arm-jtag-ew.lo -MD -MP -MF src/jtag/drivers/.deps/libocdjtagdrivers_la-arm-jtag-ew.Tpo -c -o src/jtag/drivers/libocdjtagdrivers_la-arm-jtag-ew.lo `test -f 'src/jtag/drivers/arm-jtag-ew.c' || echo './'`src/jtag/drivers/arm-jtag-ew.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I./src -I./src -DPKGDATADIR=\"/home/tommy/Downloads/riscv-openocd/installed-tools/share/openocd\" -DBINDIR=\"/home/tommy/Downloads/riscv-openocd/installed-tools/bin\" -DFD_SETSIZE=128 -I./jimtcl -I./jimtcl -isystem /usr/include/libusb-1.0 -I/usr/include/libftdi1 -I/usr/include/libusb-1.0 -I/usr/include/hidapi -I./src/jtag/drivers/libjaylink/libjaylink -I./src/jtag/drivers/libjaylink -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Wundef -Wno-error=deprecated-declarations -Werror -DFD_SETSIZE=128 -g -O2 -MT src/jtag/drivers/libocdjtagdrivers_la-arm-jtag-ew.lo -MD -MP -MF src/jtag/drivers/.deps/libocdjtagdrivers_la-arm-jtag-ew.Tpo -c src/jtag/drivers/arm-jtag-ew.c -o src/jtag/drivers/libocdjtagdrivers_la-arm-jtag-ew.o
mv -f src/jtag/drivers/.deps/libocdjtagdrivers_la-arm-jtag-ew.Tpo src/jtag/drivers/.deps/libocdjtagdrivers_la-arm-jtag-ew.Plo
```
```
./configure --enable-verbose --enable-dummy --enable-ftdi --enable-ftdi-cjtag --enable-stlink --enable-ulink --enable-usb-blaster-2 --enable-ft232r --enable-cmsis-dap-v2 --enable-usbprog --enable-cmsis-dap --enable-nulink --enable-usb-blaster --enable-openjtag --enable-jlink --enable-remote-bitbang --prefix=`pwd`/installed-tools
make
...
/bin/bash ./libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.   -I./src -I./src -DPKGDATADIR=\"/home/tommy/Downloads/riscv-openocd/installed-tools/share/openocd\" -DBINDIR=\"/home/tommy/Downloads/riscv-openocd/installed-tools/bin\" -DFD_SETSIZE=128 -I./jimtcl -I./jimtcl -isystem /usr/include/libusb-1.0 -I/usr/include/libftdi1 -I/usr/include/libusb-1.0  -I/usr/include/hidapi  -I./src/jtag/drivers/libjaylink/libjaylink -I./src/jtag/drivers/libjaylink  -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Wundef -Wno-error=deprecated-declarations -Werror -DFD_SETSIZE=128 -g -O2 -MT src/jtag/drivers/libocdjtagdrivers_la-arm-jtag-ew.lo -MD -MP -MF src/jtag/drivers/.deps/libocdjtagdrivers_la-arm-jtag-ew.Tpo -c -o src/jtag/drivers/libocdjtagdrivers_la-arm-jtag-ew.lo `test -f 'src/jtag/drivers/arm-jtag-ew.c' || echo './'`src/jtag/drivers/arm-jtag-ew.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I./src -I./src -DPKGDATADIR=\"/home/tommy/Downloads/riscv-openocd/installed-tools/share/openocd\" -DBINDIR=\"/home/tommy/Downloads/riscv-openocd/installed-tools/bin\" -DFD_SETSIZE=128 -I./jimtcl -I./jimtcl -isystem /usr/include/libusb-1.0 -I/usr/include/libftdi1 -I/usr/include/libusb-1.0 -I/usr/include/hidapi -I./src/jtag/drivers/libjaylink/libjaylink -I./src/jtag/drivers/libjaylink -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Wundef -Wno-error=deprecated-declarations -Werror -DFD_SETSIZE=128 -g -O2 -MT src/jtag/drivers/libocdjtagdrivers_la-arm-jtag-ew.lo -MD -MP -MF src/jtag/drivers/.deps/libocdjtagdrivers_la-arm-jtag-ew.Tpo -c src/jtag/drivers/arm-jtag-ew.c -o src/jtag/drivers/libocdjtagdrivers_la-arm-jtag-ew.o
src/jtag/drivers/arm-jtag-ew.c: In function ‘armjtagew_debug_buffer’:
src/jtag/drivers/arm-jtag-ew.c:796:22: error: ‘%04x’ directive output may be truncated writing between 4 and 8 bytes into a region of size 5 [-Werror=format-truncation=]
  796 |   snprintf(line, 5, "%04x", i);
      |                      ^~~~
src/jtag/drivers/arm-jtag-ew.c:796:21: note: directive argument in the range [0, 2147483632]
  796 |   snprintf(line, 5, "%04x", i);
      |                     ^~~~~~
In file included from /usr/include/stdio.h:867,
                 from ./jimtcl/jim.h:74,
                 from ./src/helper/jim-nvp.h:47,
                 from ./src/helper/command.h:28,
                 from ./src/helper/log.h:28,
                 from ./src/jtag/jtag.h:26,
                 from ./src/jtag/interface.h:28,
                 from src/jtag/drivers/arm-jtag-ew.c:23:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:67:10: note: ‘__builtin___snprintf_chk’ output between 5 and 9 bytes into a destination of size 5
[... cut ...]
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   68 |        __bos (__s), __fmt, __va_arg_pack ());
      |        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [Makefile:3680: src/jtag/drivers/libocdjtagdrivers_la-arm-jtag-ew.lo] Error 1
make[2]: Leaving directory '/home/tommy/Downloads/riscv-openocd'
make[1]: *** [Makefile:4820: all-recursive] Error 1
make[1]: Leaving directory '/home/tommy/Downloads/riscv-openocd'
```

TommyMurphyTM1234, 2022-12-17:
OK -  the `--enable-<interface>` options are a red herring. The issue is that when `--enable-verbose` is specified to `configure` then code that triggers the compilation error is enabled.

TommyMurphyTM1234, 2022-12-17:
Changing the code in `arm-jtag-ew.c` and `opendous.c` as follows eliminates the compilation error but I'm not yet sure if it's the "correct" fix. I'll try to follow up on it upstream.
```
	char line[81];
	char s[4];
        /* int i; --> change this to uint16_t */
	uint16_t i;**
	int j;

	for (i = 0; i < length; i += BYTES_PER_LINE) {
		snprintf(line, 5, "%04x", i);
```

TommyMurphyTM1234, 2022-12-17:
Ive logged an OpenOCD ticket upstream here: https://sourceforge.net/p/openocd/tickets/376/

borneoa, 2022-12-17:
Please check https://review.openocd.org/7403/
@everetto89 I can add also you as Reported-by in the commit message, but I would need your email contact

TommyMurphyTM1234, 2022-12-18:
> A workaround, unless you explicitly need these two specific debug interfaces, is to add the following to your configure command and then build: `--disable-armjtagew --disable-opendous`. Both Olimex ARM-JTAG-EW Programmer and eStick/opendous JTAG Programmer suffer from the same compilation error.

A simpler workaround is to omit `--enable-verbose` when invoking `configure`.

TommyMurphyTM1234, 2022-12-18:
> Please check https://review.openocd.org/7403/ @everetto89 I can add also you as Reported-by in the commit message, but I would need your email contact

Thanks a lot @borneoa for the quick response - I've followed up over there.

TommyMurphyTM1234, 2022-12-20:
@borneoa's change to address this issue seems to have been merged upstream so I guess that this issue here can be closed and next time the RISC-V OpenOCD fork is sync'd with upstream OpenOCD the fix will appear here. In the meantime he change can be manually applied here if needed.

closed 2023-03-15 by timsifive