riscv-collab/riscv-openocd #774: Error: dtmcontrol is 0. Check JTAG connectivity/board power.

athurwo, 2022-12-01:
Hello,
I had installed riscv-openocd,  and when I run it:
`/usr/local/bin/openocd -f openocd.cfg`

the infomation like below:
`Open On-Chip Debugger 0.11.0+dev-02442-gf7a5f4719 (2022-12-01-16:07)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
0
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 5000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x43651093 (mfg: 0x049 (Xilinx), part: 0x3651, ver: 0x4)
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Warn : target riscv.cpu examination failed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
`
the jtag tap had been getted,  but dtmcontrol is 0,
what it mean ? Does that means abits in dtmcs is 0 ?
How to resovle it ?
Thanks.

TommyMurphyTM1234, 2022-12-01:
You'll probably need to attach a verbose log captured using `openocd -d3`. Based on the info posted the FPGA (I presume) TAP is responding but the target RISC-V may not be - e.g. may be responding with all zeros on TDO.

athurwo, 2022-12-01:
> You'll probably need to attach a verbose log captured using `openocd -d3`. Based on the info posted the FPGA (I presume) TAP is responding but the target RISC-V may not be - e.g. may be responding with all zeros on TDO.

I had add -d3, the info is :

Open On-Chip Debugger 0.11.0+dev-02442-gf7a5f4719 (2022-12-01-16:07)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 3 2 options.c:63 configuration_output_handler(): debug_level: 3
User : 4 2 options.c:63 configuration_output_handler(): 
Debug: 5 2 options.c:244 add_default_dirs(): bindir=/usr/local/bin
Debug: 6 2 options.c:245 add_default_dirs(): pkgdatadir=/usr/local/share/openocd
Debug: 7 2 options.c:246 add_default_dirs(): exepath=/usr/local/bin
Debug: 8 2 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 9 2 configuration.c:44 add_script_search_dir(): adding /home/fso/.config/openocd
Debug: 10 2 configuration.c:44 add_script_search_dir(): adding /home/fso/.openocd
Debug: 11 2 configuration.c:44 add_script_search_dir(): adding /usr/local/bin/../share/openocd/site
Debug: 12 2 configuration.c:44 add_script_search_dir(): adding /usr/local/bin/../share/openocd/scripts
Debug: 13 2 command.c:166 script_debug(): command - ocd_find openocd-nexys-hs2.cfg
Debug: 14 2 configuration.c:99 find_file(): found openocd-nexys-hs2.cfg
Debug: 15 2 command.c:166 script_debug(): command - adapter driver ftdi
Debug: 16 2 command.c:166 script_debug(): command - ftdi vid_pid 0x0403 0x6014
Debug: 17 2 command.c:166 script_debug(): command - transport select jtag
Debug: 18 2 command.c:166 script_debug(): command - ftdi layout_init 0x00e8 0x60eb
Debug: 19 2 command.c:166 script_debug(): command - reset_config none
Debug: 20 2 command.c:166 script_debug(): command - adapter speed 5000
Debug: 21 2 adapter.c:176 adapter_config_khz(): handle adapter khz
Debug: 22 2 adapter.c:140 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 23 2 adapter.c:140 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 24 2 command.c:166 script_debug(): command - jtag newtap riscv cpu -irlen 5 -expected-id 0x43651093
Debug: 25 2 tcl.c:568 jim_newtap_cmd(): Creating New Tap, Chip: riscv, Tap: cpu, Dotted: riscv.cpu, 4 params
Debug: 26 2 tcl.c:593 jim_newtap_cmd(): Processing option: -irlen
Debug: 27 2 tcl.c:593 jim_newtap_cmd(): Processing option: -expected-id
Debug: 28 2 core.c:1468 jtag_tap_init(): Created Tap: riscv.cpu @ abs position 0, irlen 5, capture: 0x1 mask: 0x3
Debug: 29 2 command.c:166 script_debug(): command - target create riscv.cpu riscv -chain-position riscv.cpu -coreid 0x0
Debug: 30 2 target.c:2214 target_free_all_working_areas_restore(): freeing all working areas
Debug: 31 3 riscv.c:356 riscv_create_target(): riscv_create_target()
User : 32 3 options.c:63 configuration_output_handler(): 0User : 33 3 options.c:63 configuration_output_handler(): 
Info : 34 3 server.c:308 add_service(): Listening on port 6666 for tcl connections
Info : 35 3 server.c:308 add_service(): Listening on port 4444 for telnet connections
Debug: 36 3 command.c:166 script_debug(): command - init
Debug: 37 3 command.c:166 script_debug(): command - target init
Debug: 38 3 command.c:166 script_debug(): command - target names
Debug: 39 3 command.c:166 script_debug(): command - riscv.cpu cget -event gdb-flash-erase-start
Debug: 40 3 command.c:166 script_debug(): command - riscv.cpu configure -event gdb-flash-erase-start reset init
Debug: 41 3 command.c:166 script_debug(): command - riscv.cpu cget -event gdb-flash-write-end
Debug: 42 3 command.c:166 script_debug(): command - riscv.cpu configure -event gdb-flash-write-end reset halt
Debug: 43 3 command.c:166 script_debug(): command - riscv.cpu cget -event gdb-attach
Debug: 44 3 command.c:166 script_debug(): command - riscv.cpu configure -event gdb-attach halt 1000
Debug: 45 3 target.c:1671 handle_target_init_command(): Initializing targets...
Debug: 46 3 riscv.c:369 riscv_init_target(): riscv_init_target()
Debug: 47 3 semihosting_common.c:130 semihosting_common_init():  
Debug: 48 3 ftdi.c:742 ftdi_initialize(): ftdi interface using shortest path jtag state transitions
Debug: 49 4 mpsse.c:423 mpsse_purge(): -
Debug: 50 4 mpsse.c:704 mpsse_loopback_config(): off
Debug: 51 4 mpsse.c:749 mpsse_set_frequency(): target 5000000 Hz
Debug: 52 4 mpsse.c:741 mpsse_rtck_config(): off
Debug: 53 4 mpsse.c:730 mpsse_divide_by_5_config(): off
Debug: 54 4 mpsse.c:710 mpsse_set_divisor(): 5
Debug: 55 4 mpsse.c:773 mpsse_set_frequency(): actually 5000000 Hz
Debug: 56 4 adapter.c:140 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 57 4 adapter.c:144 adapter_khz_to_speed(): have adapter set up
Debug: 58 4 mpsse.c:749 mpsse_set_frequency(): target 5000000 Hz
Debug: 59 4 mpsse.c:741 mpsse_rtck_config(): off
Debug: 60 4 mpsse.c:730 mpsse_divide_by_5_config(): off
Debug: 61 4 mpsse.c:710 mpsse_set_divisor(): 5
Debug: 62 4 mpsse.c:773 mpsse_set_frequency(): actually 5000000 Hz
Debug: 63 4 adapter.c:140 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 64 4 adapter.c:144 adapter_khz_to_speed(): have adapter set up
Info : 65 4 adapter.c:104 adapter_init(): clock speed 5000 kHz
[... 21 lines cut ...]
Debug: 87 5 riscv.c:328 dtmcontrol_scan(): DTMCONTROL: 0x0 -> 0x0
Debug: 88 5 riscv.c:1225 riscv_examine(): dtmcontrol=0x0
Debug: 89 5 riscv.c:1227 riscv_examine():   version=0x0
Debug: 90 5 riscv-011.c:2365 init_target(): init
[... cut ...]
Debug: 106 5 pld.c:205 handle_pld_init_command(): Initializing PLDs...
Debug: 107 5 command.c:166 script_debug(): command - tpiu init
Info : 108 5 gdb_server.c:3835 gdb_target_start(): starting gdb server for riscv.cpu on 3333
Info : 109 5 server.c:308 add_service(): Listening on port 3333 for gdb connections
Debug: 110 106 target.c:1857 target_call_event_callbacks(): target event 0 (gdb-halt) for core riscv.cpu
Debug: 111 106 target.c:3072 handle_target(): [riscv.cpu] target_poll() -> -4, next attempt in 201ms

TommyMurphyTM1234, 2022-12-01:
Looks to me like once it tries to communicate with the RISC-V debug block via the FPGA TAP it's getting just zeros back on TDO. That would suggest some hardware side problem. E.g. target stuck in reset, debug signals not wired out correctly, timing issues etc. Impossible to tell without more info. Have you successfully simulated your design including debug access? What RISC-V implemention are you using? How do you know that it's functional, including debug access? Nothing here points to an issue with OpenOCD itself so far.

athurwo, 2022-12-01:
> Looks to me like once it tries to communicate with the RISC-V debug block via the FPGA TAP it's getting just zeros back on TDO. That would suggest some hardware side problem. E.g. target stuck in [...]

uint32_t dtmcontrol = dtmcontrol_scan(target, 0);

static uint32_t dtmcontrol_scan(struct target *target, uint32_t out)
{
	struct scan_field field;
	uint8_t in_value[4];
	uint8_t out_value[4] = { 0 };

	buf_set_u32(out_value, 0, 32, out);

	jtag_add_ir_scan(target->tap, &select_dtmcontrol, TAP_IDLE);

	field.num_bits = 32;
	field.out_value = out_value;
	field.in_value = in_value;
	jtag_add_dr_scan(target->tap, 1, &field, TAP_IDLE);

	/* Always return to dbus. */
	jtag_add_ir_scan(target->tap, &select_dbus, TAP_IDLE);

	int retval = jtag_execute_queue();
	if (retval != ERROR_OK) {
		LOG_ERROR("failed jtag scan: %d", retval);
		return retval;
	}

	uint32_t in = buf_get_u32(field.in_value, 0, 32);
	LOG_DEBUG("DTMCONTROL: 0x%x -> 0x%x", out, in);

	return in;
}

in_value just define, but no initial, default is 0,  I cannot find function buf_get_u32 defination, 
out is 0,  in= buf_get_u32(0,0,32);

TommyMurphyTM1234, 2022-12-01:
I don't understand the point of your last post above.
You need to check/verify that your hardware target and debug block are working.
Either through simulation or maybe by bringing signals out for probing via Vivado or scope.
Looking at the OpenOCD source isn't going to help in this context, in my opinion.

athurwo, 2022-12-02:
> I don't understand the point of your last post above.
> You need to check/verify that your hardware target and debug block are working.
> Either through simulation or maybe by bringing signals out for probing via Vivado or scope.
> Looking at the OpenOCD source code isn't going to help in this context, in my opinion.
> We know that that code works if the hardware target is functional.
> It seems most likely to me that yours is not and you need to check that first.

my hardware target is good, I had tested by a led and key bitstream. 
 last post, I want to know where is define of buf_get_u32() .

TommyMurphyTM1234, 2022-12-02:
> my hardware target is good, I had tested by a led and key bitstream. 

That just means that the CPU can execute some code. It doesn't mean that the debug block is functional or wired out to the FPGA TAP correctly and works.

> last post, I want to know where is define of buf_get_u32() .

`grep` or any source code indexer tool will find it for you.

athurwo, 2022-12-02:
> > my hardware target is good, I had tested by a led and key bitstream.
> 
> That just means that the CPU can execute some code. It doesn't mean that the debug block is functional or wired out to the FPGA TAP correctly. And all zeros when trying to contact it suggests that it's not.
> 
> > last post, I want to know where is define of buf_get_u32() .
> 
> `grep` or any source code indexer tool will find it for you.

maybe you are right,  I try it.
thanks.

athurwo, 2022-12-02:
> That just means that the CPU can execute some code. It doesn't mean that the debug block is functional or wired out to the FPGA TAP correctly. And all zeros when trying to contact it suggests that [...]

I modify some code in riscv.c like below:

`
#define DTMCONTROL					0x0`

I run openocd again with -d3, info like below:

Open On-Chip Debugger 0.11.0+dev-02442-gf7a5f4719 (2022-12-01-16:07)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 3 3 options.c:63 configuration_output_handler(): debug_level: 3
User : 4 3 options.c:63 configuration_output_handler(): 
Debug: 5 3 options.c:244 add_default_dirs(): bindir=/usr/local/bin
Debug: 6 3 options.c:245 add_default_dirs(): pkgdatadir=/usr/local/share/openocd
Debug: 7 3 options.c:246 add_default_dirs(): exepath=/usr/local/bin
Debug: 8 3 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 9 3 configuration.c:44 add_script_search_dir(): adding /home/fso/.config/openocd
Debug: 10 3 configuration.c:44 add_script_search_dir(): adding /home/fso/.openocd
Debug: 11 3 configuration.c:44 add_script_search_dir(): adding /usr/local/bin/../share/openocd/site
Debug: 12 3 configuration.c:44 add_script_search_dir(): adding /usr/local/bin/../share/openocd/scripts
Debug: 13 3 command.c:166 script_debug(): command - ocd_find openocd-nexys-hs2.cfg
Debug: 14 3 configuration.c:99 find_file(): found openocd-nexys-hs2.cfg
Debug: 15 3 command.c:166 script_debug(): command - adapter driver ftdi
Debug: 16 3 command.c:166 script_debug(): command - ftdi vid_pid 0x0403 0x6014
Debug: 17 3 command.c:166 script_debug(): command - transport select jtag
Debug: 18 3 command.c:166 script_debug(): command - ftdi layout_init 0x00e8 0x60eb
Debug: 19 3 command.c:166 script_debug(): command - reset_config none
Debug: 20 3 command.c:166 script_debug(): command - adapter speed 5000
Debug: 21 3 adapter.c:176 adapter_config_khz(): handle adapter khz
Debug: 22 3 adapter.c:140 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 23 3 adapter.c:140 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 24 3 command.c:166 script_debug(): command - jtag newtap riscv cpu -irlen 5 -expected-id 0x43651093
Debug: 25 3 tcl.c:568 jim_newtap_cmd(): Creating New Tap, Chip: riscv, Tap: cpu, Dotted: riscv.cpu, 4 params
Debug: 26 3 tcl.c:593 jim_newtap_cmd(): Processing option: -irlen
Debug: 27 3 tcl.c:593 jim_newtap_cmd(): Processing option: -expected-id
Debug: 28 3 core.c:1468 jtag_tap_init(): Created Tap: riscv.cpu @ abs position 0, irlen 5, capture: 0x1 mask: 0x3
Debug: 29 3 command.c:166 script_debug(): command - target create riscv.cpu riscv -chain-position riscv.cpu -coreid 0x0
Debug: 30 3 target.c:2214 target_free_all_working_areas_restore(): freeing all working areas
Debug: 31 3 riscv.c:356 riscv_create_target(): riscv_create_target()
User : 32 3 options.c:63 configuration_output_handler(): 0User : 33 3 options.c:63 configuration_output_handler(): 
Info : 34 3 server.c:308 add_service(): Listening on port 6666 for tcl connections
Info : 35 3 server.c:308 add_service(): Listening on port 4444 for telnet connections
Debug: 36 3 command.c:166 script_debug(): command - init
Debug: 37 3 command.c:166 script_debug(): command - target init
Debug: 38 3 command.c:166 script_debug(): command - target names
Debug: 39 3 command.c:166 script_debug(): command - riscv.cpu cget -event gdb-flash-erase-start
Debug: 40 3 command.c:166 script_debug(): command - riscv.cpu configure -event gdb-flash-erase-start reset init
Debug: 41 3 command.c:166 script_debug(): command - riscv.cpu cget -event gdb-flash-write-end
Debug: 42 3 command.c:166 script_debug(): command - riscv.cpu configure -event gdb-flash-write-end reset halt
Debug: 43 3 command.c:166 script_debug(): command - riscv.cpu cget -event gdb-attach
Debug: 44 3 command.c:166 script_debug(): command - riscv.cpu configure -event gdb-attach halt 1000
Debug: 45 3 target.c:1671 handle_target_init_command(): Initializing targets...
Debug: 46 3 riscv.c:369 riscv_init_target(): riscv_init_target()
Debug: 47 3 semihosting_common.c:130 semihosting_common_init():  
Debug: 48 4 ftdi.c:742 ftdi_initialize(): ftdi interface using shortest path jtag state transitions
Debug: 49 4 mpsse.c:423 mpsse_purge(): -
Debug: 50 5 mpsse.c:704 mpsse_loopback_config(): off
Debug: 51 5 mpsse.c:749 mpsse_set_frequency(): target 5000000 Hz
Debug: 52 5 mpsse.c:741 mpsse_rtck_config(): off
Debug: 53 5 mpsse.c:730 mpsse_divide_by_5_config(): off
Debug: 54 5 mpsse.c:710 mpsse_set_divisor(): 5
Debug: 55 5 mpsse.c:773 mpsse_set_frequency(): actually 5000000 Hz
Debug: 56 5 adapter.c:140 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 57 5 adapter.c:144 adapter_khz_to_speed(): have adapter set up
Debug: 58 5 mpsse.c:749 mpsse_set_frequency(): target 5000000 Hz
Debug: 59 5 mpsse.c:741 mpsse_rtck_config(): off
Debug: 60 5 mpsse.c:730 mpsse_divide_by_5_config(): off
[... 66 lines cut ...]
Debug: 127 709 riscv.c:328 dtmcontrol_scan(): DTMCONTROL: 0x0 -> 0x6db66dcf
Debug: 128 709 riscv.c:1225 riscv_examine(): dtmcontrol=0x6db66dcf
Debug: 129 709 riscv.c:1227 riscv_examine():   version=0xf
Error: 130 709 riscv.c:348 get_target_type(): [riscv.cpu] Unsupported DTM version: 15
Debug: 131 709 target.c:1857 target_call_event_callbacks(): target event 20 (examine-fail) for core riscv.cpu
[... cut ...]
^CDebug: 143 1876 server.c:616 sig_handler(): Terminating on Signal 2
Debug: 144 1877 command.c:166 script_debug(): command - shutdown
User : 145 1877 server.c:762 handle_shutdown_command(): shutdown command invoked
Debug: 146 1877 riscv.c:422 riscv_deinit_target(): riscv_deinit_target()
Error: 147 1877 riscv.c:348 get_target_type(): [riscv.cpu] Unsupported DTM version: 15
Debug: 148 1877 target.c:2214 target_free_all_working_areas_restore(): freeing all working areas

now, that is not zeros.
why?

TommyMurphyTM1234, 2022-12-02:
> now, that is not zeros. why?

No idea because we have no visibility into your hardware implementation. `DMCONTROL` should be `0x10` in a standard implementation of the RISC-V debug block. I'm not sure why you're defining out as `0x0` or why it changes the behaviour (and makes it work?).

athurwo, 2022-12-02:
> > now, that is not zeros. why?
> 
> No idea because we have no visibility into your hardware implementation. `DMCONTROL` should be `0x10` in a standard implementation of the RISC-V debug block. I'm not sure why you're defining it as `0x0` or why it changes the behaviour. But it obviously doesn't make things work.

thanks, I change back, try other ways.

TommyMurphyTM1234, 2022-12-02:
Why don't you debug/probe the hardware like I suggested to see if debug related signals are working correctly? Seems quite likely to me that the debug block may not be functional at all or in this particular implemention in this FPGA. What unit testing was done on the debug block? Simulation? What RISC-V and debug block implementation are you using?

athurwo, 2022-12-08:
> Why don't you debug/probe the hardware like I suggested to see if debug related signals are working correctly? Seems quite likely to me that the debug block may not be functional at all or in this [...]

How to debug hardware related signals are working correctly?
Do you have some demo about debug?

TommyMurphyTM1234, 2022-12-08:
> How to debug hardware related signals are working correctly? Do you have some demo about debug?

This is way outside the scope of this forum/repo.
You need to study how to bring signals out from your design in order to probe/monitor them using Vivado or an oscilloscope etc.
I presume that the Vivado documentation might have some info about this.

athurwo, 2022-12-08:
> 

> This is way outside the scope of this forum/repo. You need to study how to bring signals out from your design in order to probe/monitor them using Vivado or an oscilloscope etc. I presume that the [...]

Thanks. I believe myself to  resovle it.

TommyMurphyTM1234, 2022-12-15:
Candidate for closure?

closed 2022-12-27 by timsifive