riscv-collab/riscv-openocd #776: Add "init resethalt" command

Alan-19950616, 2022-12-07:
Sometimes Openocd cannot halt the CPU due to a software or hardware problem that accesses the empty address.

At this time If you use “init reset“or"init halt" command in the configuration file, the connection cannot be established.

The "init resethalt" command is designed to solve this problem.

Please help to see if this is reasonable.

add 'init resethalt' command #775

JanMatCodasip, 2022-12-08:
Hi Alan,

I think if you need to issue additional debug commands to work around specific problem (bug?) in the given hardware, you can do it with what OpenOCD already provides. I've further commented on that at https://github.com/riscv/riscv-openocd/pull/775#pullrequestreview-1209581249 Let me know if that would work for you.

sifiverobert, 2022-12-08:
I agree - adding new comment to battle specific problem is not the way to go.
I recall doing it a script (before 'init' is called).
You can do HW reset (by pin ) and you CAN execute different DM actions (by writing to registers ...).

TommyMurphyTM1234, 2022-12-08:
The other thing to bear in mind is that, without a compelling case for new functionality such as this, the chances of it being accepted in the "master" OpenOCD project (when the RISC-V OpenOCD changes are periodically upstreamed) will presumably be slim. I am aware of some cases (particularly in FPGA implementations) where things can happen on the hardware side making examine and/or reset [init|halt] difficult but I thought that some recent changes in this context by @timsifive (and maybe others) may have helped make this more reliable?

Alan-19950616, 2022-12-12:
> Hi Alan,
> 
> I think if you need to issue additional debug commands to work around specific problem (bug?) in the given hardware, you can do it using commands that OpenOCD already provides. I've further commented on that at [#775 (review)](https://github.com/riscv/riscv-openocd/pull/775#pullrequestreview-1209581249). Let me know if that would work for you.

Hi JanMatCodasip, 

I try to use **dmi_write** before **init**, that doesn't seem to be allowed.
[image]

I try to use **dmi_write** after **init**,  it does do something similar, But it still doesn't solve my problem. The problem I have now is that **init** will fail.
[image]

JanMatCodasip, 2022-12-12:
Hi Alan, `riscv dmi_write` & `riscv dmi_read` should be used only after `init`.

You can follow these steps:

1) Call `init`. If your processor is in an incorrect state and won't halt, then the examination (which is part of `init`) will of course fail.
2) After the failed init, you can try to revive your target using `dmi_write` & `dmi_read`.
3) When ready, you can re-examine the target by `<target_name> arp_examine`.

Alan-19950616, 2022-12-12:
[image]
[image]

I did what you asked, **arp_examine** does not appear to be successful. 
But **init resethalt** does.

timsifive, 2022-12-12:
You might have to disable/reenable polling before/after your dmi_write sequence. Otherwise poll() will write to dmcontrol, and clear resethaltreq before you get a chance to issue reset.

Another possible fix is to change reset so that it can execute even when examine() failed.

Alan-19950616, 2022-12-13:
Hi, timsifive
thank you for your time.

> You might have to disable/reenable polling before/after your dmi_write sequence. Otherwise poll() will write to dmcontrol, and clear resethaltreq before you get a chance to issue reset.

The following figure shows that the CPU has been successfully reset and halted.
[image]

I did what you said, and it's still the same.
[image]
[image]

> Another possible fix is to change reset so that it can execute even when examine() failed.

What should I do?

Alan-19950616, 2023-09-26:
Well, in that case then this feature is fine for me to maintain locally.

closed 2023-09-26 by Alan-19950616