riscv-collab/riscv-openocd #779: [Question]: Accessing two DMs with OpenOCD

sajjadahmed677, 2022-12-22:
Hi all,

I have question about running OpenOCD with a system which has two debug modules over same DMI. How i can access or address two debug modules with OpenOCD. is there specific configuration commands for doing this? also same question for GDB?

JanMatCodasip, 2022-12-23:
Hi @sajjadahmed677

As for OpenOCD -

Are you referring to multiple debug modules in the DMI address range, on certain address offsets, as indicated by the`nextdm` register? This is unfortunately not yet supported by `riscv-openocd`.

As for GDB - 

There are two options:
- a) debug the targets independently by two separate GDB instances
- b) debug the targets by one GDB instance as multiple threads (use SMP + hwthread mode in OpenOCD)
  - the seems to be only little documentation about this approach - see [1] and [2]
  - example configuration file [3]

[1] https://openocd.org/doc-release/html/Config-File-Guidelines.html#Define-CPU-targets-working-in-SMP
[2] https://openocd.org/doc-release/html/GDB-and-OpenOCD.html#Using-OpenOCD-SMP-with-GDB
[3] https://github.com/riscv-software-src/riscv-tests/blob/master/debug/targets/RISC-V/spike-2-hwthread.cfg

sajjadahmed677, 2022-12-24:
Thanks for the response @JanMatCodasip

closed 2022-12-24 by sajjadahmed677