riscv-collab/riscv-openocd #784: [Question]: Support for SPI based DTM

sajjadahmed677, 2023-01-05:
Hi,

I am working on project where i need to access a hardware debug module(based on RISCV) through both JTAG and SPI slave interfaces using OpenOCD. I am clear on JTAG interface and done pretty much debugging using JTAG as transport.
But i haven't found any example or reference where SPI is used as transport for debugging. Does ```riscv-openocd``` support access to DM over SPI transport?

TommyMurphyTM1234, 2023-01-05:
I'm pretty sure that neither upstream OpenOCD nor this fork has support for debugging over SPI. As opposed to SPI support for, say, programming flash - e.g.:

* https://github.com/riscv/riscv-openocd/blob/riscv/src/flash/nor/fespi.c
* https://github.com/riscv/riscv-openocd/blob/riscv/src/flash/nor/spi.c
* https://github.com/riscv/riscv-openocd/blob/riscv/src/flash/nor/stmqspi.c

There's no sign of any SPI based debug interface here:

* https://github.com/riscv/riscv-openocd/tree/riscv/tcl/interface

Can you share any details of the RISC-V debug implemention that allows debugging over SPI or otherwise summarise how this would work?

JanMatCodasip, 2023-01-06:
Hi @sajjadahmed677, the RISC-V Debug Spec currently includes the JTAG as the debug transport (DTM). See [riscv-debug-stable.pdf](https://raw.githubusercontent.com/riscv/riscv-debug-spec/master/riscv-debug-stable.pdf), page 92. 

If you'd like to use a different interface to access the debug module's registers (e.g. SPI), that is certainly possible and the spec allows for it (see "custom DTM", page 92). But it would be a custom transport protocol and you would have to implement the support in both hardware and software (in OpenOCD) yourself.

timsifive, 2023-01-06:
If you are building a system that uses SPI as a RISC-V debug interface, the RISC-V Debug group would love to review what you're doing, so something can be standardized in the spec.

sajjadahmed677, 2023-01-15:
Thanks all for your response,
@timsifive the feature is in discussion right now. i would love to share the details once finalized.

TommyMurphyTM1234, 2023-01-16:
> the feature is in discussion right now. 

So this is inaccurate then?

> I am working on project where i need to access a hardware debug module(based on RISCV) through both JTAG and SPI slave interfaces using OpenOCD.

TommyMurphyTM1234, 2023-03-27:
Any update @sajjadahmed677?

en-sc, 2024-01-22:
@sajjadahmed677, @TommyMurphyTM1234, I would like to close this issue, since the question was answered and it is not really OpenOCD-related.

TommyMurphyTM1234, 2024-01-22:
I'm happy for it to be closed. 👍

closed 2024-01-26 by en-sc