riscv-collab/riscv-openocd #785: [question/feature] facilities to hide some CSRs from the user

aap-sc, 2023-01-10:
Hi!

Currently, OpenOCD exposes most of the CSRs defined by the riscv-privileged spec.

This results in some confusion or minor usability issues for the user:

1. Not every CSR is guarded by an extension. An example is `sedeleg` register. The register is deprecated (that is, the latest spec does not have it), however, the encoding still has its definition (for compatibility reasons, I guess). Since we write such registers to the gdb target .xml this leads to some confusion because gdb can query the state of non-existent registers (like when we do `info all-registers` command)

2. When we debug bare-metal applications often we don't need access to registers like CSR_MHPMCOUNTER* or manual access to trigger registers. We could even use IDE like Eclipse of VisualStudio. When the debugger is in control, these IDEs display all CSRs and attempt to retrieve their values. These activities slow down debugging and, in some cases, may cause instability.

On the other hand, OpenOCD has a concept of `hidden` registers - these are registers  not listed by `reg` command and which are not included in the target.xml sent to GDB. To avoid the above issues, I'd like to have the ability to explicitly mark CSRs as hidden ones. In fact, I have a patch that implements this.

However, I'm not sure if there are any objections regarding inclusion of such functionality in the upstream. Thus the question.

TommyMurphyTM1234, 2023-01-10:
There are a few older issues logged that might also be relevant here?

* https://github.com/riscv/riscv-openocd/issues/245
* https://github.com/riscv/riscv-openocd/issues/175
* https://github.com/riscv/riscv-openocd/issues/136

And some other possibly relevant content:

* https://stackoverflow.com/questions/72759791/why-does-gdb-do-not-show-all-risc-v-csrs-when-debugging-bare-metal-program-runni
* https://sourceware.org/pipermail/gdb-patches/2020-June/169563.html
* https://github.com/eclipse-embed-cdt/eclipse-plugins/issues/345

I'm just posting these links because, from past experience, changes in this context can lead to unintended consequences in all sorts of related tools.

aap-sc, 2023-01-10:
@TommyMurphyTM1234 , from the above list the most relevant is #175, I think.

I've decided to create a PR (#787), since changes are relatively small. The idea is to allow user/vendor to decide which CSRs are exposed during the target configuration stage.

TommyMurphyTM1234, 2023-01-10:
Thanks @aap-sc - just to clarify, my previous post wasn't meant as any sort of objection to what you propose - doing something about the potentially huge RISC-V register list(s) (internet, floating point, CSRs) is probably a good thing. As far as I recall, some work was done on this at different times (by @timsifive, @mwachs5 I think, and maybe others). But I'm but sure if any holistic and systematic work was done on it exploiting all of the relevant capabilities of the various tools (OpenOCD, gdb, lldb?, Remote Serial Protocol, XML "manifests" etc.). I could be wrong, but it struck me as more piecemeal and reactive work as demanded by issues that arose with various tools. However, the point of my previous post was to highlight the fact that changes in this context potentially impact many other tools so careful analysis, implemention, and testing may be required in order to avoid unintended consequences and problems. Hope that makes sense! 🙂

timsifive, 2023-01-11:
Personally, I think that OpenOCD should expose all 4096 CSRs, since there is no way to know for OpenOCD which ones are actually implemented. It is up to a user interface to display the ones that are appropriate. Early versions did this. However, if you expose that many registers to Eclipse then it would crash, which is why it has the current behavior.

Adding features to OpenOCD to fix problems in gdb feels like the wrong approach. Despite this, a lot of code in OpenOCD is dedicated to make the world look the way gdb expects it to. But for the specific case of which registers to show, that seems like something that is under gdb's control.

aap-sc, 2023-01-11:
> But for the specific case of which registers to show, that seems like something that is under gdb's control.

Regarding this one. My understanding is that the set of registers to display is defined by target desctiption xml sent by remote target. In our case - the exact set should be under OpenOCD and target configuration control.

> Personally, I think that OpenOCD should expose all 4096 CSRs, since there is no way to know for OpenOCD which ones are actually implemented.

I agree, but the OpenOCD user could be aware of the actual HW configuration. And the suggestion is to allow user to exclude some registers using target configuration (in a manner similar to how expose_csr works). See #787 : no register is hidden unless user explicitly told to do so.

TommyMurphyTM1234, 2023-01-12:
> See #787 : no register is hidden unless user explicitly told to do so.

If there is any likelihood that this PR will be accepted/merged then I would like to review it but it may be a bit later today or tomorrow.

timsifive, 2023-01-12:
The change is small enough, and I'm not completely convinced this should be gdb's job, so I'm happy letting this functionality in.

aap-sc, 2023-02-27:
#787  merged, so the issue can be closed.

closed 2023-02-27 by aap-sc