riscv-collab/riscv-openocd #788: [bug] looks like the current implementation of breakpoints/watchpoints is broken (for some implementatins)

aap-sc, 2023-01-13:
In fact, this is what we observe on the current OpenOCD TOT with our (Syntacore) designs.

I've investigated the observed failures and I belive a have an understing of the problem. Please, see below.

A while ago there was a change in how OpenOCD processes configuration of the triggers #725. More specifically, the region of intrest is this commit: https://github.com/riscv/riscv-openocd/pull/725/commits/5a5c5e02ae6aa3c455cb7fe7881fa83e0ef4ba36 file riscv.c:627 and below.

Previously, we had the following code:

```
	for (i = 0; i < r->trigger_count; i++) {
		if (r->trigger_unique_id[i] != -1)
			continue;

		riscv_set_register(target, GDB_REGNO_TSELECT, i);

		uint64_t tdata1;
		int result = riscv_get_register(target, &tdata1, GDB_REGNO_TDATA1);
		if (result != ERROR_OK)
			return result;
		int type = get_field(tdata1, CSR_TDATA1_TYPE(riscv_xlen(target)));

		result = ERROR_OK;
		switch (type) {
			case 1:
				result = maybe_add_trigger_t1(target, trigger, tdata1);
				break;
			case 2:
				result = maybe_add_trigger_t2(target, trigger, tdata1);
				break;
			case 6:
				result = maybe_add_trigger_t6(target, trigger, tdata1);
				break;
			default:
				LOG_DEBUG("trigger %d has unknown type %d", i, type);
				continue;
		}
               // continued
```

That is we iterated over each available trigger `for (i = 0; i < r->trigger_count; i++) {` in an attempt to find one suitable for the requested configuration.

The loop was removed and now code looks as follows:

```
 do {                                                                                                         
   ret = maybe_add_trigger_t1(target, trigger);                                                               
   if (ret == ERROR_OK)                                                                                       
     break;                                                                                                   
   ret = maybe_add_trigger_t2(target, trigger);                                                               
   if (ret == ERROR_OK)                                                                                       
     break;                                                                                                   
   ret = maybe_add_trigger_t6(target, trigger);                                                               
   if (ret == ERROR_OK)                                                                                       
     break;                                                                                                   
  } while (0);
```

And now we expect that the respected `maybe_add_trigger_t*` should find a suitable trigger. To me it seems that the underlying implementation is **not** correct. Looks like the current OpenOCD codebase assumes that all triggers of the same type are equivalent. Meaning if we, say, have two triggers of type "2" then there is an assumption that they all provide the same functionality. 

Looks like @wxjstz  mentioned this in his comment in the respected review: 
https://github.com/riscv/riscv-openocd/pull/725/commits/5a5c5e02ae6aa3c455cb7fe7881fa83e0ef4ba36#r979250834

> The new patch adds type records in riscv_enumerate_triggers, so finding triggers no longer requires writing tselect and reading tdata1

However, I don't believe that the debug spec implies that. I believe that the following implementation is allowed:

**Example:**

the system can have 2 triggers of MCONTROL type:

- TRIGGER 0 (type2): supports only breakpoints (that is allows only tdata.execute) 
- TRIGGER 1 (type2): supports only watchpints (that is allows only tdata.(store|load) )

With the current implementation, it is not possible to set a watchpoint for such a HW implementation. The reason is that the respected `maybe_add_trigger_t2` does not iterate over the set of available triggers, but just selects the first available (which will be the one that supports only breakpoints).

Previous implementation worked fine.

@timsifive , @wxjstz would you kindly take a look at the description of the problem - does it make sence to you? Or do you believe that the described HW implementation somehow contradicts the debug specification?

timsifive, 2023-01-17:
You are right on both counts:
1. The debug spec allows triggers like you suggest.
2. OpenOCD doesn't currently handle that.

Modifying maybe_add_trigger_t* to iterate over triggers seems like a reasonable solution.

aap-sc, 2023-03-17:
fixed by https://github.com/riscv/riscv-openocd/pull/800

closed 2023-03-17 by aap-sc