riscv-collab/riscv-openocd #789: cp: cannot stat 'libgcc_s_sjlj-1.dll': No such file or directory

Cherish-forever, 2023-01-13:
I am building riscv-openocd in my local by using act
`act -j package -P ubuntu-18.04=-self-hosted`

I have add "-static-libgcc -static-libstdc++" but no use, who can help me?

TommyMurphyTM1234, 2023-01-13:
Can you post (attach, not inline) a full build log, please?

Cherish-forever, 2023-01-13:
@TommyMurphyTM1234 
In build time
I am not in docker to build, I am using act to run in my local PC, using docker takes too lang time, because GFW.
my PC is ubuntu 22.04, i686-w64-mingw32-gcc version is 10-win32 20220113.

TommyMurphyTM1234, 2023-01-13:
You haven't posted the full build log.
Maybe the DLL is simply not available at build time?

Cherish-forever, 2023-01-13:
@TommyMurphyTM1234  
I am using act to run github workflow yaml
https://github.com/nektos/act

here is my log:
|  /usr/bin/mkdir -p '/home/weig/.cache/act/87a8fe6f73051261/build/openocd_dde802bf8_i686-w64-mingw32/usr/share/man/man1'
|  /usr/bin/install -c -m 644 /home/weig/.cache/act/87a8fe6f73051261/hostexecutor/doc/openocd.1 '/home/weig/.cache/act/87a8fe6f73051261/build/openocd_dde802bf8_i686-w64-mingw32/usr/share/man/man1'
|  /usr/bin/mkdir -p '/home/weig/.cache/act/87a8fe6f73051261/build/openocd_dde802bf8_i686-w64-mingw32/usr/share/openocd'
|  /usr/bin/mkdir -p '/home/weig/.cache/act/87a8fe6f73051261/build/openocd_dde802bf8_i686-w64-mingw32/usr/share/openocd/contrib'
|  /usr/bin/install -c -m 644  /home/weig/.cache/act/87a8fe6f73051261/hostexecutor/contrib/60-openocd.rules '/home/weig/.cache/act/87a8fe6f73051261/build/openocd_dde802bf8_i686-w64-mingw32/usr/share/openocd/contrib'
|  /usr/bin/mkdir -p '/home/weig/.cache/act/87a8fe6f73051261/build/openocd_dde802bf8_i686-w64-mingw32/usr/share/openocd/contrib/libdcc'
|  /usr/bin/install -c -m 644  /home/weig/.cache/act/87a8fe6f73051261/hostexecutor/contrib/libdcc/dcc_stdio.c /home/weig/.cache/act/87a8fe6f73051261/hostexecutor/contrib/libdcc/dcc_stdio.h /home/weig/.cache/act/87a8fe6f73051261/hostexecutor/contrib/libdcc/example.c /home/weig/.cache/act/87a8fe6f73051261/hostexecutor/contrib/libdcc/README '/home/weig/.cache/act/87a8fe6f73051261/build/openocd_dde802bf8_i686-w64-mingw32/usr/share/openocd/contrib/libdcc'
| make  install-data-hook
| make[5]: Entering directory '/home/weig/.cache/act/87a8fe6f73051261/build/i686-w64-mingw32-build/openocd'
| for i in $(find /home/weig/.cache/act/87a8fe6f73051261/hostexecutor/tcl -name '*.cfg' -o -name '*.tcl' -o -name '*.txt' | sed -e 's,^/home/weig/.cache/act/87a8fe6f73051261/hostexecutor/tcl,,'); do \
| 	j="/home/weig/.cache/act/87a8fe6f73051261/build/openocd_dde802bf8_i686-w64-mingw32/usr/share/openocd/scripts/$i" && \
| 	mkdir -p "$(dirname $j)" && \
| 	/usr/bin/install -c -m 644 /home/weig/.cache/act/87a8fe6f73051261/hostexecutor/tcl/$i $j; \
| done
| make[5]: Leaving directory '/home/weig/.cache/act/87a8fe6f73051261/build/i686-w64-mingw32-build/openocd'
| make[4]: Leaving directory '/home/weig/.cache/act/87a8fe6f73051261/build/i686-w64-mingw32-build/openocd'
| make[3]: Leaving directory '/home/weig/.cache/act/87a8fe6f73051261/build/i686-w64-mingw32-build/openocd'
| make[2]: Leaving directory '/home/weig/.cache/act/87a8fe6f73051261/build/i686-w64-mingw32-build/openocd'
| make[1]: Leaving directory '/home/weig/.cache/act/87a8fe6f73051261/build/i686-w64-mingw32-build/openocd'
| cp: cannot stat 'libgcc_s_sjlj-1.dll': No such file or directory
[OpenOCD Snapshot/package]   ❌  Failure - Main Package OpenOCD for windows
[OpenOCD Snapshot/package] exit status 1
[OpenOCD Snapshot/package] 🏁  Job failed
Error: Job 'package' failed

Cherish-forever, 2023-01-13:
@TommyMurphyTM1234 
I found in /usr/x86_64-w64-mingw32/lib on ubuntu 22.04 and i686-w64-mingw32-gcc version is 10-win32 20220113.
there is no libgcc_s_sjlj-1.dll but it have libwinpthread-1.dll, so I think maybe it relate with i686-w64-mingw32-gcc version.

Cherish-forever, 2023-01-13:
@TommyMurphyTM1234 
I have tested, in ubuntu 18.04, it run no error.

TommyMurphyTM1234, 2023-01-13:
Again you haven't posted the full build log or heeded what I asked about the issue occurring with the upstream OpenOCD sources.

Almost certainly this is a general MinGW (cross) building issue and nothing to do with RISC-V, or maybe even OpenOCD, at all. You should probably be asking about this in a MinGW forum.

TommyMurphyTM1234, 2023-01-13:
> @TommyMurphyTM1234 I found in /usr/x86_64-w64-mingw32/lib on ubuntu 22.04 and i686-w64-mingw32-gcc version is 10-win32 20220113. there is no libgcc_s_sjlj-1.dll 

So the DLL is missing.
That's your problem.
Nothing to do with RISC-V, RISC-V OpenOCD, or OpenOCD at all.

Cherish-forever, 2023-01-14:
> 

Yes, that is my problem.

closed 2023-01-14 by Cherish-forever

lz-bro, 2023-12-14:
I have also encountered this problem. Did you solved it ？

TommyMurphyTM1234, 2023-12-14:
> I have also encountered this problem. Did you solved it ？

You need to find or build any DLL dependencies for OpenOCD yourself. 

Might be worth checking @ilg-ul's xPack approach to building OpenOCD as mentioned here:

* https://github.com/riscv/riscv-openocd/issues/789#issuecomment-1382311887