riscv-collab/riscv-openocd #790: Alignment with riscv-org repository

natirapaport, 2023-01-17:
Can anyone tell how much riscv-openocd repo, is aligned with official [openocd-org/openocd](https://github.com/openocd-org/openocd) repository?
The latest riscv-openocd release is from Dec-22, 2018 , but on the other hand the version of riscv-openocd is 0.11 which is very close to openocd-org/openocd latest version (0.12) .
From here, I couldn’t tell where is the latest pull-request from riscv-openocd fork to openocd-org/openocd repo. Can someone please point at that commit?
Thanks,
Nati

TommyMurphyTM1234, 2023-01-17:
For what it's worth, I think that this is the last time that this repo was sync'd with the upstream master upstream OpenOCD.

* https://github.com/riscv/riscv-openocd/pull/696

This happens every so often and changes here are upstreamef periodically too.

You could also always clone both repos and diff them or use git to check for differences.

oferShinaar, 2023-01-17:
@timsifive this repo is contributing back to the openocd repo? if so, can you share the schedule points?
On the other hand, if all riscv-openocd is in this repo what is the solid-working branch (no tags or releases I saw)?

thanks

timsifive, 2023-01-17:
> @timsifive this repo is contributing back to the openocd repo? if so, can you share the schedule points?

Yes, very slowly, when I have time. I'd love help with this. There is no schedule.

> On the other hand, if all riscv-openocd is in this repo what is the solid-working branch (no tags or releases I saw)?

master in this branch should always be working well. There are no official tags or releases.

natirapaport, 2023-01-18:
@timsifive - do you know to tell if riscv-openocd pass the same , or similar testing procedures like the openocd-org one? In other words, how much it is considered to be reliable comparing to openocd-org?

timsifive, 2023-01-18:
I'm not aware of any systematic testing done by upstream OpenOCD. Since riscv-openocd is behind mainline, there are almost certainly bugfixes there that are not here.

timsifive, 2023-01-18:
Multi-hart cores is a very common use case, so I would expect it to work well. If there are problems, feel free to open an issue and include the `openocd -d` log.

TommyMurphyTM1234, 2023-03-21:
Just noting this PR here for reference purposes:

* https://github.com/riscv/riscv-openocd/pull/816

TommyMurphyTM1234, 2023-03-28:
Hi @timsifive - quick question which, offhand, I don't see mentioned elsewhere.... Do you have any feel for how far off the day might be that further RISC-V development/enhancement can be done directly upstream obviating the need for a RISC-V fork of OpenOCD? Or, maybe put another way, what are the main big tasks outstanding before this might be feasible? Thanks.

timsifive, 2023-03-29:
At the current pace, this will never happen.

The branches are drifting further apart, and I don't have the bandwidth to bring them together. The process to bring them together is pretty straightforward:
1. Merge changes from upstream into this branch.
2. Merge changes from this branch into upstream.

TommyMurphyTM1234, 2023-03-29:
Thanks @timsifive. I fully understand that you (and others who contribute) have limited bandwidth and/or too much to do.

Just for my own edification, is there anywhere any outline of stuff that remains to be implemented? Including anything that arises from further revisions of the debug (and trace?!) spec(s)? 

Apologies for all the questions. I hope that they're ok and that this is an appropriate forum?

timsifive, 2023-03-30:
There is no todo list. OpenOCD development tends to be done as-needed. Lately the amount of work done in this branch is actually increasing as more people are using RISC-V chips and OpenOCD. I think it's unlikely this pace will slow in the foreseeable future.

TommyMurphyTM1234, 2023-03-30:
> There is no todo list. OpenOCD development tends to be done as-needed. Lately the amount of work done in this branch is actually increasing as more people are using RISC-V chips and OpenOCD. I think it's unlikely this pace will slow in the foreseeable future.

Thanks @timsifive - that's fine. I was just wondering if there was any sort of outline of a schedule/task list. No worries if there's not.

aap-sc, 2023-03-30:
Hi @timsifive  would you kindly clarify on this one:

> The branches are drifting further apart, and I don't have the bandwidth to bring them together. The process to bring them together is pretty straightforward:
> Merge changes from upstream into this branch.
> Merge changes from this branch into upstream.

Is the issue only with this process (likely, both merges require significant amount of time)? Or there are some other issues preventing riscv-openocd development to move to the mainline?

timsifive, 2023-03-30:
The issue is only with the process. There are no legal or technical reasons the branches cannot be brought together.

zqb-all, 2023-04-04:
> Merge changes from upstream into this branch.
> Merge changes from this branch into upstream.

If someone helps with this, what tests should be done after the merge to make sure the merge didn't introduce any problems？

timsifive, 2023-04-04:
> If someone helps with this, what tests should be done after the merge to make sure the merge didn't introduce any problems?

I rely on https://github.com/riscv-software-src/riscv-tests/tree/master/debug. I run `make all` there. Then I also run the tests against HiFive Unleashed, and if riscv-011 has changed, HiFive1 as well. For the hardware platforms some of the tests fail. Unfortunately that's not documented anywhere.

mrv96, 2023-05-02:
I'm waiting too about upstream merging. Is someone working on this?

mrv96, 2023-05-02:
Thank you for the quick reply.

I'm already a contributor of this repository... however upstream merge is quite tricky and potentially harmful, hence is better if done by a maintainer of the repository.

Despite this, I was willing to try by myself in case of negative answer to my question

timsifive, 2023-05-02:
I'm working on this here and there. My general process is:
1. Merge down changes from upstream. We are considerably behind. #844 would make us less behind. It's waiting https://github.com/riscv-software-src/riscv-tests/pull/469 before we can merge it.
2. Upstream changes.

We don't *have* to do things in that order, and any upstream change (however small) the reduces the diff between the two branches would be helpful.