riscv-collab/riscv-openocd #791: unable to build openocd

vignajeth, 2023-01-17:
Hi,
    i am trying to install riscv-openocd in my windows computer using cygwin.

  i was able to successfully run the following steps

 ```
git clone https://github.com/riscv/riscv-openocd
cd riscv-openocd
./bootstrap
./configure --enable-jtag_vpi --enable-ftdi
```

However, when i run `make`, i get this following error

```
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I./src -I./src -DPKGDATADIR=\"/usr/local/share/openocd\" -DBINDIR=\"/usr/local/bin\" -DFD_SETSIZE=128 -I./jimtcl -I./jimtcl -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Wundef -Wno-error=deprecated-declarations -Werror -DFD_SETSIZE=128 -g -O2 -MT src/server/telnet_server.lo -MD -MP -MF src/server/.deps/telnet_server.Tpo -c src/server/telnet_server.c -o src/server/telnet_server.o
In file included from ./src/helper/system.h:38,
                 from ./src/helper/replacements.h:29,
                 from ./src/server/server.h:33,
                 from src/server/telnet_server.h:28,
                 from src/server/telnet_server.c:29:
src/server/telnet_server.c: In function 'telnet_auto_complete':
src/server/telnet_server.c:624:73: error: array subscript has type 'char' [-Werror=char-subscripts]
  624 |         while ((usr_cmd_pos < t_con->line_cursor) && isspace(t_con->line[usr_cmd_pos]))
      |                                                              ~~~~~~~~~~~^~~~~~~~~~~~~
src/server/telnet_server.c:635:65: error: array subscript has type 'char' [-Werror=char-subscripts]
  635 |                 if ((i < usr_cmd_len - 1) && isspace(t_con->line[usr_cmd_pos + i])
      |                                                      ~~~~~~~~~~~^~~~~~~~~~~~~~~~~
src/server/telnet_server.c:636:55: error: array subscript has type 'char' [-Werror=char-subscripts]
  636 |                                 && isspace(t_con->line[usr_cmd_pos + i + 1])) {
      |                                            ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [Makefile:3176: src/server/telnet_server.lo] Error 1
make[2]: Leaving directory '/homedir/riscv-openocd'
make[1]: *** [Makefile:4568: all-recursive] Error 1
make[1]: Leaving directory '/homedir/riscv-ope
```

i would be happy if someone can help me here ;)

vignajeth, 2023-01-17:
some of the checks done by ./configure 

```

TommyMurphyTM1234, 2023-01-17:
In my experience, building OpenOCD on Windows is a pain and it's easier to cross compile it from Linux. In particular, the xPack Project approach makes it more predictable.

* https://xpack.github.io/openocd/
* https://xpack.github.io/openocd/#build-details

By default, it builds from its own tarballs, but you can change the following file to get it to build from this repo instead by entering the appropriate repo URL, branch name and commit.

* https://github.com/xpack-dev-tools/openocd-xpack/blob/4233b2eb4d6571ad04f3950b8d0e037b9784beb6/scripts/defs-source.sh#L40

vignajeth, 2023-01-17:
Thanks for your speedy reply

i was able to successfully install openocd by following your instructions

closed 2023-01-17 by vignajeth