riscv-collab/riscv-openocd #808: Error Building Example for RiscV SiFive E31

hparpinel, 2023-03-03:
Hi, 
I wasn't sure where to post this. I am trying to run the SiFive E31 Arty example in eclipse but I am getting the error below.
make all 
Building file: ../xpacks/micro-os-plus-semihosting/src/syscalls-semihosting.cpp
Invoking: GNU RISC-V Cross C++ Compiler
riscv-none-elf-g++ -march=rv32imac -mabi=ilp32 -mcmodel=medany -msmall-data-limit=8 -mno-save-restore -Og -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -fno-move-loop-invariants -Wall -Wextra -g3 -DDEBUG -DOS_USE_SEMIHOSTING_SYSCALLS -DOS_USE_CPP_INTERRUPTS -DTRACE -DOS_USE_TRACE_SEMIHOSTING_DEBUG -DSIFIVE_E31ARTY -DSIFIVE_E31ARTY_BOARD -I"../include" -I"../xpacks/micro-os-plus-diag-trace/include" -I"../xpacks/micro-os-plus-semihosting/include" -I"../xpacks/sifive-devices/include" -I"../xpacks/sifive-arty-boards/include" -I"../xpacks/micro-os-plus-c-libs/include" -I"../xpacks/micro-os-plus-riscv-arch/include" -I"../xpacks/micro-os-plus-startup/include" -std=gnu++14 -fabi-version=0 -fno-exceptions -fno-rtti -fno-use-cxa-atexit -fno-threadsafe-statics -MMD -MP -MF"xpacks/micro-os-plus-semihosting/src/syscalls-semihosting.d" -MT"xpacks/micro-os-plus-semihosting/src/syscalls-semihosting.o" -c -o "xpacks/micro-os-plus-semihosting/src/syscalls-semihosting.o" "../xpacks/micro-os-plus-semihosting/src/syscalls-semihosting.cpp"
../xpacks/micro-os-plus-semihosting/src/syscalls-semihosting.cpp:1569:3: error: 'int select(int, fd_set*, fd_set*, fd_set*, timeval*)' aliased to undefined symbol '_ZN5posix6selectEiP13_types_fd_setS1_S1_P7timeval'
 1569 |   select (int nfds, fd_set* readfds, fd_set* writefds, fd_set* errorfds,
      |   ^~~~~~
make: *** [xpacks/micro-os-plus-semihosting/src/subdir.mk:23: xpacks/micro-os-plus-semihosting/src/syscalls-semihosting.o] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

TommyMurphyTM1234, 2023-03-03:
This has absolutely nothing to do with OpenOCD so it seems bizarre that you posted about it here. You would be better off posting in the SiFive forums.

* https://forums.sifive.com/

TommyMurphyTM1234, 2023-03-15:
Candidate for closure?

closed 2023-03-15 by timsifive