riscv-collab/riscv-openocd #813: Error while setting 4th watchpoint

neanderthal148, 2023-03-15:
Hi,
I am getting an error when I try to set the 4th watchpoint using the wp command. I am able to set 4 breakpoints or a combination of 4 breakpoints and watchpoints but for some reason I cannot set 4 watchpoints.

[image]

JanMatCodasip, 2023-03-15:
Hi @neanderthal148, could you please provide us:

- commit ID from which you built your copy of `riscv-openocd`
- verbose debug log (start `openocd` with argument `-d3`)

Thank you!

neanderthal148, 2023-03-16:
Commit hash: d6bf022560d54f83bb7d73d5b1b9ae78cb5eaf11

[wp.debug.log](https://github.com/riscv/riscv-openocd/files/10989471/wp.debug.log)

TommyMurphyTM1234, 2023-03-16:
Also, what is the nature and provenance of the RISC-V implementation in your debug target?

TommyMurphyTM1234, 2023-03-16:
I've tried this using Spike as the debug target and can't reproduce the problem.
```
tommy@kingfisher:~/Downloads/sandbox$ telnet localhost 4444
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
> wp 0x00 4 a
[riscv.cpu] Found 4 triggers
> wp 0x04 4 a
> wp 0x08 4 a
> wp 0x0c 4 a
> wp 0x10 4 a
can't add access watchpoint at 0x00000010, unrecognized error
Failure setting watchpoints

> 
```

TommyMurphyTM1234, 2023-03-16:
In your log, the error arises here:
```
Error: 17293 239286 breakpoints.c:457 watchpoint_add_internal(): can't add access watchpoint at 0x00000020, unrecognized error
Error: 17294 239286 target.c:4160 handle_wp_command(): Failure setting watchpoints
Debug: 17295 239286 command.c:556 run_command(): Command 'wp' failed with error code -4
```
But I can't explain (or reproduce!) this so far.

timsifive, 2023-03-17:
You encounter a message like this one several times:
```
Debug: 8730 148571 riscv.c:523 set_trigger(): Trigger doesn't support what we need; After writing 0x28000000000010db to tdata1 it contains 0x280000000000105b; tdata1_ignore_mask=0x7e0000000000000
```

I don't see that message anywhere in the current source. Can you try building the latest riscv-openocd and share a log from that?

TommyMurphyTM1234, 2023-03-26:
Any update @neanderthal148?

neanderthal148, 2023-03-27:
Hi guys, thanks for the help. Issue seems to be solved after building the latest openocd

closed 2023-03-28 by neanderthal148