riscv-collab/riscv-openocd #814: OPENOCD command: read_memory $address 64 $count, displaying 32-bit values with count=1

neanderthal148, 2023-03-15:
Hi, 
   Whenever I try to use read_memory 64, I only get 32 bit output (data1) on my terminal for both abstract and sysbus access. This issue is not present while reading 64 bit using mdd.

[image]

TommyMurphyTM1234, 2023-03-15:
Please post (attach) a verbose `openocd -d3` log for the operation in question.

neanderthal148, 2023-03-16:
Attached the log 
[readmemory.debug.log](https://github.com/riscv/riscv-openocd/files/10989464/readmemory.debug.log)

TommyMurphyTM1234, 2023-03-16:
OK - looks like the issue is here:

* https://github.com/riscv/riscv-openocd/blob/riscv/src/target/target.c#L4730

and fixed in OpenOCD upstream here:

* https://github.com/openocd-org/openocd/blob/master/src/target/target.c#L4726

with this patch:

* https://sourceforge.net/p/openocd/mailman/message/37710818/

which has not yet been integrated downstream (here).

TommyMurphyTM1234, 2023-03-20:
Confirmed fixed by https://github.com/riscv/riscv-openocd/pull/816 so this issue can be closed.
```
$ telnet localhost 4444
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
> poll
background polling: on
TAP: riscv.cpu (enabled)

> mdw 0x10000
0x00010000: 464c457f 

> mdw 0x10004
0x00010004: 00010102 

> mdd 0x10000
0x00010000: 00010102464c457f 

> read_memory 0x10000 64 1
0x10102464c457f
> 
```

closed 2023-03-29 by neanderthal148