riscv-collab/riscv-openocd #817: remote bitbang failing on halt command and Wrong IR capture

ElectronicBatman, 2023-03-20:
I am running remote bitbang server with RTL simulation(CVA6 ariane core). When I try to connect openocd I am getting following error:
```
Error: 222 127573 core.c:1242 jtag_validate_ircapture(): riscv.cpu: IR capture error; saw 0x0a not 0x01
Debug: 223 127573 core.c:327 jtag_call_event_callbacks(): jtag event: TAP reset
Warn : 224 127573 core.c:1462 jtag_init_inner(): Bypassing JTAG setup events due to errors
Debug: 225 127573 command.c:143 script_debug(): command - dap dap init
Debug: 226 127573 log.c:409 keep_alive(): keep_alive() was not invoked in the 1000ms timelimit (127573). This may cause trouble with GDB connections.
Debug: 229 127573 arm_dap.c:105 dap_init_all(): Initializing all DAPs ...
Debug: 230 127573 openocd.c:158 handle_init_command(): Examining targets...
Debug: 231 127573 target.c:1616 target_call_event_callbacks(): target event 17 (examine-start) for core riscv.cpu
Debug: 232 127573 riscv.c:1015 riscv_examine(): riscv_examine()
Debug: 233 136596 riscv.c:445 dtmcontrol_scan(): DTMCONTROL: 0x0 -> 0x20e2
Debug: 234 136596 riscv.c:1025 riscv_examine(): dtmcontrol=0x20e2
Debug: 235 136596 riscv.c:1027 riscv_examine():   version=0x2
Error: 236 136596 riscv.c:465 get_target_type(): Unsupported DTM version: 2
Debug: 237 136596 openocd.c:160 handle_init_command(): target examination failed
Debug: 238 136596 command.c:143 script_debug(): command - flash flash init
Debug: 239 136597 log.c:409 keep_alive(): keep_alive() was not invoked in the 1000ms timelimit (9024). This may cause trouble with GDB connections.
Debug: 242 136597 tcl.c:1222 handle_flash_init_command(): Initializing flash devices...
Debug: 243 136597 command.c:143 script_debug(): command - nand nand init
Debug: 245 136597 tcl.c:498 handle_nand_init_command(): Initializing NAND devices...
Debug: 246 136597 command.c:143 script_debug(): command - pld pld init
Debug: 248 136597 pld.c:206 handle_pld_init_command(): Initializing PLDs...
Debug: 249 136597 gdb_server.c:3503 gdb_target_start(): starting gdb server for riscv.cpu on 3333
Info : 250 136597 server.c:311 add_service(): Listening on port 3333 for gdb connections
Debug: 251 136597 command.c:143 script_debug(): command - halt halt
Debug: 253 136597 target.c:3070 handle_halt_command(): -
Error: 254 136597 target.c:567 target_halt(): Target not examined yet
Debug: 255 136597 command.c:630 run_command(): Command 'halt' failed with error code -4
```
The configuration file I use is as below:
```
interface remote_bitbang
remote_bitbang_host localhost
remote_bitbang_port 8088

set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 5

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME -coreid 0

gdb_report_data_abort enable

riscv set_reset_timeout_sec 120
riscv set_command_timeout_sec 120

init
halt
```
The DTMCONTROL as per RTL is:
```
 dtmcs_d  = '{
                      zero1        : '0,
                      dmihardreset : 1'b0,
                      dmireset     : 1'b0,
                      zero0        : '0,
                      idle         : 3'd1, // 1: Enter Run-Test/Idle and leave it immediately
                      dmistat      : dmi_error_i, // 0: No error, 2: Op failed, 3: too fast
                      abits        : 6'd7, // The size of address in dmi
                      version      : 4'd1  // Version described in spec version 0.13 (and later?)
                    };
```

TommyMurphyTM1234, 2023-03-20:
Based on the small amount of info provided, the following log message suggests that the issue may be the target debug block not working correctly as opposed to any issue with OpenOCD itself:

> ```
> Error: 222 127573 core.c:1242 jtag_validate_ircapture(): riscv.cpu: IR capture error; saw 0x0a not 0x01
> ```

Unless you can provide more compelling evidence as to why you think that this is an OpenOCD issue/bug then I suspect that this is a question better asked over at the CVA6 project:

* https://github.com/openhwgroup/cva6#debugging
* https://github.com/openhwgroup/cva6/issues

ElectronicBatman, 2023-03-21:
Sorry, I forgot to paste few lines from log:

```
Debug: 214 6 core.c:327 jtag_call_event_callbacks(): jtag event: TAP reset
Debug: 215 6 core.c:1060 jtag_examine_chain(): DR scan interrogation for IDCODE/BYPASS
Debug: 216 6 core.c:327 jtag_call_event_callbacks(): jtag event: TAP reset
Info : 217 136498 core.c:1110 jtag_examine_chain(): TAP riscv.cpu has invalid IDCODE (0x2)
Debug: 218 136498 core.c:1304 jtag_tap_init(): Created Tap: auto0.tap @ abs position 1, irlen 0, capture: 0x1 mask: 0x3
```

The problem I see here in cfg file I am giving irlen 5 but Debug message says irlen 0

TommyMurphyTM1234, 2023-03-21:
> Sorry, I forgot to paste few lines from log:
> 
> ```
> Debug: 214 6 core.c:327 jtag_call_event_callbacks(): jtag event: TAP reset
> Debug: 215 6 core.c:1060 jtag_examine_chain(): DR scan interrogation for IDCODE/BYPASS
> Debug: 216 6 core.c:327 jtag_call_event_callbacks(): jtag event: TAP reset
> Info : 217 136498 core.c:1110 jtag_examine_chain(): TAP riscv.cpu has invalid IDCODE (0x2)
> Debug: 218 136498 core.c:1304 jtag_tap_init(): Created Tap: auto0.tap @ abs position 1, irlen 0, capture: 0x1 mask: 0x3
> ```
> 
> The problem I see here in cfg file I am giving irlen 5 but Debug message says irlen 0 Attaching full log file.
> 
> [ocdlog.log](https://github.com/riscv/riscv-openocd/files/11025278/ocdlog.log)

That doesn't change my opinion that this looks more like an issue with the target's debug block rather than an issue with the RISC-V OpenOCD fork. Especially the fact that the IDCODE is coming can invalid.

TommyMurphyTM1234, 2023-03-21:
You also seem to be running and old OpenOCD:

> `Open On-Chip Debugger 0.10.0+dev-00849-gcbb15587d (2023-03-20-12:06)`

An up to date build from this repo gives this:

> `Open On-Chip Debugger 0.11.0+dev-02545-g3387015af-dirty (2023-03-20-14:23)`

You should probably try again with the latest RISC-V OpenOCD just in case you're hitting some old issue.

TommyMurphyTM1234, 2023-03-27:
Any update @ElectronicBatman?

ElectronicBatman, 2023-03-27:
@TommyMurphyTM1234  Issue is with debug module rather than openocd, the RTL is sending data (TDO)on posedge of TCK rather than negedge.

closed 2023-03-27 by ElectronicBatman