riscv-collab/riscv-openocd #827: OpenOCD Connection time out

hparpinel, 2023-03-27:
Hi,

I am using Eclipse CDT and trying to debug with GDB OpenOCD Debugging but I get ^error,msg="localhost:3333: Connection timed out."
The board I am working with is an Arty A7 100T. The board is programmed with the core; the green LED is on. 
My settings are below. What am I missing, do I suppose to start GDB somewhere else?
---MAIN
C/C++ Application: app.elf
---DEBUGGER
Start OpenOCD locally: -f board/arty.cfg -f interface/ftdi/digilent-hs1a.cfg
Start GDB Session: 
set mem inaccessible-by-default off
set $target_riscv = 1
set arch riscv:rv32
set remotetimeout 250

TommyMurphyTM1234, 2023-03-27:
You really need to attach a verbose OpenOCD log (`openocd -d3 ...`) 

* https://openocd.org/doc/html/Running.html

and a GDB Remote Serial Protocol log:

* https://sourceware.org/gdb/onlinedocs/gdb/Remote-Configuration.html

hparpinel, 2023-03-27:
I tried what you suggested from OpenOCD I get the output below but from GDB I dont get anything
xPack Open On-Chip Debugger 0.12.0-01004-g9ea7f3d64-dirty (2023-01-30-15:03)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 3 1 options.c:52 configuration_output_handler(): debug_level: 3
User : 4 1 options.c:52 configuration_output_handler(): 
Debug: 5 1 options.c:233 add_default_dirs(): bindir=/__w/openocd-xpack/openocd-xpack/build/linux-x64/application/bin
Debug: 6 1 options.c:234 add_default_dirs(): pkgdatadir=/__w/openocd-xpack/openocd-xpack/build/linux-x64/application/openocd
Debug: 7 1 options.c:235 add_default_dirs(): exepath=/home/bluespec/.local/xPacks/@xpack-dev-tools/openocd/0.12.0-1.1/.content/bin
Debug: 8 1 options.c:236 add_default_dirs(): bin2data=../openocd
Debug: 9 1 configuration.c:33 add_script_search_dir(): adding /home/bluespec/.config/openocd
Debug: 10 1 configuration.c:33 add_script_search_dir(): adding /home/bluespec/.openocd
Debug: 11 1 configuration.c:33 add_script_search_dir(): adding /home/bluespec/.local/xPacks/@xpack-dev-tools/openocd/0.12.0-1.1/.content/bin/../openocd/site
Debug: 12 1 configuration.c:33 add_script_search_dir(): adding /home/bluespec/.local/xPacks/@xpack-dev-tools/openocd/0.12.0-1.1/.content/bin/../openocd/scripts
Debug: 13 1 command.c:155 script_debug(): command - gdb_port 3333
Debug: 14 1 command.c:155 script_debug(): command - telnet_port 4444
Debug: 15 1 command.c:155 script_debug(): command - tcl_port 6666
Debug: 16 1 command.c:155 script_debug(): command - ocd_find board/digilent_arty.cfg
Debug: 17 1 configuration.c:88 find_file(): found /home/bluespec/.local/xPacks/@xpack-dev-tools/openocd/0.12.0-1.1/.content/bin/../openocd/scripts/board/digilent_arty.cfg
Debug: 18 1 command.c:155 script_debug(): command - echo DEPRECATED! use 'adapter driver' not 'interface'
User : 19 1 command.c:685 handle_echo(): DEPRECATED! use 'adapter driver' not 'interface'
Debug: 20 1 command.c:155 script_debug(): command - adapter driver ftdi
Debug: 21 1 command.c:155 script_debug(): command - echo DEPRECATED! use 'ftdi device_desc' not 'ftdi_device_desc'
User : 22 1 command.c:685 handle_echo(): DEPRECATED! use 'ftdi device_desc' not 'ftdi_device_desc'
Debug: 23 1 command.c:155 script_debug(): command - ftdi device_desc Digilent USB Device
Debug: 24 1 command.c:155 script_debug(): command - echo DEPRECATED! use 'ftdi vid_pid' not 'ftdi_vid_pid'
User : 25 1 command.c:685 handle_echo(): DEPRECATED! use 'ftdi vid_pid' not 'ftdi_vid_pid'
Debug: 26 1 command.c:155 script_debug(): command - ftdi vid_pid 0x0403 0x6010
Debug: 27 1 command.c:155 script_debug(): command - echo DEPRECATED! use 'ftdi channel' not 'ftdi_channel'
User : 28 1 command.c:685 handle_echo(): DEPRECATED! use 'ftdi channel' not 'ftdi_channel'
Debug: 29 1 command.c:155 script_debug(): command - ftdi channel 0
Debug: 30 1 command.c:155 script_debug(): command - echo DEPRECATED! use 'ftdi layout_init' not 'ftdi_layout_init'
User : 31 1 command.c:685 handle_echo(): DEPRECATED! use 'ftdi layout_init' not 'ftdi_layout_init'
Debug: 32 1 command.c:155 script_debug(): command - ftdi layout_init 0x0088 0x008b
Debug: 33 1 command.c:155 script_debug(): command - reset_config none
Debug: 34 1 command.c:155 script_debug(): command - echo DEPRECATED! use 'adapter speed' not 'adapter_khz'
User : 35 1 command.c:685 handle_echo(): DEPRECATED! use 'adapter speed' not 'adapter_khz'
Debug: 36 1 command.c:155 script_debug(): command - adapter speed 10000
Debug: 37 1 adapter.c:251 adapter_config_khz(): handle adapter khz
Debug: 38 1 adapter.c:215 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 39 1 adapter.c:215 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 40 1 command.c:155 script_debug(): command - ocd_find cpld/xilinx-xc7.cfg
Debug: 41 1 configuration.c:88 find_file(): found /home/bluespec/.local/xPacks/@xpack-dev-tools/openocd/0.12.0-1.1/.content/bin/../openocd/scripts/cpld/xilinx-xc7.cfg
Debug: 42 1 command.c:155 script_debug(): command - transport select
Info : 43 1 transport.c:265 jim_transport_select(): auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Debug: 44 1 command.c:155 script_debug(): command - transport select
Debug: 45 1 command.c:155 script_debug(): command - expr  [ string first "jtag" $_TRANSPORT ] != -1 
Debug: 46 1 command.c:155 script_debug(): command - jtag newtap xc7 tap -irlen 6 -ignore-version -expected-id 0x03622093 -expected-id 0x03620093 -expected-id 0x037C4093 -expected-id 0x0362F093 -expected-id 0x037C8093 -expected-id 0x037C7093 -expected-id 0x037C3093 -expected-id 0x0362E093 -expected-id 0x037C2093 -expected-id 0x0362D093 -expected-id 0x0362C093 -expected-id 0x03632093 -expected-id 0x03631093 -expected-id 0x03636093 -expected-id 0x03647093 -expected-id 0x0364C093 -expected-id 0x03651093 -expected-id 0x03747093 -expected-id 0x03656093 -expected-id 0x03752093 -expected-id 0x03751093 -expected-id 0x03671093 -expected-id 0x036B3093 -expected-id 0x036B7093 -expected-id 0x036BB093 -expected-id 0x036BF093 -expected-id 0x03667093 -expected-id 0x03682093 -expected-id 0x03687093 -expected-id 0x03692093 -expected-id 0x03691093 -expected-id 0x03696093 -expected-id 0x036D5093 -expected-id 0x036D9093 -expected-id 0x036DB093
Debug: 47 1 tcl.c:557 jim_newtap_cmd(): Creating New Tap, Chip: xc7, Tap: tap, Dotted: xc7.tap, 73 params
[... cut ...]
Debug: 169 8 tcl.c:1375 handle_flash_init_command(): Initializing flash devices...
Debug: 170 8 command.c:155 script_debug(): command - nand init
Debug: 171 8 tcl.c:487 handle_nand_init_command(): Initializing NAND devices...
Debug: 172 8 command.c:155 script_debug(): command - pld init
Debug: 173 8 pld.c:194 handle_pld_init_command(): Initializing PLDs...
Debug: 174 8 command.c:155 script_debug(): command - tpiu init
Debug: 175 8 gdb_server.c:3819 gdb_target_add_one(): skip gdb server for target xc7.proxy

TommyMurphyTM1234, 2023-03-28:
You're not using the latest OpenOCD built from this repo.
You should do that before reporting issues here.
You should also address the deprecated warnings in your OpenOCD scripts.
And you should also clarify the details of your target RISC-V implementation, how you have verified that the debug block is fully functional, what scripts you are using for debugging etc.
There are too many unknowns for anybody to provide any meaningful comment.

TommyMurphyTM1234, 2023-03-28:
> The board I am working with is an Arty A7 100T. The board is programmed with the core; 

What core/RISC-V implementation?

hparpinel, 2023-03-28:
I have the latest OpenOCD build. I was just trying different settings to figure out what was going on.
Update for future reference: I believe the issue was that my Ubuntu machine had a gdb installation and might have been intercepting the connection. I removed that installation and now its working.
Thanks for your help

TommyMurphyTM1234, 2023-03-28:
> I have the latest OpenOCD build. 

This is not a build of the latest `riscv-openocd`.

> ```
> xPack Open On-Chip Debugger 0.12.0-01004-g9ea7f3d64-dirty (2023-01-30-15:03)
> ```

> I was just trying different settings to figure out what was going on. Update for future reference: I believe the issue was that my Ubuntu machine had a gdb installation and might have been intercepting the connection. I removed that installation and now its working. Thanks for your help

Ok, can you close the issue if it's resolved so, please?

closed 2023-03-28 by hparpinel