riscv-collab/riscv-openocd #832: What does openocd use to judge the soc XLEN？My soc is rv32,and misa = 40000100,but openocd shows XLEN=64.

Damien-Wu, 2023-04-04:


TommyMurphyTM1234, 2023-04-04:
This is how OpenOCD detects the target XLEN (I'm assuming that your target is not using a "legacy" 0.11 debug block):

* https://github.com/riscv/riscv-openocd/blob/78231cda38b6e01e87ba69d9878a9a7d56dd0c14/src/target/riscv/riscv-013.c#L1705

You probably need to provide more details - for example:

* Details of your target
* OpenOCD verbose log using `openocd -d3 ...`
* Details of your OpenOCD/GDB setup
* What happens if you just use the OpenOCD telnet interface to eliminate GDB from the picture temporarily

TommyMurphyTM1234, 2023-04-05:
Any update @Damien-Wu?

Damien-Wu, 2023-04-06:
@TommyMurphyTM1234  Thanks for answer. This is my openocd.cfg 

`interface remote_bitbang
remote_bitbang_port 44853
remote_bitbang_host localhost

# Target configuration for the hummingbird E203 chip

if { [info exists CHIPNAME] } {
   set  _CHIPNAME $CHIPNAME
} else {
   set  _CHIPNAME riscv
}

if { [info exists CPUTAPID ] } {
   set _CPUTAPID $CPUTAPID
} else {
   set _CPUTAPID 0x00360a79
}

jtag newtap $_CHIPNAME tap -irlen 5 -expected-id $_CPUTAPID
set _TARGETNAME $_CHIPNAME.tap
target create $_TARGETNAME.0 riscv -chain-position $_TARGETNAME 

# Configure work area in on-chip SRAM[
$_TARGETNAME.0 configure -work-area-phys 0x20000000 -work-area-size 8192 -endian little

riscv expose_csrs 3040-3071

# Be verbose about GDB errors
gdb_report_data_abort enable
gdb_report_register_access_error enable

# Increase timeouts in simulation
riscv set_command_timeout_sec 1200

riscv set_reset_timeout_sec 10`
Additionly,i used 'set arch riscv:rv32' before 'target remote'.
I just read the riscv-013.c and confused about the function `register_read_abstract`,the return value is always `ERRO_OK`.

Damien-Wu, 2023-04-06:
@TommyMurphyTM1234  Thanks for answer. This is my openocd.cfg 

```
interface remote_bitbang
remote_bitbang_port 44853
remote_bitbang_host localhost
if { [info exists CHIPNAME] } {
   set  _CHIPNAME $CHIPNAME
} else {
   set  _CHIPNAME riscv
}
if { [info exists CPUTAPID ] } {
   set _CPUTAPID $CPUTAPID
} else {
   set _CPUTAPID 0x00360a79
}
jtag newtap $_CHIPNAME tap -irlen 5 -expected-id $_CPUTAPID
set _TARGETNAME $_CHIPNAME.tap
target create $_TARGETNAME.0 riscv -chain-position $_TARGETNAME 
$_TARGETNAME.0 configure -work-area-phys 0x20000000 -work-area-size 8192 -endian little
riscv expose_csrs 3040-3071
gdb_report_data_abort enable
gdb_report_register_access_error enable
riscv set_command_timeout_sec 1200
riscv set_reset_timeout_sec 10
```
Additionly,i used `set arch riscv:rv32` before `target remote`,this is my gdb.cfg:
```
set architecture riscv:rv32 
target extended-remote :3333
load /home/wh/Project/Agcsoc/agcsoc_sw/app/dummy/dummy`
```
When i execute It shows
```
warning: Selected architecture riscv:rv32 is ambiguous withreported target architecture riscv:rv64
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
```
I just read the riscv-013.c and confused about the function `register_read_abstract`,the return value is always `ERRO_OK`.
When i connect with openocd ,it shows debug log  `examine(): hart0: XLEN=64，misa=0x40000100`. 
****

TommyMurphyTM1234, 2023-04-06:
You haven't provided the other info that I suggested earlier, in particular the verbose OpenOCD log.

Damien-Wu, 2023-04-06:
@TommyMurphyTM1234  Sorry,this is OpenOCD log：
```
Open On-Chip Debugger 0.12.0+dev-01139-g99ec57609 (2023-04-04-11:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 3 1 options.c:52 configuration_output_handler(): debug_level: 3User : 4 1 options.c:52 configuration_output_handler(): 
Debug: 5 1 configuration.c:33 add_script_search_dir(): adding ./verilator/util/openocd/
Debug: 6 2 options.c:233 add_default_dirs(): bindir=/home/ww/riscv/riscv/bin
Debug: 7 2 options.c:234 add_default_dirs(): pkgdatadir=/home/ww/riscv/riscv/share/openocd
Debug: 8 2 options.c:235 add_default_dirs(): exepath=/home/ww/riscv/riscv/bin
Debug: 9 2 options.c:236 add_default_dirs(): bin2data=../share/openocd
Debug: 10 2 configuration.c:33 add_script_search_dir(): adding /home/ww/.config/openocd
Debug: 11 2 configuration.c:33 add_script_search_dir(): adding /home/ww/.openocd
Debug: 12 2 configuration.c:33 add_script_search_dir(): adding /home/ww/riscv/riscv/bin/../share/openocd/site
Debug: 13 2 configuration.c:33 add_script_search_dir(): adding /home/ww/riscv/riscv/bin/../share/openocd/scripts
Debug: 14 2 command.c:152 script_debug(): command - ocd_find hbird-e203.cfg
Debug: 15 2 configuration.c:88 find_file(): found ./verilator/util/openocd//openocd.cfg
Debug: 16 2 command.c:152 script_debug(): command - echo DEPRECATED! use 'adapter driver' not 'interface'
User : 17 2 command.c:690 handle_echo(): DEPRECATED! use 'adapter driver' not 'interface'
Debug: 18 2 command.c:152 script_debug(): command - adapter driver remote_bitbang
Info : 19 2 transport.c:107 allow_transports(): only one transport option; autoselecting 'jtag'
Debug: 20 2 command.c:152 script_debug(): command - echo DEPRECATED! use 'remote_bitbang port' not 'remote_bitbang_port'
User : 21 2 command.c:690 handle_echo(): DEPRECATED! use 'remote_bitbang port' not 'remote_bitbang_port'
Debug: 22 2 command.c:152 script_debug(): command - remote_bitbang port 44853
Debug: 23 2 command.c:152 script_debug(): command - echo DEPRECATED! use 'remote_bitbang host' not 'remote_bitbang_host'
User : 24 2 command.c:690 handle_echo(): DEPRECATED! use 'remote_bitbang host' not 'remote_bitbang_host'
Debug: 25 2 command.c:152 script_debug(): command - remote_bitbang host localhost
Debug: 26 2 command.c:152 script_debug(): command - jtag newtap riscv tap -irlen 5 -expected-id 0x00360a79
Debug: 27 2 tcl.c:561 jim_newtap_cmd(): Creating New Tap, Chip: riscv, Tap: tap, Dotted: riscv.tap, 4 params
Debug: 28 2 tcl.c:579 jim_newtap_cmd(): Processing option: -irlen
Debug: 29 2 tcl.c:579 jim_newtap_cmd(): Processing option: -expected-id
Debug: 30 2 core.c:1474 jtag_tap_init(): Created Tap: riscv.tap @ abs position 0, irlen 5, capture: 0x1 mask: 0x3
Debug: 31 2 command.c:152 script_debug(): command - target create riscv.tap.0 riscv -chain-position riscv.tap
Debug: 32 2 target.c:2193 target_free_all_working_areas_restore(): freeing all working areas
Debug: 33 2 riscv.c:432 riscv_create_target(): riscv_create_target()
Debug: 34 2 command.c:152 script_debug(): command - riscv.tap.0 configure -work-area-phys 0x20000000 -work-area-size 8192 -endian little
Debug: 35 2 target.c:2193 target_free_all_working_areas_restore(): freeing all working areas
Debug: 36 2 target.c:2193 target_free_all_working_areas_restore(): freeing all working areas
Debug: 37 2 command.c:152 script_debug(): command - riscv expose_csrs 3040-3071
Debug: 38 2 command.c:152 script_debug(): command - gdb_report_data_abort enable
Debug: 39 2 command.c:152 script_debug(): command - gdb_report_register_access_error enable
Debug: 40 2 command.c:152 script_debug(): command - riscv set_command_timeout_sec 1200
Debug: 41 2 command.c:152 script_debug(): command - riscv set_reset_timeout_sec 10
User : 42 2 options.c:52 configuration_output_handler(): 32User : 43 2 options.c:52 configuration_output_handler(): 
Info : 44 2 server.c:297 add_service(): Listening on port 6666 for tcl connections
Info : 45 2 server.c:297 add_service(): Listening on port 4444 for telnet connections
Debug: 46 2 command.c:152 script_debug(): command - init
Debug: 47 2 command.c:152 script_debug(): command - target init
Debug: 48 2 command.c:152 script_debug(): command - target names
Debug: 49 2 command.c:152 script_debug(): command - riscv.tap.0 cget -event gdb-flash-erase-start
Debug: 50 2 command.c:152 script_debug(): command - riscv.tap.0 configure -event gdb-flash-erase-start reset init
Debug: 51 2 command.c:152 script_debug(): command - riscv.tap.0 cget -event gdb-flash-write-end
Debug: 52 2 command.c:152 script_debug(): command - riscv.tap.0 configure -event gdb-flash-write-end reset halt
Debug: 53 2 command.c:152 script_debug(): command - riscv.tap.0 cget -event gdb-attach
Debug: 54 2 command.c:152 script_debug(): command - riscv.tap.0 configure -event gdb-attach halt 1000
Debug: 55 2 target.c:1651 handle_target_init_command(): Initializing targets...
Debug: 56 2 riscv.c:445 riscv_init_target(): riscv_init_target()
Debug: 57 2 semihosting_common.c:109 semihosting_common_init():  
Info : 58 2 remote_bitbang.c:315 remote_bitbang_init(): Initializing remote_bitbang driver
Info : 59 2 remote_bitbang.c:237 remote_bitbang_init_tcp(): Connecting to localhost:44853
Info : 60 2 remote_bitbang.c:326 remote_bitbang_init(): remote_bitbang driver initialized
Info : 61 2 adapter.c:158 adapter_init(): Note: The adapter "remote_bitbang" doesn't support configurable speed
[... cut ...]
Debug: 1368 7582 target.c:1837 target_call_event_callbacks(): target event 8 (gdb-end) for core riscv.tap.0
Debug: 1369 7582 target.c:1837 target_call_event_callbacks(): target event 23 (gdb-detach) for core riscv.tap.0
Debug: 1370 7582 riscv.c:491 riscv_deinit_target(): riscv_deinit_target()
Debug: 1371 7582 riscv-013.c:1520 deinit_target(): riscv_deinit_target()
Debug: 1372 7582 target.c:2193 target_free_all_working_areas_restore(): freeing all working areas
Info : 1373 7582 remote_bitbang.c:164 remote_bitbang_quit(): remote_bitbang interface quit
```

TommyMurphyTM1234, 2023-04-06:
>```
> Debug: 233 123 riscv-013.c:1750 examine():  hart 0: XLEN=64, misa=0x40000100
>```

This means that the 64-bit read of register `s0` in `examine()` succeeds so OpenOCD detects the target's XLEN as 64.

* https://github.com/riscv/riscv-openocd/blob/21d2787d64952c7d794591ccda40edd21cab9cf1/src/target/riscv/riscv-013.c#L1697

You need to figure out why this is the case.

You also still haven't clarified the nature and provenance of your RISC-V target which might shed some light on what's going on/wrong.

TommyMurphyTM1234, 2023-04-06:
At a glance, this seems to be the read of register `s0` for the purposes of XLEN detection:
```
Debug: 180 108 riscv-013.c:776 execute_abstract_command(): command=0x321008; access register, size=64, postexec=0, transfer=1, write=0, regno=0x1008
Debug: 181 109 riscv-013.c:391 scan(): 42b w 00321008 @17 -> + 004003a2 @11; 8i
Debug: 182 109 riscv-013.c:401 scan():  ->  impebreak allhalted anyhalted authenticated hasresethaltreq version=2
Debug: 183 111 riscv-013.c:391 scan(): 42b r 00000000 @16 -> + 004003a2 @11; 8i
Debug: 184 111 riscv-013.c:401 scan():  ->  impebreak allhalted anyhalted authenticated hasresethaltreq version=2
Debug: 185 112 riscv-013.c:391 scan(): 42b - 00000000 @16 -> + 0f000001 @16; 8i
Debug: 186 112 riscv-013.c:401 scan():  ->  progbufsize=15 datacount=1
```

TommyMurphyTM1234, 2023-04-06:
Ok, I presume that this
```
Info : 73 33 core.c:1133 jtag_examine_chain_display(): JTAG tap: riscv.tap tap/device found: 0x00360a79 (mfg: 0x53c (Efinix Inc), part: 0x0360, ver: 0x0)
```
means that you're using an Efinix RISC-V (based on/derived from VexRiscv) target?

That being the case, you might want to ask Efinix about this odd XLEN detection behaviour especially because there seem to be several other OpenOCD forks associated with Efinix/VexRiscv so maybe you're expected to use one of those and not this `riscv-openocd` fork?

Damien-Wu, 2023-04-07:
My target is designed for learning by myself,and the JTAG id just conflicts with Efinix.

TommyMurphyTM1234, 2023-04-07:
As I said, you need to figure out why a 64-bit read of register `s0` is succeeding if your target RISC-V has XLEN=32. If the XLEN actually is 32 then that read should fail and OpenOCD should detect that the target XLEN is 32.

Damien-Wu, 2023-04-07:
Yes,we took write it from scratch, just implemented RV32I，i wrote the core and DM module is written by another one.Through analyzing the wave ,the reason of a 64-bit read of register `s0` is the `cmderr` field of `dmstatuscs` didn`t keep enough time.We didn`t use any existing IP, and the implementation of Debug Mode might have some bugs .I`m just trying to debug.

TommyMurphyTM1234, 2023-04-07:
> And I found the mistake！The `cmderr` doesn`t maintain .

What does that mean?
If the issue is with your target and not OpenOCD then could you close this issue as invalid, please?

Damien-Wu, 2023-04-07:
Yes,we took write it from scratch, just implemented RV32I，i wrote the core and DM module is written by another one.Through analyzing the wave ,the reason of a 64-bit read of register `s0`  is succeeding is the `cmderr` field of `dmstatuscs` didn`t keep enough time.We didn`t use any existing IP, and the implementation of Debug Mode might have some bugs .I`m just trying to debug. 
- If the issue is with your target and not OpenOCD then could you close this issue as invalid, please?
Thanks . I will close the issue.

closed 2023-04-07 by Damien-Wu