riscv-collab/riscv-openocd #838: expose_csr not work when config target smp

zqb-all, 2023-04-21:
When OpenOCD is configured with the following lines:
```
set _TARGETNAME_0 $_CHIPNAME.cpu0
set _TARGETNAME_1 $_CHIPNAME.cpu1
set _TARGETNAME_2 $_CHIPNAME.cpu2
set _TARGETNAME_3 $_CHIPNAME.cpu3
target create $_TARGETNAME_0 riscv -chain-position $_CHIPNAME.cpu -rtos hwthread
target create $_TARGETNAME_1 riscv -chain-position $_CHIPNAME.cpu -coreid 1
target create $_TARGETNAME_2 riscv -chain-position $_CHIPNAME.cpu -coreid 2
target create $_TARGETNAME_3 riscv -chain-position $_CHIPNAME.cpu -coreid 3
target smp $_TARGETNAME_0 $_TARGETNAME_1 $_TARGETNAME_2 $_TARGETNAME_3
```
the configuration  `riscv expose_csrs ` does not work, and OpenOCD shows a warning message like:
```
Warn : Register csr_xxx does not exist in riscv.cpu0, which is part of an SMP group where this register does exist
```
Is there any other configuration that I need to use to make this work?

timsifive, 2023-04-21:
`expose_csrs` functions on the current target only. If you have multiple targets, it needs to be executed for each one. See e.g. https://github.com/riscv-software-src/riscv-tests/blob/master/debug/targets/RISC-V/spike-2.cfg#L21

zqb-all, 2023-04-21:
> `expose_csrs` functions on the current target only. If you have multiple targets, it needs to be executed for each one. See e.g. https://github.com/riscv-software-src/riscv-tests/blob/master/debug/targets/RISC-V/spike-2.cfg#L21

Thanks, I will try that.

closed 2023-04-21 by zqb-all