riscv-collab/riscv-openocd #845: probable issue with a loop inside riscv/program.c

bieganski, 2023-05-08:
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/program.c#L75

does that loop always performs 0 iterations?

if there is some magic and the answer is "no", then I think it should be marked with a comment.

TommyMurphyTM1234, 2023-05-08:
> does that loop always has no effect?

Why do you think that this is the case?
Isn't the loop executed `riscv_debug_buffer_size(p->target)` times?

* https://github.com/riscv/riscv-openocd/blob/80d529cad3706a9f8ad23ac8f73ae7434d655dc7/src/target/riscv/riscv.c#L4739
* https://github.com/riscv/riscv-openocd/blob/80d529cad3706a9f8ad23ac8f73ae7434d655dc7/src/target/riscv/riscv-013.c#L1695
* https://github.com/riscv/riscv-openocd/blob/80d529cad3706a9f8ad23ac8f73ae7434d655dc7/src/target/riscv/riscv-013.c#L2396
* https://github.com/riscv/riscv-openocd/blob/80d529cad3706a9f8ad23ac8f73ae7434d655dc7/src/target/riscv/riscv-013.c#L1626

TommyMurphyTM1234, 2023-05-08:
Certainly looks like the loop has no effect given the `if` statement inside the loop!
```
	for (size_t i = 0; i < riscv_debug_buffer_size(p->target); ++i)
		if (i >= riscv_debug_buffer_size(p->target))
			p->debug_buffer[i] = riscv_read_debug_buffer(t, i);
```

TommyMurphyTM1234, 2023-05-12:
I presume that this issue can now be closed as a result of PR #846?

en-sc, 2024-01-22:
@TommyMurphyTM1234, thanks for noticing and linking the fix!

closed 2024-01-22 by en-sc