riscv-collab/riscv-openocd #849: Identify new hart on the fly

hananyasegal1, 2023-05-15:
Hi All,

I would like to ask if Openocd can identify a new hart after the initialization. 
For example if at init stage only hart0 is awaken and after some period of time during regular operation hart1 has been started, Can Openocd identify hart1?

Thanks,
Hananya

hananyasegal1, 2023-05-15:
Hi All,

One more question, if I want to change the Openocd source code itself, where is the preferred location to start with?

timsifive, 2023-05-15:
If your changes are only relevant to RISC-V targets, they should be made here. If they are more general, or in target-independent code, then ideally they are made upstream as described in https://github.com/riscv/riscv-openocd/blob/riscv/HACKING

hananyasegal1, 2023-05-15:
Hi Tim,

Thanks for your answer.
The targey is Riscv.
What do you mean 'here' in your answer?

Thanks,

hananyasegal1, 2023-05-15:
Hi Tim,

My question was where  is the recommended source code location to statt the invedtigation.

Thanks,
Hananya

hananyasegal1, 2023-05-15:
Yes, I know.
Where do you think is the right place i the sourcecode  to start the investigation with?

TommyMurphyTM1234, 2023-05-15:
> Yes, I know. Where do you think is the right place in the sourcecode to start the investigation with?

@timsifive already answered this question above.

* https://github.com/riscv/riscv-openocd/issues/849#issuecomment-1548080885

hananyasegal1, 2023-05-16:
Hi Tommy,

Thanks for your great answer.
The current solution is to actively send a command to Openocd to connect with command arp_examine. I want Openocd to do a scan in specific time interval or per step for example. I don't want that the user will type a specific command for this purpose.

TommyMurphyTM1234, 2023-05-16:
In my opinion that sounds like something that might be implemented as a generic target feature - i.e. not specific to RISC-V harts but available to other targets with cores in a similar configuration. In which case it sounds like something that might be more suitable for discussion on the upstream OpenOCD development mailing list and implementation in the upstream source base:

* https://openocd.org/pages/mailing-lists.html
* https://openocd.org/pages/repos.html

hananyasegal1, 2023-05-16:
Thank you very much for detailed answer. It helps me a lot. I'll carfully look at it.

en-sc, 2024-01-22:
@hananyasegal1, seems like the issue is resolved. Can I close it?

closed 2024-01-26 by en-sc