riscv-collab/riscv-openocd #863: riscv_set_current_hartid/riscv_select_current_hart/riscv_current_hartid no defined

wjzhang, 2023-06-09:
Hi:   When I build latest RISCV-OPENOCD with old flash driver. found the riscv_set_current_hartid() API is removed by this commit : Remove riscv_info_t.current_hartid(Change-Id: Ibe8d5e156523397f245edd6ec0a5df3239b717bf

But it did not remove the declaration in src\target\riscv\riscv.h.

How to fix old flash driver? just delete these APIs call? or Is there have replace API?

wjzhang, 2023-06-11:
> What flash driver are you referring to?

Hi Tommy:
The flash driver source code did not commit(pull request) to riscv-openocd
riscv branch.  it's in the https://github.com/hpmicro/riscv-openocd repo.
the hpmicro HPM6750 SPI flash driver:  hpm_xpi.
\src\flash\nor\hpm_xpi.c

TommyMurphyTM1234, 2023-06-12:
> Hi Tommy: The flash driver source code did not commit(pull request) to riscv-openocd riscv branch. it's in the https://github.com/hpmicro/riscv-openocd repo. the hpmicro HPM6750 SPI flash driver: hpm_xpi. \src\flash\nor\hpm_xpi.c
> […](#)
> -- Have a nice day! Best regards! Zhangwenjin

I don't understand.
Why is a flash driver calling target (RISC-V) specific APIs?
That seems very odd.

timsifive, 2023-06-12:
You shouldn't need any of these functions. The code is (generally) clean enough that you can just call the functions you need with the appropriate target struct, and the code inside will take care of selecting the correct hart.

I agree with Tommy that you shouldn't need to call any of these functions directly from the flash driver. Here's an example of a heavily optimized one that barely calls any riscv functions at all: https://github.com/riscv/riscv-openocd/blob/riscv/src/flash/nor/fespi.c

wjzhang, 2023-06-12:
Awesome, thanks for the info!

en-sc, 2024-01-22:
The issue seems to be resolved. @wjzhang, are you OK with me closing it?

wjzhang, 2024-01-22:
Yes, go ahead.

closed 2024-01-23 by en-sc