riscv-collab/riscv-openocd #869: Passing and failing riscv-tests/debug on v0.11 - request for confirmation

JanMatCodasip, 2023-06-22:
We have run `riscv-tests/debug` on HiFive1 board (Rev A01) with RISC-V debug implementation v0.11. Each test was repeated 30 times in order to catch sporadic fails, if any.

The results are in table below.

@timsifive - We would like to ask for confirmation whether you are getting the same results, so that we can establish a baseline for regression testing of the 0.11 debug support in riscv-openocd.

Thank you.

riscv-openocd commit: `e0dd44a53c9df2206d854526082c34635866ea0b`
riscv-tests commit: `7b52ba3b7167acb4d8b38f4d4633112b4699cb26`

| target  	| test                              	| HiFive1 Rev A01, HiFive1.py 	| HiFive1 Rev A01, HiFive1-flash.py 	|
|---------	|-----------------------------------	|-----------------------------	|-----------------------------------	|
| HiFive1 	| CeaseMultiTest                    	| not applicable              	| not applicable                    	|
| HiFive1 	| CeaseRunTest                      	| not applicable              	| not applicable                    	|
| HiFive1 	| CeaseStepiTest                    	| not applicable              	| not applicable                    	|
| HiFive1 	| CheckMisa                         	| pass                        	| pass                              	|
| HiFive1 	| CustomRegisterTest                	| not applicable              	| not applicable                    	|
| HiFive1 	| DebugBreakpoint                   	| pass                        	| exception                         	|
| HiFive1 	| DebugChangeString                 	| pass                        	| pass                              	|
| HiFive1 	| DebugCompareSections              	| pass                        	| pass                              	|
| HiFive1 	| DebugExit                         	| pass                        	| exception                         	|
| HiFive1 	| DebugFunctionCall                 	| pass                        	| exception                         	|
| HiFive1 	| DebugSymbols                      	| pass                        	| pass                              	|
| HiFive1 	| DebugTurbostep                    	| pass                        	| pass                              	|
| HiFive1 	| DisconnectTest                    	| pass                        	| pass                              	|
| HiFive1 	| DownloadTest                      	| pass                        	| pass                              	|
| HiFive1 	| EbreakTest                        	| pass                        	| pass                              	|
| HiFive1 	| EtriggerTest                      	| fail                        	| exception                         	|
| HiFive1 	| FreeRtosTest                      	| not applicable              	| not applicable                    	|
| HiFive1 	| Hwbp1                             	| pass                        	| exception                         	|
| HiFive1 	| Hwbp2                             	| pass                        	| exception                         	|
| HiFive1 	| HwbpManual                        	| not applicable              	| not applicable                    	|
| HiFive1 	| IcountTest                        	| fail                        	| fail                              	|
| HiFive1 	| InfoTest                          	| pass                        	| pass                              	|
| HiFive1 	| InstantChangePc                   	| pass                        	| pass                              	|
| HiFive1 	| InstantHaltTest                   	| fail                        	| fail                              	|
| HiFive1 	| InterruptTest                     	| pass                        	| pass                              	|
| HiFive1 	| ItriggerTest                      	| exception                   	| exception                         	|
| HiFive1 	| JumpHbreak                        	| pass                        	| pass                              	|
| HiFive1 	| MemorySampleMixed                 	| exception                   	| exception                         	|
| HiFive1 	| MemorySampleSingle                	| exception                   	| exception                         	|
| HiFive1 	| MemTest16                         	| pass                        	| pass                              	|
| HiFive1 	| MemTest32                         	| pass                        	| pass                              	|
| HiFive1 	| MemTest64                         	| pass                        	| pass                              	|
| HiFive1 	| MemTest8                          	| pass                        	| pass                              	|
| HiFive1 	| MemTestBlock0                     	| pass                        	| pass                              	|
| HiFive1 	| MemTestBlock1                     	| pass                        	| pass                              	|
| HiFive1 	| MemTestBlock2                     	| pass                        	| pass                              	|
| HiFive1 	| MemTestBlockReadInvalid           	| not applicable              	| not applicable                    	|
| HiFive1 	| MemTestReadInvalid                	| exception                   	| exception                         	|
| HiFive1 	| MulticoreRegTest                  	| not applicable              	| not applicable                    	|
| HiFive1 	| MulticoreRtosSwitchActiveHartTest 	| not applicable              	| not applicable                    	|
| HiFive1 	| MulticoreRunAllHaltOne            	| not applicable              	| not applicable                    	|
| HiFive1 	| PrivChange                        	| not applicable              	| not applicable                    	|
| HiFive1 	| PrivRw                            	| pass                        	| pass                              	|
| HiFive1 	| ProgramHwWatchpoint               	| pass                        	| pass                              	|
| HiFive1 	| ProgramSwWatchpoint               	| pass                        	| pass                              	|
| HiFive1 	| Registers                         	| pass                        	| exception                         	|
| HiFive1 	| RepeatReadTest                    	| exception                   	| exception                         	|
| HiFive1 	| Semihosting                       	| fail                        	| fail                              	|
| HiFive1 	| SemihostingFileio                 	| fail                        	| fail                              	|
| HiFive1 	| SimpleF18Test                     	| pass                        	| pass                              	|
| HiFive1 	| SimpleNoExistTest                 	| pass                        	| pass                              	|
| HiFive1 	| SimpleS0Test                      	| pass                        	| pass                              	|
| HiFive1 	| SimpleS1Test                      	| pass                        	| pass                              	|
| HiFive1 	| SimpleT0Test                      	| pass                        	| pass                              	|
| HiFive1 	| SimpleT1Test                      	| pass                        	| pass                              	|
| HiFive1 	| SimpleV13Test                     	| pass                        	| pass                              	|
| HiFive1 	| SmpSimultaneousRunHalt            	| not applicable              	| not applicable                    	|
| HiFive1 	| StepTest                          	| pass                        	| pass                              	|
| HiFive1 	| StepThread2Test                   	| not applicable              	| not applicable                    	|
| HiFive1 	| Sv32Test                          	| not applicable              	| not applicable                    	|
| HiFive1 	| Sv39Test                          	| not applicable              	| not applicable                    	|
| HiFive1 	| Sv48Test                          	| not applicable              	| not applicable                    	|
| HiFive1 	| TooManyHwbp                       	| pass                        	| exception                         	|
| HiFive1 	| TriggerDmode                      	| pass                        	| pass                              	|
| HiFive1 	| TriggerExecuteInstant             	| pass                        	| pass                              	|
[... cut ...]
| HiFive1 	| TriggerStoreAddressInstant        	| pass                        	| pass                              	|
| HiFive1 	| UserInterrupt                     	| pass                        	| exception                         	|
| HiFive1 	| VectorTest                        	| not applicable              	| not applicable                    	|
| HiFive1 	| WriteCsrs                         	| pass                        	| pass                              	|
| HiFive1 	| WriteGprs                         	| pass                        	| pass                              	|

TommyMurphyTM1234, 2023-06-22:
Hi @JanMatCodasip - I have the same board so might be able to repeat these tests for you but I cannot find anything explaining how to run the (a?) test suite. Maybe you or somebody else can point me in the right direction please? Thanks.

timsifive, 2023-06-22:
@TommyMurphyTM1234 You found the tests. :-) If the README there isn't sufficient to let you run them, let me know and I'll try to make it more clear.

@JanMatCodasip This list looks alright to me. When I run the HiFive1 tests (rarely), I'm generally happy as long as most of the tests are passing, because there is very little hardware out there that implements 0.11 (I only know of HiFive1). It would be nice to have a list of known not-passing tests for proper regression testing.

JanMatCodasip, 2023-06-23:
> Tim: This list looks alright to me. (...) It would be nice to have a list of known not-passing tests for proper regression testing.

@timsifive Thank you for looking at this. Unless you or @TommyMurphyTM1234 disagree with some items in the table within a week or so, I will take the table above as the base for our regression testing of v0.11 debug support.

timsifive, 2023-06-23:
That sounds good to me. I assume you'll then commit a new skip file using the mechanism from https://github.com/riscv-software-src/riscv-tests/pull/477

JanMatCodasip, 2023-06-26:
Sure, we can create one such file with the known failing tests on HiFive1 and commit it to riscv-tests.

TommyMurphyTM1234, 2023-06-28:
> @timsifive Thank you for looking at this. Unless you or @TommyMurphyTM1234 disagree with some items in the table within a week or so, I will take the table above as the base for our regression [...]

Hi @JanMatCodasip - I still plan to perform the tests for comparison with your results - hopefully today. Apologies for the delay but other stuff cropped up recently.

timsifive, 2023-06-28:
There should be no need to have them in the system path. I have all my tools under /opt/riscv. They are in my PATH.

Etrigger test is expected to fail on HiFive1 because it does not implement exception triggers. This should really be resolved by adding a property in testlib.Target the says whether etrigger is supported, checking for that in EtriggerTest, and setting the property for the spike targets.

TommyMurphyTM1234, 2023-06-28:
Hi @timsifive - thanks for the quick reply.

> There should be no need to have them in the system path. I have all my tools under /opt/riscv. They are in my PATH.

I had them in my `PATH` but something in the scripts still insisted on at least Spike and OpenOCD being in `/bin` and failed when they were not - so I had to manually copy those two binaries into `/bin`. I'll take that offline and log a separate issue in the tests repo if necessary.

> Etrigger test is expected to fail on HiFive1 because it does not implement exception triggers. 

I'm not yet running against the HiFive1 - I'm running the smoke test against Spike.
The instructions suggest that this should "just work"?
Am I missing something?

* https://github.com/riscv-software-src/riscv-tests/tree/master/debug#usage

> To run a quick smoke test against spike, run `make`.

timsifive, 2023-06-28:
https://github.com/riscv/riscv-openocd/blob/riscv/.github/workflows/spike-openocd-tests.yml might serve as a concrete set of instructions that are run regularly.

TommyMurphyTM1234, 2023-06-28:
> https://github.com/riscv/riscv-openocd/blob/riscv/.github/workflows/spike-openocd-tests.yml might serve as a concrete set of instructions that are run regularly.

I tried running the 32-bit tests:

* https://github.com/riscv/riscv-openocd/blob/4be6a5683f9de5705c5d8a28f4ed9ada2250dccb/.github/workflows/spike-openocd-tests.yml#L106

as follows (all tools are on my `PATH`):
```
./gdbserver.py targets/RISC-V/spike32.py --print-failures
```
but I get this:
```
Using $misa from hart definition: 0x4034112d
[CeaseMultiTest] Starting > logs/20230628-210201-spike32-CeaseMultiTest.log
[CeaseMultiTest] not_applicable in 0.00s
[CeaseRunTest] Starting > logs/20230628-210201-spike32-CeaseRunTest.log
[CeaseRunTest] not_applicable in 0.00s
[CeaseStepiTest] Starting > logs/20230628-210201-spike32-CeaseStepiTest.log
[CeaseStepiTest] not_applicable in 0.00s
[CheckMisa] Starting > logs/20230628-210201-spike32-CheckMisa.log
[CheckMisa] exception in 3.83s
Test: CheckMisa
Target: spike32
---------------------------------[ Compile ]----------------------------------
+ riscv64-unknown-elf-gcc -g programs/checksum.c programs/tiny-malloc.c programs/infinite_loop.S -DDEFINE_MALLOC -DDEFINE_FREE programs/entry.S programs/init.c -DNHARTS=1 -I ../env -T targets/RISC-V/spike32.lds -nostartfiles -mcmodel=medany -DXLEN=32 -o spike32_checksum-4034112d -march=rv32imafdcv -mabi=ilp32
-------------------------[ /tmp/spike-2oumi2ve.log ]--------------------------
+ spike -p1 --isa RV32IMAFDCV --dm-auth --dmi-rti 4 --dm-no-abstract-fpr --dm-no-halt-groups --varch=vlen:128,elen:64 -m0x10100000:0x10000000 --rbb-port 0 spike32_checksum-4034112d
spike: unrecognized option --dm-no-abstract-fpr
Try 'spike --help' for more information.

--------------------------------[ Traceback ]---------------------------------
Traceback (most recent call last):
  File "/home/user/Downloads/riscv-tests/debug/testlib.py", line 1203, in run
    self.classSetup()
  File "/home/user/Downloads/riscv-tests/debug/testlib.py", line 1257, in classSetup
    BaseTest.classSetup(self)
  File "/home/user/Downloads/riscv-tests/debug/testlib.py", line 1166, in classSetup
    self.target_process = self.target.create()
  File "/home/user/Downloads/riscv-tests/debug/targets/RISC-V/spike32.py", line 23, in create
    return testlib.Spike(self, isa="RV32IMAFDCV", dmi_rti=4,
  File "/home/user/Downloads/riscv-tests/debug/testlib.py", line 117, in __init__
    raise TestLibError("Didn't get spike message about bitbang "
testlib.TestLibError: Didn't get spike message about bitbang connection
-------------------------------[ End of logs ]--------------------------------

...
```
Most or all tests generate these 
```
spike: unrecognized option --dm-no-abstract-fpr

testlib.TestLibError: Didn't get spike message about bitbang connection
```
Am I supposed to do something special when compiling Spike and/or OpenOCD?

OpenOCD supports these interfaces:
```
openocd -c "adapter list"
Open On-Chip Debugger 0.12.0+dev-02946-g4be6a5683 (2023-06-28-09:36)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
The following debug adapters are available:
1: ftdi
2: usb_blaster
3: esp_usb_jtag
4: ft232r
5: usbprog
6: vsllink
7: rlink
8: ulink
9: arm-jtag-ew
10: buspirate
11: remote_bitbang
12: hla
13: osbdm
14: opendous
15: cmsis-dap
16: xds110
17: st-link

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: Debug Adapter has to be specified, see "adapter driver" command
```

TommyMurphyTM1234, 2023-06-28:
> `spike: unrecognized option --dm-no-abstract-fpr`

Sigh... The Spike pointed to by `riscv-gnu-toolchain` predates the addition of this option.

* https://github.com/riscv-software-src/riscv-isa-sim/blob/a0298a33e7b2091ba8d9f3a20838d96dc1164cac/spike_main/spike.cc#L74

So I guess that the `riscv-gnu-toolchain` has to be built using `--with-spike-src` to point to the latest Spike repo contents.

* https://github.com/riscv-collab/riscv-gnu-toolchain#use-source-tree-other-than-riscv-gnu-toolchain

TommyMurphyTM1234, 2023-06-29:
> Sigh... The Spike pointed to by `riscv-gnu-toolchain` predates the addition of this option.
>
> * [...]

I did this - built the latest Spike using `--with-spike-src` - and now the tests seem to be running OK.
I guess that the second error here was a red herring caused by the first:
```
spike: unrecognized option --dm-no-abstract-fpr
testlib.TestLibError: Didn't get spike message about bitbang connection
```
I'll try the tests against the HiFive1 hardware tomorrow hopefully.

TommyMurphyTM1234, 2023-06-29:
Thanks @timsifive for your assistance. 
I am now able to run the tests against the HiFive1 Rev A01 board.
@JanMatCodasip - how can I compare my results against yours?
Is there some tool/script to produce your results table?
```
./gdbserver.py targets/SiFive/HiFive1.py 
Using $misa from hart definition: 0x40001105
[CeaseMultiTest] Starting > logs/20230629-123924-HiFive1-CeaseMultiTest.log
[CeaseMultiTest] not_applicable in 0.00s
[CeaseRunTest] Starting > logs/20230629-123924-HiFive1-CeaseRunTest.log
[CeaseRunTest] not_applicable in 0.00s
[CeaseStepiTest] Starting > logs/20230629-123924-HiFive1-CeaseStepiTest.log
[CeaseStepiTest] not_applicable in 0.00s
[CheckMisa] Starting > logs/20230629-123924-HiFive1-CheckMisa.log
[CheckMisa] pass in 1.75s
[CustomRegisterTest] Starting > logs/20230629-123926-HiFive1-CustomRegisterTest.log
[CustomRegisterTest] not_applicable in 0.00s
[DebugBreakpoint] Starting > logs/20230629-123926-HiFive1-DebugBreakpoint.log
[DebugBreakpoint] exception in 0.26s
[DebugChangeString] Starting > logs/20230629-123926-HiFive1-DebugChangeString.log
[DebugChangeString] exception in 0.23s
[DebugCompareSections] Starting > logs/20230629-123926-HiFive1-DebugCompareSections.log
[DebugCompareSections] exception in 0.24s
[DebugExit] Starting > logs/20230629-123926-HiFive1-DebugExit.log
[DebugExit] exception in 0.24s
[DebugFunctionCall] Starting > logs/20230629-123927-HiFive1-DebugFunctionCall.log
[DebugFunctionCall] exception in 0.24s
[DebugSymbols] Starting > logs/20230629-123927-HiFive1-DebugSymbols.log
[DebugSymbols] exception in 0.23s
[DebugTurbostep] Starting > logs/20230629-123927-HiFive1-DebugTurbostep.log
[DebugTurbostep] exception in 0.24s
[DisconnectTest] Starting > logs/20230629-123927-HiFive1-DisconnectTest.log
[DisconnectTest] pass in 9.81s
[DownloadTest] Starting > logs/20230629-123937-HiFive1-DownloadTest.log
[DownloadTest] exception in 0.13s
[EbreakTest] Starting > logs/20230629-123937-HiFive1-EbreakTest.log
[EbreakTest] exception in 0.12s
[EtriggerTest] Starting > logs/20230629-123937-HiFive1-EtriggerTest.log
[EtriggerTest] exception in 0.24s
[FreeRtosTest] Starting > logs/20230629-123938-HiFive1-FreeRtosTest.log
[FreeRtosTest] not_applicable in 0.00s
[Hwbp1] Starting > logs/20230629-123938-HiFive1-Hwbp1.log
[Hwbp1] exception in 0.24s
[Hwbp2] Starting > logs/20230629-123938-HiFive1-Hwbp2.log
[Hwbp2] exception in 0.23s
[HwbpManual] Starting > logs/20230629-123938-HiFive1-HwbpManual.log
[HwbpManual] exception in 0.24s
[IcountTest] Starting > logs/20230629-123938-HiFive1-IcountTest.log
[IcountTest] exception in 0.12s
[InfoTest] Starting > logs/20230629-123938-HiFive1-InfoTest.log
[InfoTest] pass in 1.75s
[InstantChangePc] Starting > logs/20230629-123940-HiFive1-InstantChangePc.log
[InstantChangePc] pass in 7.03s
[InstantHaltTest] Starting > logs/20230629-123947-HiFive1-InstantHaltTest.log
[InstantHaltTest] fail in 5.28s
[InterruptTest] Starting > logs/20230629-123953-HiFive1-InterruptTest.log
[InterruptTest] exception in 0.13s
[ItriggerTest] Starting > logs/20230629-123953-HiFive1-ItriggerTest.log
[ItriggerTest] exception in 0.13s
[JumpHbreak] Starting > logs/20230629-123953-HiFive1-JumpHbreak.log
[JumpHbreak] exception in 0.12s
[MemTest16] Starting > logs/20230629-123953-HiFive1-MemTest16.log
[MemTest16] pass in 3.00s
[MemTest32] Starting > logs/20230629-123956-HiFive1-MemTest32.log
[MemTest32] pass in 3.23s
[MemTest64] Starting > logs/20230629-123959-HiFive1-MemTest64.log
[MemTest64] pass in 3.12s
[MemTest8] Starting > logs/20230629-124002-HiFive1-MemTest8.log
[MemTest8] pass in 3.34s
[MemTestBlock0] Starting > logs/20230629-124006-HiFive1-MemTestBlock0.log
[... 287 lines cut ...]
   ItriggerTest > logs/20230629-124321-HiFive1Flash-ItriggerTest.log
   JumpHbreak > logs/20230629-124321-HiFive1Flash-JumpHbreak.log
   MemTestReadInvalid > logs/20230629-124349-HiFive1Flash-MemTestReadInvalid.log
   MemorySampleMixed > logs/20230629-124359-HiFive1Flash-MemorySampleMixed.log
   MemorySampleSingle > logs/20230629-124359-HiFive1Flash-MemorySampleSingle.log
   PrivChange > logs/20230629-124359-HiFive1Flash-PrivChange.log
   PrivRw > logs/20230629-124359-HiFive1Flash-PrivRw.log
   ProgramHwWatchpoint > logs/20230629-124359-HiFive1Flash-ProgramHwWatchpoint.log
   ProgramSwWatchpoint > logs/20230629-124359-HiFive1Flash-ProgramSwWatchpoint.log
   Registers > logs/20230629-124400-HiFive1Flash-Registers.log
   RepeatReadTest > logs/20230629-124400-HiFive1Flash-RepeatReadTest.log
   Semihosting > logs/20230629-124400-HiFive1Flash-Semihosting.log
   SemihostingFileio > logs/20230629-124400-HiFive1Flash-SemihostingFileio.log
   StepTest > logs/20230629-124452-HiFive1Flash-StepTest.log
   TooManyHwbp > logs/20230629-124452-HiFive1Flash-TooManyHwbp.log
   TriggerDmode > logs/20230629-124452-HiFive1Flash-TriggerDmode.log
   TriggerExecuteInstant > logs/20230629-124452-HiFive1Flash-TriggerExecuteInstant.log
   TriggerLoadAddressInstant > logs/20230629-124452-HiFive1Flash-TriggerLoadAddressInstant.log
   TriggerStoreAddressInstant > logs/20230629-124452-HiFive1Flash-TriggerStoreAddressInstant.log
   UserInterrupt > logs/20230629-124453-HiFive1Flash-UserInterrupt.log
   WriteCsrs > logs/20230629-124453-HiFive1Flash-WriteCsrs.log
   WriteGprs > logs/20230629-124453-HiFive1Flash-WriteGprs.log
1 tests returned fail
   InstantHaltTest > logs/20230629-124314-HiFive1Flash-InstantHaltTest.log
```

TommyMurphyTM1234, 2023-06-29:
> @JanMatCodasip - how can I compare my results against yours? Is there some tool/script to produce your results table? Or do you want me to simply compress and upload a tarball of the logs?

Why not... :-)

[hifive1-a01-tests-logs.tar.gz](https://github.com/riscv/riscv-openocd/files/11905281/hifive1-a01-tests-logs.tar.gz)

TommyMurphyTM1234, 2023-06-29:
Looks like some stuff is not up to date with the toolchain post separation of the `Zicsr`/`Zifencei` extensions out from the base integer ISA?
```
Test: DebugExit
Target: HiFive1
---------------------------------[ Compile ]----------------------------------
+ riscv64-unknown-elf-gcc -g programs/debug.c programs/checksum.c programs/tiny-malloc.c -DDEFINE_MALLOC -DDEFINE_FREE programs/entry.S programs/init.c -DNHARTS=1 -I ../env -T targets/SiFive/HiFive1.lds -nostartfiles -mcmodel=medany -DXLEN=32 -o HiFive1_debug-40001105 -march=rv32imac -mabi=ilp32
 programs/entry.S: Assembler messages:
programs/entry.S:30: Error: unrecognized opcode `csrw mtvec,t0', extension `zicsr' required
```

TommyMurphyTM1234, 2023-06-29:
> programs/entry.S:30: Error: unrecognized opcode `csrw mtvec,t0', extension `zicsr' required

> ./gdbserver.py targets/RISC-V/spike32.py --print-failures

I made the change mentioned here:

* https://github.com/riscv-software-src/riscv-tests/issues/482#issuecomment-1613594248

and ran the tests again with these results:
```
./gdbserver.py targets/SiFive/HiFive1.py
Using $misa from hart definition: 0x40001105
[CeaseMultiTest] Starting > logs/20230629-202019-HiFive1-CeaseMultiTest.log
[CeaseMultiTest] not_applicable in 0.00s
[CeaseRunTest] Starting > logs/20230629-202019-HiFive1-CeaseRunTest.log
[CeaseRunTest] not_applicable in 0.00s
[CeaseStepiTest] Starting > logs/20230629-202019-HiFive1-CeaseStepiTest.log
[CeaseStepiTest] not_applicable in 0.00s
[CheckMisa] Starting > logs/20230629-202019-HiFive1-CheckMisa.log
[CheckMisa] pass in 1.89s
[CustomRegisterTest] Starting > logs/20230629-202021-HiFive1-CustomRegisterTest.log
[CustomRegisterTest] not_applicable in 0.00s
[DebugBreakpoint] Starting > logs/20230629-202021-HiFive1-DebugBreakpoint.log
[DebugBreakpoint] pass in 5.55s
[DebugChangeString] Starting > logs/20230629-202027-HiFive1-DebugChangeString.log
[DebugChangeString] pass in 5.53s
[DebugCompareSections] Starting > logs/20230629-202032-HiFive1-DebugCompareSections.log
[DebugCompareSections] pass in 4.43s
[DebugExit] Starting > logs/20230629-202037-HiFive1-DebugExit.log
[DebugExit] pass in 3.55s
[DebugFunctionCall] Starting > logs/20230629-202040-HiFive1-DebugFunctionCall.log
[DebugFunctionCall] pass in 5.61s
[DebugSymbols] Starting > logs/20230629-202046-HiFive1-DebugSymbols.log
[DebugSymbols] pass in 4.47s
[DebugTurbostep] Starting > logs/20230629-202050-HiFive1-DebugTurbostep.log
[DebugTurbostep] pass in 9.14s
[DisconnectTest] Starting > logs/20230629-202059-HiFive1-DisconnectTest.log
[DisconnectTest] pass in 9.64s
[DownloadTest] Starting > logs/20230629-202109-HiFive1-DownloadTest.log
[DownloadTest] pass in 4.70s
[EbreakTest] Starting > logs/20230629-202114-HiFive1-EbreakTest.log
[EbreakTest] pass in 4.94s
[EtriggerTest] Starting > logs/20230629-202119-HiFive1-EtriggerTest.log
[EtriggerTest] fail in 8.18s
[FreeRtosTest] Starting > logs/20230629-202127-HiFive1-FreeRtosTest.log
[FreeRtosTest] not_applicable in 0.00s
[Hwbp1] Starting > logs/20230629-202127-HiFive1-Hwbp1.log
[Hwbp1] pass in 4.88s
[Hwbp2] Starting > logs/20230629-202132-HiFive1-Hwbp2.log
[Hwbp2] pass in 6.10s
[HwbpManual] Starting > logs/20230629-202138-HiFive1-HwbpManual.log
[HwbpManual] not_applicable in 4.85s
[IcountTest] Starting > logs/20230629-202143-HiFive1-IcountTest.log
[IcountTest] fail in 6.51s
[InfoTest] Starting > logs/20230629-202149-HiFive1-InfoTest.log
[InfoTest] pass in 1.64s
[InstantChangePc] Starting > logs/20230629-202151-HiFive1-InstantChangePc.log
[InstantChangePc] pass in 6.59s
[InstantHaltTest] Starting > logs/20230629-202157-HiFive1-InstantHaltTest.log
[InstantHaltTest] fail in 6.62s
[InterruptTest] Starting > logs/20230629-202204-HiFive1-InterruptTest.log
[InterruptTest] pass in 14.23s
[ItriggerTest] Starting > logs/20230629-202218-HiFive1-ItriggerTest.log
[ItriggerTest] exception in 47.22s
[JumpHbreak] Starting > logs/20230629-202305-HiFive1-JumpHbreak.log
[JumpHbreak] pass in 4.44s
[MemTest16] Starting > logs/20230629-202310-HiFive1-MemTest16.log
[MemTest16] pass in 3.46s
[MemTest32] Starting > logs/20230629-202313-HiFive1-MemTest32.log
[MemTest32] pass in 3.44s
[... 245 lines cut ...]
:::::::::::::::::::::::::::[ ran 71 tests in 745s ]:::::::::::::::::::::::::::
17 tests returned not_applicable
36 tests returned pass
14 tests returned exception
   DebugBreakpoint > logs/20230629-202812-HiFive1Flash-DebugBreakpoint.log
   DebugExit > logs/20230629-202913-HiFive1Flash-DebugExit.log
   DebugFunctionCall > logs/20230629-203001-HiFive1Flash-DebugFunctionCall.log
   EtriggerTest > logs/20230629-203132-HiFive1Flash-EtriggerTest.log
   Hwbp1 > logs/20230629-203142-HiFive1Flash-Hwbp1.log
   Hwbp2 > logs/20230629-203231-HiFive1Flash-Hwbp2.log
   ItriggerTest > logs/20230629-203406-HiFive1Flash-ItriggerTest.log
   MemTestReadInvalid > logs/20230629-203527-HiFive1Flash-MemTestReadInvalid.log
   MemorySampleMixed > logs/20230629-203536-HiFive1Flash-MemorySampleMixed.log
   MemorySampleSingle > logs/20230629-203551-HiFive1Flash-MemorySampleSingle.log
   Registers > logs/20230629-203636-HiFive1Flash-Registers.log
   RepeatReadTest > logs/20230629-203725-HiFive1Flash-RepeatReadTest.log
   TooManyHwbp > logs/20230629-203852-HiFive1Flash-TooManyHwbp.log
   UserInterrupt > logs/20230629-204013-HiFive1Flash-UserInterrupt.log
4 tests returned fail
   IcountTest > logs/20230629-203326-HiFive1Flash-IcountTest.log
   InstantHaltTest > logs/20230629-203344-HiFive1Flash-InstantHaltTest.log
   Semihosting > logs/20230629-203732-HiFive1Flash-Semihosting.log
   SemihostingFileio > logs/20230629-203742-HiFive1Flash-SemihostingFileio.log
```
[hifive1-a01-tests-logs.tar.gz](https://github.com/riscv/riscv-openocd/files/11909850/hifive1-a01-tests-logs.tar.gz)

TommyMurphyTM1234, 2023-06-29:
I'm not sure that all of the tests are running reliably in this case (HiFive1 Rev A01).
Take the DebugExit one for example.
When run by `HiFive1.py` this passes.
But when run by `HiFive1-flash.py` it gets an exception.
This is part of the GDB log for the latter test:
```
(gdb) b _exit
b _exit
Breakpoint 2 at 0x204008bc: file programs/init.c, line 36.
(gdb) monitor riscv reset_delays 131
monitor riscv reset_delays 131
(gdb) c
c
Continuing.
^C
Program received signal SIGINT, Interrupt.
handle_trap (mcause=7, mepc=0x2040028a <rot13+182>, sp=0x80000088) at programs/init.c:28
28	    while (1)
(gdb) info breakpoints
info breakpoints
Num     Type           Disp Enb Address    What
2       breakpoint     keep y   0x204008bc in _exit at programs/init.c:36
(gdb) x/20i $pc
x/20i $pc
=> 0x204008a8 <handle_trap+94>:	j	0x204008a8 <handle_trap+94>
   0x204008aa <handle_trap+96>:	lw	ra,44(sp)
   0x204008ac <handle_trap+98>:	lw	s0,40(sp)
   0x204008ae <handle_trap+100>:	addi	sp,sp,48
   0x204008b0 <handle_trap+102>:	ret
```
For some reason, after the breakpoint is set on `_exit` and execution is continued, instead of hitting the breakpoint the target experiences an exception/fault with an `mcause` of `7` which is Store/AMO access fault.

This is a common pattern with other "exception" results.
Is there, perhaps, a problem with the test programs or the tests themselves that they fail for some reason other than failure of the capability that they're trying to test?

TommyMurphyTM1234, 2023-06-29:
There's also stuff like this:
```
(gdb) monitor riscv reset_delays 127
monitor riscv reset_delays 127
"monitor" command not supported by this target.
```
which is happening because the `monitor` command is being used before the `target extended-remote` command has been used to connect to the embedded target (via OpenOCD).

timsifive, 2023-06-29:
That should be cleaned up, but I doubt it affects anything.

JanMatCodasip, 2023-06-30:
Hi @TommyMurphyTM1234 ,

thank you for your time spent on this matter.

> TommyMurphyTM1234: Looks like some stuff is not up to date with the toolchain post separation of the Zicsr/Zifencei extensions out from the base integer ISA? (https://github.com/riscv/riscv-openocd/issues/869#issuecomment-1613014637)

Agreed, looks like for the newer compiler versions, the `march` has to be updated the way you suggested. For my test runs, I have used an [older compiler version](https://github.com/xpack-dev-tools/riscv-none-embed-gcc-xpack/releases/tag/v10.2.0-1.2), and therefore I haven't noticed.

> TommyMurphyTM1234: I made the change mentioned here: https://github.com/riscv-software-src/riscv-tests/issues/482#issuecomment-1613594248 and ran the tests again with these results: ....

I have compared 
- your test results (from [this comment](https://github.com/riscv/riscv-openocd/issues/869#issuecomment-1613699658)) 
- and my results (from [this comment](https://github.com/riscv/riscv-openocd/issues/869#issue-1769176709))

... and we're getting identical outcomes :-) 

Thanks for this independent confirmation.

JanMatCodasip, 2023-06-30:
So I believe this issue can be considered done, we have some sort of baseline for HiFive1 Rev A01 regressions.

I will close the ticket in a couple of days, unless there are further comments.

TommyMurphyTM1234, 2023-06-30:
> So I believe this issue can be considered done, we have some sort of baseline for HiFive1 Rev A01 regressions.

Sounds good. Thanks for comparing my results to yours and confirming that they're consistent.

> I will close the ticket in a couple of days, unless there are further comments.

Also sounds good. I'll take other issues (e.g. [the `zicsr` issue](https://github.com/riscv-software-src/riscv-tests/issues/482) , possible issues with some of the tests themselves etc.) offline from this issue and into separate issues as necessary.

MarekVCodasip, 2023-06-30:
@timsifive The corresponding test exclusion file has been submitted here: https://github.com/riscv-software-src/riscv-tests/pull/485

TommyMurphyTM1234, 2023-06-30:
> @timsifive The corresponding test exclusion file has been submitted here: [riscv-software-src/riscv-tests#485](https://github.com/riscv-software-src/riscv-tests/pull/485)

As I mentioned earlier, it seems to me that some of the tests result in a failure or exception status due to problems with the test programs and/or debug sessions themselves as opposed to being genuine results from the test. I am trying to investigate but my point in mentioning this is that I don't think that the HiFive1 results that I and @JanMatCodasip have achieved are "correct" and I think that it's prmature to mark some of the de-facto failure/exception tests as to be excluded.

JanMatCodasip, 2023-06-30:
My main goal in this thread was to clearly document he current state - known passing and known failing tests - regardless if it is a test issue or DUT issue.

I agree that there will most likely be issues in the test themselves that can be fixed or improved. But that can be the follow-up steps, outside of this thread, and whether that will happen depends on how much time people are able to commit to this :)

TommyMurphyTM1234, 2023-06-30:
> My main goal in this thread was to clearly document he current state - known passing and known failing tests on HiFive1 Rev A01 - regardless if it is a test issue or DUT issue.
> 
> I agree that there will most likely be issues in the test themselves that can be fixed or improved. But that can be follow-up steps, outside of this thread, and whether that will happen depends on how much time people are able to commit to this :)

OK - that makes sense. :-)

timsifive, 2023-06-30:
> I agree that there will most likely be issues in the test themselves that can be fixed or improved. But that can be follow-up steps, outside of this thread, and whether that will happen depends on how much time people are able to commit to this :)

I just filed https://github.com/riscv-software-src/riscv-tests/issues/488 for this. I'm unlikely to work on it in the foreseeable future.

TommyMurphyTM1234, 2023-06-30:
> > I agree that there will most likely be issues in the test themselves that can be fixed or improved. But that can be follow-up steps, outside of this thread, and whether that will happen depends on how much time people are able to commit to this :)
> 
> I just filed [riscv-software-src/riscv-tests#488](https://github.com/riscv-software-src/riscv-tests/issues/488) for this. I'm unlikely to work on it in the foreseeable future.

FWIW I will look at some or all of the HiFive1 failing/exception tests to see if any or them are failing because of the test program/debug session itself as opposed to the actual test failing.