riscv-collab/riscv-openocd #877: Error: Check JTAG interface, etc

cmenendezzz, 2023-07-04:
Hi, I have encountered a technical issue  working on a project in the NEXYS A7 100t board. Specifically, I am facing difficulties in establishing a connection with the JTAG interface with OPENOCD.

Due to the fact that J10 ports are not being recognized by Vivado, I have resorted to using a PMOD to extract the signals for JTAG. I have connected the JTAG HS2 and modified the OpenOCD configuration file accordingly. However, I am currently unable to establish a successful connection.

I have thoroughly reviewed my setup and configuration to ensure that all connections and settings are correct. Despite my efforts, I have been unable to resolve the connectivity issue. I have also consulted relevant documentation and online resources, but the problem persists.

Here you can see the error  and the configuration files.

Thank you in advance for your time.

[image]

[image]

[image]

TommyMurphyTM1234, 2023-07-04:
Whatever JTAG dongle is being used here seems to have TDO stuck at 1 so nothing is going to work in terms of OpenOCD/JTAG communication with the target.

I don't really understand the nature of the JTAG dongle you're using with OpenOCD here. Maybe you can explain in more detail? Explaining the schematics and setup of this may help.

cmenendezzz, 2023-07-04:
The idea is as follows: I want to synthesize the NEIRV32 on a NEXYS A7. I have been able to do this (including generating the bitstream and programming the device). Once this is done, the next step I want to take is to enable debugging using the JTAG HS2 probe. For this, on the one hand, I need to download GDB and configure it with my IDE (if I understood correctly), and on the other hand, configure OpenOCD. Currently, I am working on the latter but encountering the errors I mentioned earlier. You can find a schematic of the problem below.

file:///home/claudia/Downloads/IMG_9138.jpg

Also, here are the results of the 'lsusb' command.
```
claudia@claudia-VirtualBox:~$ lsusb
Bus 001 Device 004: ID 0403:6014 Future Technology Devices International, Ltd FT232H Single HS USB-UART/FIFO IC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

```
Finally, initially, I took inspiration from your HS2 config file. However, I had to add underscores ('_') in some of the commands. I was wondering if this was due to an older version of FTDI, or because I'm in a Linux environment, or something else. Thank you in advance.

cmenendezzz, 2023-07-04:
[image]

TommyMurphyTM1234, 2023-07-04:
> For this, on the one hand, I need to download GDB and configure it with my IDE (if I understood correctly), and on the other hand, configure OpenOCD. 

I don't really understand your points here.

If you want to use the latest OpenOCD with RISC-V support then build it as follows:
```
# Install any prerequisites - includingg those required in order to enable the required JTAG interface(s)
git clone https://github.com/riscv/riscv-openocd
cd riscv-openocd
./bootstrap
./configure --prefix=..
make install
```
> Currently, I am working on the latter but encountering the errors I mentioned earlier. 

Your main problem is that TDO is stuck at 1 so no communication with your target is going to work.
You need to figure out why TDO is stuck at 1.

> Finally, initially, I took inspiration from your HS2 config file. However, I had to add underscores ('_') in some of the commands. 

Again you are not explaining the issue here clearly enough.
What config file?
What "inspiration"? 
What commands?

TommyMurphyTM1234, 2023-07-04:
In case there's anything useful here...

* https://github.com/riscv/riscv-openocd/issues/597

cmenendezzz, 2023-07-04:
> Again you are not explaining the issue here clearly enough.
> What config file?
> What "inspiration"?
> What commands?

By executing the configuration file of the hs2 ( which can be found in riscv-openocd/tcl/interface/ftdi) an error occurs: 

```claudia@claudia-VirtualBox:~/riscv-openocd/tcl/interface/ftdi$ openocd -f digilent-hs2.cfg
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
digilent-hs2.cfg:6: Error: invalid command name "ftdi"
in procedure 'script' 
at file "embedded:startup.tcl", line 26
at file "digilent-hs2.cfg", line 6
```
Configuration file:
```
adapter driver ftdi
ftdi device_desc "Digilent Adept USB Device"
ftdi vid_pid 0x0403 0x6014

ftdi channel 0
ftdi layout_init 0x00e8 0x60eb

reset_config none
```

Whereas if a do the following modification , this error disappears. My question is if there is a mistake in the original file or my Linux environment is supposed to know those commands and he actually don't.

New configuration file:
```
adapter driver ftdi
ftdi_device_desc "Digilent Adept USB Device"
ftdi_vid_pid 0x0403 0x6014

ftdi_channel 0
ftdi_layout_init 0x00e8 0x60eb

reset_config none
```

The error disappears: 
```
claudia@claudia-VirtualBox:~/riscv-openocd/tcl/interface/ftdi$ openocd -f digilent-hs2.cfg
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6014, description 'Digilent Adept USB Device', serial '*' at bus location '*'
```

cmenendezzz, 2023-07-04:
>Your main problem is that TDO is stuck at 1 so no communication with your target is going to work.
You need to figure out why TDO is stuck at 1.
It might also be an idea - from a process of elimination point of view - to get your HS2 based JTAG connection working with a known good target first just to eliminate your FPGA RISC-V as being the culprit.
>

It is the first time I am doing this, that is why is quite frustrating sometimes... I am going to keep looking why it is stuck at 1.

Also, I forgot to precise on the last message that I used the latest OpenOCD with RISC-V support as you told me to. 

Finally, concerning the results of the `lsusb -v `command: 
```
claudia@claudia-VirtualBox:~/riscv-openocd/tcl/interface/ftdi$ lsusb -v

Bus 001 Device 004: ID 0403:6014 Future Technology Devices International, Ltd FT232H Single HS USB-UART/FIFO IC
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0403 Future Technology Devices International, Ltd
  idProduct          0x6014 FT232H Single HS USB-UART/FIFO IC
  bcdDevice            9.00
  iManufacturer           1 Digilent
  iProduct                2 Digilent USB Device
  iSerial                 3 210249B868A9
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 Digilent USB Device
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
[... 143 lines cut ...]
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval             255

```

TommyMurphyTM1234, 2023-07-04:
OK - my recommendations:

1. Clone, configure and build the latest RISC-V Openocd as explained earlier.
2. Make sure that it supports the Diligent HS2 interface (check the configuration log and the output of `openocd -c adapter list"`
3. Use the Diligent HS2 configuration script as-is - you should no longer need to modify it
4. If possible check that OpenOCD and Diligent HS2 work to connect to a known good debug target - not your FPGA design which may or may not be working
5. Finally try to connec to your FPGA design - if TDO remains stuck then you probably have some problem with the design or configuration of your target.

cmenendezzz, 2023-07-04:
Thank you very much for your help! I am going to follow you recommendations step by step and erase the older openocd that I downloaded. Again, thank you for your time, all your answers have been very useful!

TommyMurphyTM1234, 2023-07-05:
You're welcome.
I would suggest that you also double check your FPGA design to ensure that the target RISC-V and, in particular, debug block and JTAG are properly functional.
Perhaps by doing some appropriate simulated tests of the HDL.

cmenendezzz, 2023-07-05:
Hello again! 

I managed to successfully install the latest version of OpenOCD and I verified the debug block. Nevertheless, I noticed that by erasing the command `ftdi device_desc "Digilent Adept USB Device"` in the configuration file, I obtain different results. Is this command very important? 

Here you can see the error before erasing this command:
```
$ sudo openocd -f riscv-openocd/tcl/interface/ftdi/digilent-hs2.cfg -f neorv32/sw/openocd/openocd_neorv32.cfg
Open On-Chip Debugger 0.12.0+dev-02948-g92c031926 (2023-07-05-09:32)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate

Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6014, description 'Digilent Adept USB Device', serial '*' at bus location '*'

```

And here afterwards: 

```
$ sudo openocd -f riscv-openocd/tcl/interface/ftdi/digilent-hs2.cfg -f neorv32/sw/openocd/openocd_neorv32.cfg
Open On-Chip Debugger 0.12.0+dev-02948-g92c031926 (2023-07-05-09:32)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
jtag
Info : clock speed 2000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: neorv32.cpu: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Warn : target neorv32.cpu.0 examination failed
Info : starting gdb server for neorv32.cpu.0 on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet
```

Finally, this is the configuration file I am using: 
```
# SPDX-License-Identifier: GPL-2.0-or-later

# this supports JTAG-HS2 (and apparently Nexys4 as well)

adapter driver ftdi
#ftdi device_desc "Digilent Adept USB Device"
ftdi vid_pid 0x0403 0x6014

ftdi channel 0 
ftdi layout_init 0x00e8 0x60eb
adapter speed 2000
transport select jtag
reset_config none
```

TommyMurphyTM1234, 2023-07-05:
What happens if you use the string from the earlier `lsusb -v` output?
```
ftdi device_desc "Digilent USB Device"
```
This may also be relevant:

* https://www.mail-archive.com/openocd-devel@lists.sourceforge.net/msg14914.html

Either way, I don't think that the `ftdi device_desc` command really matters.

You still have the original problem - i.e. that TDO is stuck at 1.
So you should ideally verify that your JTAG connection works with a known good debug target such as an off the shelf MCU/board (e.g. some arm Cortex-M target if possible) and also recheck your FPGA design for issues, including simulating it to verify that the debug block/JTAG is functionally correct if possible.

BTW you shouldn't have to use `sudo` to run OpenOCD and doing so is not good practice. If, when you run it without `root` privileges then the issue is almost certainly that you need to install the OpenOCD `udev` rules file.

Usually this involves:

1. `sudo cp <path-into-openocd-installation>/60-openocd.rules /etc/udev/rules.d`
2. `udevadm trigger`
3. make sure that your user is a member of the `plugdev` group
4. In some cases you may also need to log out/in again or reboot
5.

TommyMurphyTM1234, 2023-07-05:
Isn't this a (the?) known good implementation of neorv32 on the Nexus A7 board? Did you try it making sure to follow the instructions to the letter?

* https://github.com/stnolting/neorv32-setups/tree/main/vivado/nexys-a7-test-setup

You might be better asking about debug issues with this combination of soft IP and FPGA board over at the neorv32 project?

* https://github.com/stnolting/neorv32-setups/issues

cmenendezzz, 2023-07-05:
Indeed I am implementing the NEORV32 on the Nexus A7 board.

Thank you very much! I did what you told me ( concerning the user privileges) and look at the doubts posted by other users and I forgot to put the reset jtag signal at high ( as it is optional i did not think about putting it at a high level).

I manage to solve all my problems ! Thank you very much for your time and patience .

closed 2023-07-06 by cmenendezzz