riscv-collab/riscv-openocd #881: which openOCD is better to use this one or the "Official OpenOCD Read-Only Mirror (no pull requests) "

InspireSemi, 2023-07-08:
How often are these two synced?  Is it better to use this one or the other one?

TommyMurphyTM1234, 2023-07-08:
> How often are these two synced?

See here:

* https://github.com/riscv/riscv-openocd/issues/790

> Is it better to use this one or the other one?

"Better" in what way exactly?
If you want the latest updates for RISC-V support then use this one.
If you want the latest updates for other targets and architecturally neutral parts of OpenOCD use upstream.
If you want both then you'll have to resolve the differences/merges yourself as long as they're out of sync.

TommyMurphyTM1234, 2023-07-12:
Not sure that this issue needs to stay open?

InspireSemi, 2023-07-12:
Yes you can choose this

TommyMurphyTM1234, 2023-07-12:
Can't you close it?
I can't (don't have the repo privileges) and I'm sure that @timsifive has enough to do.
Thanks.

closed 2023-07-19 by InspireSemi