riscv-collab/riscv-openocd #887: Error: Target neorv32.cpu.0: Failed to read memory (addr=0x3ce44)  Error:   progbuf=failed, sysbus=disabled, abstract=disabled

cmenendezzz, 2023-07-18:
Hi! I am currently trying to debug the NEORV32 open-source processor. I have successfully synthesized it and installed OpenOCD and GDB. For debugging, I am using the JTAG HS2 adapter and launching the debug session from Eclipse with a "hello world" code.

However, I encountered the following error:
```

Open On-Chip Debugger 0.12.0+dev-02955-g8032b7877 (2023-07-17-10:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
Info : clock speed 2000 kHz
Info : JTAG tap: neorv32.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Info : [neorv32.cpu.0] datacount=1 progbufsize=2
Info : Disabling abstract command reads from CSRs.
Info : Disabling abstract command writes to CSRs.
Info : [neorv32.cpu.0] Examined RISC-V core; found 1 harts
Info : [neorv32.cpu.0]  XLEN=32, misa=0x40801104
[neorv32.cpu.0] Target successfully examined.
Info : starting gdb server for neorv32.cpu.0 on 3333
Info : Listening on port 3333 for gdb connections
Target HALTED.
Ready for remote connections.
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"
Info : JTAG tap: neorv32.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Warn : Failed to read memory via program buffer.
Error: Target neorv32.cpu.0: Failed to read memory (addr=0x3ce44)
Error:   progbuf=failed, sysbus=disabled, abstract=disabled
Error: Target neorv32.cpu.0: Failed to read memory (addr=0x72578)
Error:   progbuf=failed, sysbus=disabled, abstract=disabled
Error: Target neorv32.cpu.0: Failed to write memory (addr=0x10094)
Error:   progbuf=failed, sysbus=disabled, abstract=disabled
Error: Target neorv32.cpu.0: Failed to write memory (addr=0x13bd0)
Error:   progbuf=failed, sysbus=disabled, abstract=disabled
Info : dropped 'gdb' connection
shutdown command invoked
```

Below is the configuration file used for OpenOCD:

```
adapter driver ftdi
ftdi vid_pid 0x0403 0x6014

ftdi channel 0
ftdi layout_init 0x00e8 0x60eb
adapter speed 2000
transport select jtag
reset_config none

ftdi layout_signal nTRST -ndata 0x0010 -noe 0x0040

# ----------------------------------------------
# Target configuration
# ----------------------------------------------
set _CHIPNAME neorv32
jtag newtap $_CHIPNAME cpu -irlen 5

set _TARGETNAME $_CHIPNAME.cpu

target create $_TARGETNAME.0 riscv -chain-position $_TARGETNAME

# ----------------------------------------------
# Access memory via program buffer
# ----------------------------------------------

riscv set_mem_access progbuf
# ----------------------------------------------
# Scratch pad RAM
# ----------------------------------------------
# work area ("scratch pad RAM"): beginning of (internal) DMEM, 256 bytes, requires(!) backup
$_TARGETNAME.0 configure -work-area-phys 0x80000000 -work-area-size 256 -work-area-backup 1

# ----------------------------------------------
# Expose NEORV32-specific CSRs
# ----------------------------------------------
riscv expose_csrs 4032

# ----------------------------------------------
# Start session
# ----------------------------------------------
init
halt

echo "Target HALTED."
echo "Ready for remote connections."
```

Thank you in advance!

cmenendezzz, 2023-07-18:
> Why is it attempting to read from/write to those addresses?

In fact, it is attempting to read these addresses because I am debugging the NEORV32. I presume it is trying to write to or read from the addresses where the debugging process should start.

TommyMurphyTM1234, 2023-07-18:
> > Why is it attempting to read from/write to those addresses?
> 
> In fact, it is attempting to read these addresses because I am debugging the NEORV32. I presume it is trying to write to or read from the addresses where the debugging process should start.

I know what you're trying to debug but that doesn't really address my question. Why **those** addresses?

And you didn't answer my other key question:

> Is there anything implemented in the memory map at these addresses?

Or, more simply, what is the memory map of your system?

If there's nothing at those addresses then it's quite possible that the error on accessing them is correct and not any indication of anything wrong, in particular with OpenOCD.

TommyMurphyTM1234, 2023-07-19:
Any update @cmenendezzz?

cmenendezzz, 2023-07-20:
Yes! Exactly, as you said the openocd is working all right. There is a problem with the addresses and the memory placement (I had to change the linker script). I am going to link this discussion with the other discussion in case someone will have the same issue as me. 

Thank you again @TommyMurphyTM1234 for your help!

cmenendezzz, 2023-07-20:
#https://github.com/stnolting/neorv32/discussions/642#

closed 2023-07-20 by cmenendezzz